aly-fly/EleksTubeHAX #1: BoingBall + FW Backup not displaying anything

saschaludwig, 2021-05-10:
If I flash the BoingBall, the displays remain all black.
On the serial I can see the fps numbers though.

I also tried flashing your original firmware: same result, black displays.
The fw dump from my clock looks a bit different in some places.
https://github.com/saschaludwig/EleksTubeHAX/blob/main/original-firmware/fw-backup-4M_sascha.bin

Any Idea what might be the difference?
My PCB says "EleksTube IPS Docker v6.1.11"

Best regards, Sascha

saschaludwig, 2021-05-10:
Thanks for pointing that out :-) my fault.
I'm a bit confused because TFT_CS and TFT_DC is redefined with different values at the end.
I will use the last ones and give it a try.

saschaludwig, 2021-05-10:
> Set width to 135. The same way you’re setting height.

That did the trick! :-)

and @frankcohen: yes, gave me goosebumps. awesome!

frankcohen, 2021-05-10:
Yeah!!!!!

---
Frank Cohen, Founder
***@***.***
http://www.votsh.com
Phone: (408) 364-5508 USA

saschaludwig, 2021-05-10:
For reference: https://www.youtube.com/watch?v=zj1XFj4SVxM