aly-fly/EleksTubeHAX #2: BoingBall+EleksTube general

jont-studioj, 2021-05-12:
Forgive me if this the wrong place / wrong way to get in touch.

I recently got the EleksTubeIPS and am interested in its hacking capability.

I have followed with interest (if not wholly absorbed) the discussion thread on Reddit.

It seems you people have already gotten a long way forward, nice work.

I _am_ a developer - but I'm now old and my brain isn't as spritely as it once was, I'd love to contribute (and still could) but fear I may be more of a hindrance than a help (as you are about to find out!)

I have done very little in the way of dealing with micro-controller projects (I've done just one Arduino project some years back) and have little experience in dealing with interface chips / libraries etc.

Riding on the shoulders of you good folk I have managed to check out the bouncing ball project and successfully got that built, running and (most importantly) restored the clock to its original f/w, using your instructions (and the "font-editor's" copy of esptool.exe, on Windows). Phew.

Wishing to see what can be done - and as a way to familiarise myself with the environment etc - I decided to see if could hack at the bouncing ball sketch to make it do something different.

C/C++ isn't my choice language (far far from it, I've managed to avoid those throughout my career - tbh, I actively dislike them!) so this would've been some bit of a way of forcing myself to learn a bit and get more familiar. But really, I am playing and not doing anything particularly useful. Yet.

Hacking at the code I came across an issue which I am struggling to overcome (I'm not expecting anyone to hand-hold me) but this issue has prompted me to write this text now.

I was struggling with the generation of graphics and the blitting of data to the displays, via the TFT_eSPI library. After much struggling trying and failing to get colours and stuff right.. I double checked the display/interface as described on https://www.aliexpress.com/item/32564432870.html and noticed that it says "Display Color: 262K"... and after some quick investigation this lead me to believe that the display takes 6bits per colour per pixel? rrrrrrbbbbbbgggggg (or whichever way round it really is)... which makes 18bits per pixel... I then peeked into the TFT_eSPI library/headers which _seemed_ (at least on casual reading) to suggest it deals with 1 4 or 8 bits per colour per pixel - and has nice 8bit/16bit data-size/buffer-size/boundaries for many of its functions.

Thinking that I want to send 135 pixels (a single display row) of 18 bits per pixel down to the display (or its controller chip or whatever) doesn't go nicely into any nice boundary/buffer-size of some byte/int or whatever.. hmm.. and this shows up even where I flail with c/c++! but never-mind.

( specifically a line like: `tft.pushPixelsDMA(&renderbuf[bufIdx][0], SCREENWIDTH);` and the preparation of said buffer etc )

It's quite possible that I am missing something.. but I wondered whether the TFT_eSPI library code is actually capable of driving this display sensibly?

Again, not familiar with how these things are done generally (or, indeed, I haven't read enough), is it the case that lots of code (including the library) can deal with nice power-of-two sized pixels/buffers (with the 6 bits of each colour padded out to a byte with zeros) and it goes all the way through that way until it reaches the actual display (or its controller)... where it is finally converted to the required 18bpp.. and that I shouldn't myself be trying to pack 3 colours of each 6-bits for a pixel spreading over a non-integer number of bytes inside buffers etc? Or is there a fundamental incompatibility between the library and the ultimate display?

Sorry for what might seem like dumb questions but I am somewhat flummoxed at the moment. Happy to delve further and work it out myself.. but not sure where I should be looking.. look harder into the library code? ..investigate the display's controller (ST7789) more? (or both, of course) - or maybe this is a genuine issue. Any clues, advice?

Regardless of whether you are able (or want) to help me, I'd like to say thank you for the work (investigation & dev) you've put in.

Many thanks in advance.

jont

frankcohen, 2021-05-12:
Hi Jont, welcome to the team! The water is warm, jump right in! :-)

I don’t much like C/C++ either. I’m much happier in Java and Python. One thing you might want to consider is getting MicroPython/CircuitPython running on the EleksTubeIPS. I’d be glad to help you.

I see your question about screen width, TFT_eSPI, and ST7789 and I wish I could say I had an answer. I don’t. My experience with these TFT displays is we stream a bunch of data to the 7789 controller and it deals with byte boundaries. I have crashed the controllers by sending it data it didn’t expect, but those crashes are few and far between.

The cool thing about the TFT_eSPI driver is all of the source code is right there for your investigation. You can see what it does to decode the image data and stream it over SPI to the display controller. And, my attitude is ‘meh, if it works that’s good enough for me’. hehehe

My next steps are to fire-up the Wifi interface on the ESP32 and see if I can stream images and video from an AWS hosted Web server over HTTP to the EleksTubeIPS. I’ll have more hacking time coming up this weekend.

You may find my other nights-and-weekends project to be interesting. Reflections is at https://github.com/frankcohen/ReflectionsOS. It’s a wrist watch that  lets me play videos of my kids while they were growing up on my wrist. It is distributed under a GPL v3 license. I’m hoping to provide the EleksTubeIPS hack with a Web browser interface by adapting the Reflections code.

Happy hacking!

-Frank

---
Frank Cohen, Founder
***@***.***
http://www.votsh.com
Phone: (408) 364-5508 USA

jont-studioj, 2021-05-12:
Thank you, Frank, for your warm welcome and generous reply.

I will re-read several times (I am slow to absorb info)... and look more deeply into the library stuff.. and/or discard what I've been playing with.

I notice that the project has moved on with more useful bits of code, while I am still playing in the sandbox that is the bouncing ball.

I guess I should be looking at those too.. to see alternate ways of dealing with the displays. I better learn from other people's experience (I wish I could learn something, anything, tbh)

Maybe the bouncing ball is a bad place for me to have started with so many unknowns (to me).. particularly with the buffered DMA updating (or however it is phrased) - that was the part that was causing me confusion - along with the bits-per-colour-per-pixel & how-the-hell-do-I-pack-these-bits! wait--you-want-me-to-squeeze-18bits-into-2bytes?!.. and all that jazz.. it's a big learning curve for me despite my long career. Getting too old for this.

I will plod on... again thanks for your reply. I will have a peruse of your other projects... thank you.

Thanks too to other contributors - on whose shoulders I am currently teetering, precariously.

Cheers
jont

frankcohen, 2021-05-12:
pssttt… (I just turned 60, you’re not too old). 

The bouncing ball demo is complicated. It comes from the TFT_eSPI examples. In Arduino_IDE use File->Examples/TFT_eSPI/Generic for easier examples. With the changes to the properties files you can get any of these to work on the EleksTubeIPS. And I’d love to see what you do.

-Frank

---
Frank Cohen, Founder
***@***.***
http://www.votsh.com
Phone: (408) 364-5508 USA

jont-studioj, 2021-05-12:
Oh you lucky young man! :-)

(I'm couple of years ahead of you, curse it).

Thanks for the extra advice.

I wanted to hack the bouncing ball example to use all the displays.. and to give you a clue.. it was to be called QuantumBall.

jont-studioj, 2021-05-12:
I have decided to junk what I have and start again (just playing, learning, sandbox-wise) with (a copy of) the EleksTubeHAX which looks to have been advanced in nice clean ways for the h/w (thanks to whoever!)

Not sure if it's my responsibility... but I will "close" this issue with this comment. Thanks again Frank.

closed 2021-05-12 by jont-studioj