aly-fly/EleksTubeHAX #3: missing Menu.h?

jont-studioj, 2021-05-13:
I just checked out the latest.. and there seems to be a header file "Menu.h" missing.. unless I've got something jolly wrong here (which, tbh, is pretty likely knowing me)

I don't know if raising an issue like this is the right thing to do... I couldn't find any way of leaving a comment.

jont

SmittyHalibut, 2021-05-13:
Curses! Did it not get added to the commit? Sorry, stand by. I’m out at lunch. I’ll add it back when I get home. 

-Mark

SmittyHalibut, 2021-05-13:
And, yes, an issue like this is a fine way to report this. :-)

jont-studioj, 2021-05-13:
cool... no real problem... I completely understand.. personally I have to use git for work but it confuses the hell out of me every time I use it. Mind you, it doesn't take much to confuse me nowadays (so git has no problem in that respect)

Thanks, hope you enjoy(ed) your lunch

jont

SmittyHalibut, 2021-05-13:
Fixed.  Sorry 'bout that.  New commit is pushed.

closed 2021-05-13 by SmittyHalibut

jont-studioj, 2021-05-13:
Many thanks - will re-checkout.