aly-fly/EleksTubeHAX #4: Joining WiFi sometimes fails. Be more graceful

SmittyHalibut, 2021-05-14:
Sometimes, when joining wifi on boot, it just keeps writing `.` to the TFT and never succeeds.   Be more graceful with wifi connection failures.

* Timeout and retry?
* Timeout and error on the TFT?
* Better error messages. Can we detect if the wifi password is incorrect so a misconfiguration and occasional failure don't look the same?

RedNax67, 2021-05-14:
Option to skip and run off RTC... (maybe people take it somewhere to demo.. who knows)

RedNax67, 2021-05-14:
Maybe this will help?

 /* Set the ESP to be a WiFi-client, otherwise by default, it acts as both a client and an access-point
  *  and can cause network-issues with other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);

This came from a very easy way to handle timezones. It's basically built in.. see here: https://github.com/G6EJD/ESP_Simple_Clock_Functions

SmittyHalibut, 2021-09-29:
I added your suggested code: `WiFi.mode(WIFI_STA);`  It seems a reasonable thing to do anyway.

aly-fly, 2023-01-10:
Mark, I think this is an old issue, never had any problems mentioned above. Probably you can close this as resolved?

SmittyHalibut, 2023-01-10:
But! But! But!  I opened it! </sarcasm>

Yeah, you're probably right.  This was firmware from ages ago, and the WiFi code has changed dramatically since then.

Resolving.

SmittyHalibut, 2023-01-17:
@ExplodingLemur just reported this problem happening to him. Reopening.

Its probably a good idea to put some retry logic around wifi connection failures.  That seems reasonable.