aly-fly/EleksTubeHAX #7: Consider adding a license

Aircoookie, 2021-05-20:
Hi and thank you for fully reverse engineering and providing an open-source alternative for the stock software!

I would like to use your code as the basis of a modification (a so-called usermod) to my [WLED](https://github.com/Aircoookie/WLED) project. WLED is an open-source ESP8266/ESP32 project to drive WS2812 and other addressable RGB LEDs with lots of possible lighting effects. It also already has support for NTP, timezones and automatic DST change implemented.

I would kindly like to ask for permission to re-use parts of your code in that context - of course giving full attribution and linking to this repo.

Aircoookie, 2021-05-21:
Once again, thank you for the amazing work!

I just went ahead and implemented the TFT driving part of your code as a WLED mod - now the backlight colors and effects are fully wirelessly controllable, as well as the IPS panel power! The bitmaps need to be uploaded to the `[Clock-IP]/edit` page of the embedded web server, as WLED uses LittleFS instead of SPIFFS. WiFi and timezone config is done via web-interface, the clock is initially reachable using the WLED-AP access point :)

Here is a binary in case you'd like to try it out: Erase flash, flash `esp32_bootloader_new.bin` to 0x0 and `elekstube_ips.bin` to 0x10000 using esptool. I would be very happy if it is ok for you if I publish the source in my repo. I have "stolen" part of the code, but giving full attribution :)
[WLED_Elekstube_IPS.zip](https://github.com/SmittyHalibut/EleksTubeHAX/files/6521155/WLED_Elekstube_IPS.zip)

RedNax67, 2021-05-21:
Hi,

Thanks for this! Maybe add that the ap pw is wled1234? As a collaborator on EleksTubeHAX i'd be very interested to see your code!

The webinterface on 4.3.2.1 looks really nice! The redirect doesn't work for me though.. but that could just be my setup,

Aircoookie, 2021-05-21:
I'm very glad you like it :)

I've now published the source at: https://github.com/Aircoookie/WLED/tree/master/usermods/EleksTube_IPS

Also made a tiny change, centering bitmaps with a height <240 vertically. This allows for cropping the black "bars" from the top and bottom of the Nixie images by about 28px in total, that way one image needs only 84 instead of 95kB. Still not enough to fit 2 complete 0-9 bitmap sets into the standard 1.5MB filesystem, but it's a start. Also makes the digits "flip" a little bit faster if multiple change at once - I love the slight effect of them changing sequentially though!

frankcohen, 2021-05-24:
Hi All, drats... I had seen WLED a while ago and forgot about it. I should have coded my Web interface as extensions to it. Instead I created my own here https://github.com/frankcohen/EleksTubeIPSHack. My code lets you upload and delete images to the SPIFFS from the browser, implements a slide show animation, and still runs the clock. I'll see about adding these to Aircookie's mod. -Frank

frankcohen, 2021-05-24:
To the issue at hand... seems to me GPL v3 is the right license, so that EleksMaker could incorporate any of our work in a future clock product, and we'd get to see the improvements. What do you think? -Frank

SmittyHalibut, 2021-09-29:
Sorry I've been, uhh, absent.  Wackiness ensuing at work that only recently got resolved.  Finally catching up to all this.

Yeah, GPL3 is good. I'll add the license.

SmittyHalibut, 2021-09-29:
Done. Resolving.

closed 2021-09-29 by SmittyHalibut

sbuxreg, 2021-11-11:
I'm hoping someone here can help me. When I go to [clock-ip]/edit to upload my digits the tab on my web browser says "ESP Editor" but the screen itself is just a white screen with a dark bar across the top and a line off towards the left side. Here is a picture for refrence: 
https://ibb.co/VMDtZQK
Can anyone tell me what might be causing this and how i can fix it?

Thank you in advance!