aly-fly/EleksTubeHAX #8: Original firmware backup does not work on all Elekstube IPS clocks

neptune2, 2021-05-22:
Firstly - thanks for creating the alternate firmware and documenting all the steps - compiled and worked first time.

I discovered that your original [fw-backup-4M.bin](https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/original-firmware/fw-backup-4M.bin) does not work on the new EleksTube IPS clock I just received.
Fortunately, I had made a backup of my firmware before uploading new Firmware and my backup firmware does restore correctly.
I checked multiple times with different baud rates - same results.

I'm not sure what is different - all the parts look the same as your documented part numbers.

Please consider adding a _Warning for prospective hackers to backup their firmware **before** uploading new firmware_.
Someting similar to:
```
# backup the firmware from the esp32
esptool --baud 115200 --port COM3 read_flash 0x0 0x400000 my_fw-backup-4M.bin
```

RedNax67, 2021-05-22:
Interesting, would you mind sharing your backup? Maybe we can find out what the difference is...

koshisan, 2021-05-27:
Sounds like my problem - only I didn't make a backup of mine ;)
@neptune2 would you care to share yours?

koshisan, 2021-05-29:
Is this still alive? Since @neptune2 never replied here again I took a dump from my second clock and the binary files indeed differ largely...
[fw-backup-jz-4M.zip](https://github.com/SmittyHalibut/EleksTubeHAX/files/6563922/fw-backup-jz-4M.zip)

frankcohen, 2021-05-29:
Hi Koshisan, you are welcome to try my binary file and alternative firmware (based on smitty's work). It's at https://github.com/frankcohen/EleksTubeIPSHack. -Frank

koshisan, 2021-05-29:
> 
> 
> Hi Koshisan, you are welcome to try my binary file and alternative firmware (based on smitty's work). It's at https://github.com/frankcohen/EleksTubeIPSHack. -Frank

I know, I am ksh_osaka in reddit ;) I was just wondering since all other contributors have been silent for a while now...

neptune2, 2021-05-29:
Hi Jens @koshisan, I have found how to convert @RedNax67 original firmware in this repo to work on my clock!

The discovery is that the MAC address in the firmware must match the factory programmed MAC address in your clock.

Here are the steps (in Windows):

1. Find the MAC address in your clock:
`esptool flash_id`
Look for the line that starts MAC:

2. Replace all 42 instances of the MAC address in your copy of https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/original-firmware/fw-backup-4M.bin.
Note that the 6 MAC bytes in the firmware are in reverse byte order from the MAC: xx:xx:xx:xx:xx:xx format in step 1.
I used WinMerge https://winmerge.org/ to edit the firmware binary with the following steps:
```
2a. Make a copy of fw-backup-4M.bin (I named it edited_fw-backup-4M.bin)
2b. Open both bin files side-by-side in WinMerge 
2c. Select the edited_fw-backup-4M.bin pane in WinMerge
2d. Edit->Replace <bh:80><bh:93><bh:2f><bh:84><bh:db><bh:e8> with <bh:xx><bh:xx><bh:xx><bh:xx><bh:xx><bh:xx>
     where the xx bytes are replaced with your clock MAC address bytes (reverse order from Step1. format). 
     The last 2-3 bytes should be the same.
2e. Save the edited_fw-backup-4M.bin file
```
3. Restore the edited_fw-backup-4M.bin firmware to your clock:
`esptool --baud 115200 --port [COM port] write_flash 0x0 edited_fw-backup-4M.bin`

Hopefully this also works on your clock. 
Please reply with you results. 頑張って

SmittyHalibut, 2021-09-29:
Thank you @neptune2  for finding this and coming up with the solution.  I've updated `original-firmware/README.md` with your instructions:

https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/original-firmware/README.md

Resolving. :-)

closed 2021-09-29 by SmittyHalibut