aly-fly/EleksTubeHAX #9: Merging my stuff

frankcohen, 2021-05-24:
Hi u/SmittyHalibut, u/RedNax67, would you like to help me merge my changes with the repository? I'm thinking it would look like this:

ElecksTubeHAX - split-out the Web UI into a web.h, web.cpp module, add support for backlights during the slide show.

TFT.h, TFT.cpp - added the JPEG decoder and slide side animation

Clock.h, clock.cpp - change error messages when needed image files not available

StoredConfig.h, StoredConfid.cpp - added setting to remember the slide show, video show, and clock settings on restart

wifi_creds.h - use to set the default Wifi values, and present it in the Web UI

menu.h, menu.cpp - keep for people wanting to use the buttons. The Web UI doesn't do anything with the menu features

And incorporate Aircookie's WLED mod.

What do you think?

-Frank

frankcohen, 2021-05-24:
And does anyone object to my adding myself as a contributor? I'd like to add this to the bottom of the readme.md file:

Frank Cohen, https://www.reddit.com/user/frankcohen, founder of StarlingWatch.com and
inventor of the Reflections ESP32/TFT wrist watch project https://github.com/frankcohen/ReflectionsOS
See Frank's alternative EleksTubeIPS firmware for a Web UX to this project at
https://github.com/frankcohen/EleksTubeIPSHack (working to merge these repositories)

RedNax67, 2021-05-25:
Hi Frank, 

No problem at all, when anything pops up, we can always go back :)

SmittyHalibut, 2021-09-29:
Sorry, @frankcohen and @RedNax67 I did all this original work while on an FMLA break from work.  Once I went back to work, my life kinda exploded.  I, uhh, don't work there anymore (see my Twitter if you're curious) so I finally have the time an energy to come back and look at this again.

I do NOT object to merging your changes in here.  Having just rebased my local code, I don't see your files in yet.  Feel free to file a pull request, or since you're a contributor, just rebase and push your changes directly.

Thank you again, both of you, for the work you've done on this.  Again, I'm really sorry for ghosting on you for the last several months.

I'll leave this Issue open until you get your code merged.

SmittyHalibut, 2021-12-26:
@frankcohen @RedNax67 Are you able to merge this code? Or, submit a PR and I'll review and merge it?

frankcohen, 2021-12-27:
Hi Mark, no worries on the project, or feeling like you ghosted anyone. I loved the time we working on the EleksTubeHAX together. Thank you!

Look, if the Elekstube people themselves aren't interested or at least indifferent to our hack's goals then why should any of us be interested. I've been to China 6 times, spoken at the Maker Fair Shenzhen, and now have very good friends in Southern China. Elekstube IPS clock met their goals for something fun, something inexpensive, and something tech savvy. And in the past year we've seen copies of IPS clock with different processors and board versions. None of them are compatible with the original. So I don't see improving the repository.

Instead, I am working on my own project called Reflections. The Reflections repository is at https://github.com/frankcohen/ReflectionsOS. It will all be open source, deliver a Web-based app and an inexpensive ESP32 board, and hopefully be fully supported by @SmittyHalibut and @RedNax67 and everyone else.

I'd love to know your thoughts, all of you!

-Frank

PS- I don't know if you checked-out my Photobooth hack:
https://www.reddit.com/r/arduino/comments/oc33jr/fun_photobooth_hack_on_the_elekstube_ips_clock/

SmittyHalibut, 2023-01-09:
Wow.  WAY behind on issues.

Awesome, @frankcohen Glad to see more projects.  :-)

Resolving this issue.

closed 2023-01-09 by SmittyHalibut