aly-fly/EleksTubeHAX #14: "DRAM segment data does not fit."

phar, 2021-07-18:
Users/User/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-gcc -nostdlib -L/Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib -L/Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/ld -T esp32_out.ld -T esp32.project.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u esp_app_desc -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/Backlights.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/Button.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/ChipSelect.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/Clock.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/EleksTubeHAX.ino.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/Menu.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/StoredConfig.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/sketch/TFTs.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Preferences/Preferences.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Adafruit_NeoPixel/esp.c.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Adafruit_NeoPixel/esp8266.c.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Adafruit_NeoPixel/kendyte_k210.c.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Adafruit_NeoPixel/rp2040.c.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/FS/FS.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/FS/vfs_api.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/SPIFFS/SPIFFS.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/TFT_eSPI/TFT_eSPI.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/SPI/SPI.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Time/DateStrings.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Time/Time.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/ETH.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFi.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiAP.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiClient.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiGeneric.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiMulti.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiSTA.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiScan.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiServer.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/WiFi/WiFiUdp.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/NTPClient/NTPClient.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/DS1307RTC/DS1307RTC.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/libraries/Wire/Wire.cpp.o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/core/core.a -lgcc -lesp_websocket_client -lwpa2 -ldetection -lesp_https_server -lwps -lhal -lconsole -lpe -lsoc -lsdmmc -lpthread -llog -lesp_http_client -ljson -lmesh -lesp32-camera -lnet80211 -lwpa_supplicant -lc -lmqtt -lcxx -lesp_https_ota -lulp -lefuse -lpp -lmdns -lbt -lwpa -lspiffs -lheap -limage_util -lunity -lrtc -lmbedtls -lface_recognition -lnghttp -ljsmn -lopenssl -lcore -lfatfs -lm -lprotocomm -lsmartconfig -lxtensa-debug-module -ldl -lesp_event -lesp-tls -lfd -lespcoredump -lesp_http_server -lfr -lsmartconfig_ack -lwear_levelling -ltcp_transport -llwip -lphy -lvfs -lcoap -lesp32 -llibsodium -lbootloader_support -ldriver -lcoexist -lasio -lod -lmicro-ecc -lesp_ringbuf -ldetection_cat_face -lapp_update -lespnow -lface_detection -lapp_trace -lnewlib -lbtdm_app -lwifi_provisioning -lfreertos -lfreemodbus -lethernet -lnvs_flash -lspi_flash -lc_nano -lexpat -lfb_gfx -lprotobuf-c -lesp_adc_cal -ltcpip_adapter -lstdc++ -lc-psram-workaround -lm-psram-workaround -Wl,--end-group -Wl,-EL -o /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/EleksTubeHAX.ino.elf
Multiple libraries were found for "WiFi.h"
 Used: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /private/var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/AppTranslocation/77AA4AAA-DDCC-4DAB-8110-06E51642DA41/d/Arduino.app/Contents/Java/libraries/WiFi
/Users/User/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /var/folders/2n/5bg52f0s13l3kxljs5ylvkx80000gn/T/arduino_build_743203/EleksTubeHAX.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/Users/User/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/User/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 69656 bytes
collect2: error: ld returned 1 exit status
Using library Preferences at version 1.0 in folder: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Preferences 
Using library Adafruit_NeoPixel at version 1.8.0 in folder: /Users/User/Documents/Arduino/libraries/Adafruit_NeoPixel 
Using library FS at version 1.0 in folder: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPIFFS 
Using library TFT_eSPI at version 2.3.61 in folder: /Users/User/Documents/Arduino/libraries/TFT_eSPI 
Using library SPI at version 1.0 in folder: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPI 
Using library Time at version 1.6 in folder: /Users/User/Documents/Arduino/libraries/Time 
Using library WiFi at version 1.0 in folder: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi 
Using library NTPClient at version 3.2.0 in folder: /Users/User/Documents/Arduino/libraries/NTPClient 
Using library DS1307RTC at version 1.4.1 in folder: /Users/User/Documents/Arduino/libraries/DS1307RTC 
Using library Wire at version 1.0.1 in folder: /Users/User/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire

pedrompcaetano, 2021-09-18:
I also ran into that error.
Simple PEBCAK.

The instructions are clear, but I inadvertently pointed the define in User_Setup_Select.h to default User_Setup.h instead of the project User_Setup.h.
After correctly following instructions build was successful.

SmittyHalibut, 2021-09-29:
@phar Does @pedrompcaetano 's comment address your issue?

phar, 2021-10-03:
"PEBCAK" was not helpful, however your nudge reminded me that this was a thing and caused me to see what happens this time, a sync on the source tree and a rebuild worked without issue, Arduino 1.8.13 and all libraries left unchanged. Appreciate the fix and the checkin.

closed 2021-10-03 by phar

SmittyHalibut, 2021-10-04:
@phar PEBKAC means "Problem exists between keyboard and chair." Its a short way of saying "I screwed up." :-)

I'm glad you got it working.

phar, 2021-10-04:
I understood the acronym, It was unhelpful and incorrect, thanks for the great project.