aly-fly/EleksTubeHAX #16: NTP Sync - Locale settings?

TheAnticitizen1, 2021-09-24:
Sorry if this is the wrong place to post this, I'm very much a n00b when it comes to Arduino. The time sync for me is wrong, I'm trying to set it to London UK but I can't see where to specify the locale?
I tried doing the offset via the menu items, but the 15 minute minimal I'd end up 13 minutes off the correct time. 

Any help greatly appreciated! :)

TheAnticitizen1, 2021-09-24:
Just a bit more of an update.. at the time of writing this comment, here's what I'm seeing:
EleksTube: 12:50:04
UTC: 15:35:04
GMT: 16:35:05

SmittyHalibut, 2021-09-29:
There are two menu settings, one for an hour offset, and one for a 15 minute offset, because some time zones are :30 or :45 minutes off UTC.  I did not try to put the timezone database in the code, instead just letting the user select the offset.

So, in the UK in daylight saving time, you're UTC +1:00.  So you'd set the Hour offset to `+1` and the 15 Minute offset to `+00`.

The clock doesn't do any DST automation.  You'll have to adjust the hour offset twice a year.  Sorry about that.

Does this address you question?

closed 2021-12-26 by SmittyHalibut