aly-fly/EleksTubeHAX #19: печатка

Yevgeniy135, 2022-01-29:
здравствуйте! Мне очень понравился ваш проект!  И чень хочется собрать самостоятельно такие часики, во время сборки естественно, будут и вопросы. можно будет вам здесь их задавать ? и можно мне их собрать? можно будет у вас попросить печатные платы ?

SmittyHalibut, 2022-01-29:
Translation according to my iPhone:

> Hello!? I really liked your project! And I really want to collect such a watch on my own, during assembly, of course, there will be questions. Can I ask you here? And can I collect them? Can I ask you for printed circuit boards?

These are not a kit. These are a product sold by several companies on eBay or AliExpress. This GitHub repo is only an after market firmware that adds NTP to the original product. 

A link to the original products can be found in README.md .

I'm glad you like it! We do too! :-)

Yevgeniy135, 2022-01-29:
Очень даже жаль, я думал что ваша прошивка подойдёт. Вот и решил собрать (естественно не за бесплатно)  хотите сказать, что эта прошивка не заработает если я сам соберу схему? Я правильно понимаю?

SmittyHalibut, 2022-01-30:
> It's a pity, I thought your firmware would do. So I decided to assemble (of course not for free) do you want to say that this firmware will not work if I assemble the scheme myself? Do I understand correctly?

If you buy the same parts and assemble it the way I describe in the reverse Engineering document, it should work. But I can't support it.

SmittyHalibut, 2022-01-30:
Can you buy one here? Will they ship to your country? https://www.ebay.com/itm/EleksTube-IPS-NIXIE-RGB-PSEUDO-Glow-Tube-Clock-Customized-Dial-Styles-Nice-Gifts-/164808970266

Yevgeniy135, 2022-02-15:
Здравствуйте ещё раз! В общем, начал я собирать и хочу спросить. Можно ещё внедрить в прошивку (и физически) поддержку температур? Хочу это обсудить в личном сообщении с человеком который мне в этом поможет! Собираю часы только  для себя. Продавать я их точно не буду! А покупать готовый вариант очень дорого (особенно для нашей страны. Заранее спасибо за помощь!

SmittyHalibut, 2022-02-15:
> Hello again! In general, I began to collect and I want to ask. Can I still introduce temperature support into the firmware (and physically)? I want to discuss this in a personal message with a person who will help me with this! I collect watches for myself. I won't be selling them! And buying a ready-made version is very expensive (especially for our country. Thanks in advance for your help! 

Yes, you can definitely add support for a temperature sensor.  I don't know which sensor hardware you want to add, but there is already an I2C bus in use for the RTC, you can add a temperature sensor to that bus.  If the temperature sensor is analog, I can't remember whether the ESP has an available ADC pin.

Unfortunately, I don't have the resources right now to help you with this.  Maybe someone else reading this thread does and will speak up.  Otherwise, there are good tutorials (in English at least) for temperature sensors and microcontrollers.  I don't know what's available in Russian, or how well Google Translate would work on a English technical tutorial.

aly-fly, 2022-02-17:
I am working on adding a temperature sensor. Clock is connected over MQTT to Smartnest cloud and SmartThings cloud. Both have display (and history) for temperature so it is straightforward to add temperature measurement. I plan to add 1-Wire DS18x20 sensor outside the housing (as clock gets pretty warm). Flollow me for updates.

[image]
[image]

Yevgeniy135, 2022-02-17:
> I am working on adding a temperature sensor. Clock is connected over MQTT to Smartnest cloud and SmartThings cloud. Both have display (and history) for temperature so it is straightforward to add [...]

Хорошая идея, я  этоимел и в виду когда спрашивал. Буду следить за вашими результатом.  Просто я не очень силен в создании скетч (прошивки) вот по этому и обратился сюда на форум))))  я уже заказал дисплеи, теперь жду посылку и буду пробовать собирать и экспериментировать

aly-fly, 2022-02-19:
_Good idea, that's what I meant when I asked. I will follow your result. It's just that I'm not very good at creating a sketch (firmware), that's why I turned to the forum here)))) I already ordered displays, now I'm waiting for the package and I'll try to assemble and experiment_
Good luck. You will have to learn:

- install Arduino IDE
- configure all the plugins and tools for ESP32
- save original firmware from your clock
- load files from GitHub to your PC
- compile and upload firmware 
- upload images

SmittyHalibut, 2023-01-09:
First off, let me say that I love we were able to use technology to help us communicate across language barriers.  :-)

I wish you luck in your project, @Yevgeniy135.  But I'm going to resolve this issue now.  :-)

closed 2023-01-09 by SmittyHalibut