aly-fly/EleksTubeHAX #20: Compile error

Rambotnik87, 2022-02-17:
I followed the instructions, but I can't compile the project. Can anyone help me?
These are some of the errors I get when compiling:

> In file included from C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\Backlights.cpp:1:0:
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\Backlights.h: In constructor 'Backlights::Backlights()':
> Backlights.h:64:37: error: 'Backlights::config' will be initialized after [-Werror=reorder]
>    StoredConfig::Config::Backlights *config;
>                                      ^
> Backlights.h:60:8: error:   'bool Backlights::pattern_needs_init' [-Werror=reorder]
>    bool pattern_needs_init;
>         ^
> Backlights.h:22:3: error:   when initialized here [-Werror=reorder]
>    Backlights() : config(NULL), pattern_needs_init(true), off(true),
>    ^
> Backlights.h:61:8: error: 'Backlights::off' will be initialized after [-Werror=reorder]
>    bool off;
>         ^
> Backlights.h:23:71: error:   base 'Adafruit_NeoPixel' [-Werror=reorder]
>      Adafruit_NeoPixel(NUM_DIGITS, BACKLIGHTS_PIN, NEO_GRB + NEO_KHZ800)
>                                                                        ^
> Backlights.h:22:3: error:   when initialized here [-Werror=reorder]
>    Backlights() : config(NULL), pattern_needs_init(true), off(true),
>    ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\Backlights.cpp: In member function 'void Backlights::setIntensity(uint8_t)':
> Backlights.cpp:51:59: error: suggest parentheses around '-' inside '>>' [-Werror=parentheses]
>    setBrightness(0xFF >> max_intensity - config->intensity - 1);
>                                                            ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\Backlights.cpp: In member function 'void Backlights::loop()':
> Backlights.cpp:69:76: error: suggest parentheses around '-' inside '>>' [-Werror=parentheses]
>          setBrightness(0xFF >> max_intensity - (BACKLIGHT_DIMMED_INTENSITY) - 1);
>                                                                             ^
> Backlights.cpp:71:65: error: suggest parentheses around '-' inside '>>' [-Werror=parentheses]
>          setBrightness(0xFF >> max_intensity - config->intensity - 1);
>                                                                  ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\Backlights.cpp: In member function 'void Backlights::rainbowPattern()':
> Backlights.cpp:185:61: error: suggest parentheses around '-' inside '>>' [-Werror=parentheses]
>      setBrightness(0xFF >> max_intensity - config->intensity - 1);  
>                                                              ^
> In file included from C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.cpp:1:0:
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.h: In member function 'bool ChipSelect::isSecondsOnes()':
> ChipSelect.h:35:71: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    bool isSecondsOnes()                            { return (digits_map&SECONDS_ONES_MAP > 0); }
>                                                                        ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.h: In member function 'bool ChipSelect::isSecondsTens()':
> ChipSelect.h:36:71: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    bool isSecondsTens()                            { return (digits_map&SECONDS_TENS_MAP > 0); }
>                                                                        ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.h: In member function 'bool ChipSelect::isMinutesOnes()':
> ChipSelect.h:37:71: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    bool isMinutesOnes()                            { return (digits_map&MINUTES_ONES_MAP > 0); }
>                                                                        ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.h: In member function 'bool ChipSelect::isMinutesTens()':
> ChipSelect.h:38:71: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    bool isMinutesTens()                            { return (digits_map&MINUTES_TENS_MAP > 0); }
>                                                                        ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.h: In member function 'bool ChipSelect::isHoursOnes()':
> ChipSelect.h:39:71: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    bool isHoursOnes()                              { return (digits_map&HOURS_ONES_MAP > 0); }
>                                                                        ^
> C:\Users\User\Desktop\latest\EleksTubeHAX-main\EleksTubeHAX\ChipSelect.h: In member function 'bool ChipSelect::isHoursTens()':
> ChipSelect.h:40:71: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    bool isHoursTens()                              { return (digits_map&HOURS_TENS_MAP > 0); }
>                                                                        ^

closed 2022-02-17 by Rambotnik87

SmittyHalibut, 2022-02-18:
Did you close this ticket on accident?  Or did you figure out your problem?

If you figured it out, please post your solution so anyone in the future who runs into the same problem can learn from your hard earned wisdom. :-)

If it was closed on accident, then let's re-open it and dig in.

Rambotnik87, 2022-02-18:
Hello thank you for your response, i figured how to compile and sent your
firmware. But with no succes, clock still black with no boot or numbers.
I tried also backup original via esptool, it sucessfully uploaded, but
nothing happened with my clock. Can you pls help me?

aly-fly, 2022-02-19:
First you upload the firmware. Then you have to upload the files with clock images.
Use the Serial monitor tool immediately after upload to see what is going on with the clock.

Which "backup" did you use? Backup only works if you have saved original firmware from your own clock. Uploading any other "backup" file will not work (MAC address of ESP32 module is stored in a FW and is used like a license key).

Rambotnik87, 2022-02-19:
First i erased flash via esptool. Installed all libraries and setuped. Compile and upload firmware with no errors and upload files with clock images via Sketch data upload SPIFFS with no errors. But displays are still black, no text is displayed when booting.
Here is serial monitor after upload: 

```
In setup().
Config size: 88
Loaded Backlights config is invalid, using default.  This is normal on first boot .                                             
Loaded Clock config is invalid, using default.  This is normal on first boot.
00:00:05
syncProvider()
Getting NTP.. Done.
NTP, RTC, Diff:
6
0
6
syncProvider()
Using RTC time.
```

[IMG1](https://postimg.cc/njt3zVQg)
[IMG2](https://postimg.cc/k6nWz2SJ)

aly-fly, 2022-02-19:
It looks like your clock is from different manufacturer. Our is from EleksMaker / EleksTube. Yours is  from Siahi or something similar. It looks similar, but it is possible that some hardware or connections are different.

Rambotnik87, 2022-02-19:
@aly-fly I found that out, too. I also contacted the dealer about that and he said I used unapproved software and he can't help me. According to him, there is no other software than theirs, that can be loaded on this clock and he cannot send me that. The only thing he can send me is a new board for $30. Is it possible that this board is blocked against overwriting and nothing else works on it?

aly-fly, 2022-02-20:
In theory, I think they could set some eFuses in ESP32. But most probably they didn't go that far, even original clock doesn't have such hard protection. I think they changed something in hardware, like pin connections so that different firmware would not work. I suggest reverse-engineering their board. And then we can include those details in our firmware to support it. If you can't do that, send board to one of us that can do it and then return it to you. I am in Europe and can help.

Rambotnik87, 2022-02-20:
> In theory, I think they could set some eFuses in ESP32. But most probably they didn't go that far, even original clock doesn't have such hard protection. I think they changed something in hardware, [...]

Can you please email me at rambotnik87@gmail.com?