aly-fly/EleksTubeHAX #24: How to use without WPS

triggad23, 2022-05-05:
Hi,

how can I use this without WPS? My Wifi APs don't support WPS.

Greets
Daniel

aly-fly, 2022-05-06:
Hello, currently only WPS is implemented. No manual WiFi selection.
You can replace the line `wifi.begin();` with `wifi.begin(YOUR_SSID, YOUR_PASSWORD);` and this should work fine for your case.

kogeki, 2022-12-21:
> Hello, currently only WPS is implemented. No manual WiFi selection. You can replace the line `wifi.begin();` with `wifi.begin(YOUR_SSID, YOUR_PASSWORD);` and this should work fine for your case.

Hello,

Which file should I change? In WiFi_WPS.cpp, there are 2 lines.

SmittyHalibut, 2022-12-21:
https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/EleksTubeHAX/EleksTubeHAX.ino#L60

SmittyHalibut, 2023-01-09:
https://github.com/SmittyHalibut/EleksTubeHAX/pull/30 implements this.  Resolving.

closed 2023-01-09 by SmittyHalibut