aly-fly/EleksTubeHAX #28: Cannot read or upload after hardware modification

Yami661995, 2022-08-25:
I have a EleksTube IPS Gen2 clock which can sync time via wi-fi. I wanted to try this custom firmware and after reading the guide, i tried hardware modification as written in the doc. But after that, i can no longer upload clock styles using EleksTube tool, i got it from here: https://elekstube.com/pages/elekstube-ips-faces-download, the "EleksTube IPS_NTP_NEW" one.
The clock is still recognized by my PC, and i still can use the tool to perform basic functions like changing styles, setting wi-fi, setting backlights. But when i try to upload styles, the tool stuck at 2%, the digit screens turn off and kind off flashing with black screen, and after some time, the clock rebooted, but the clock styles have not changed at all. 
I also cannot backup using _ESP32 save flash 4MB.cmd file, it keeps showing "Connecting" and then "A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header". I have already tried different baud rates.
Sorry but I'm noob at this. What could i done wrong? How can i check if i broke something?

aly-fly, 2022-08-25:
Hi, I am not aware of IPS Gen2 clock hardware. All development described here was done on original hardware. It looks like something was changed on Gen2 and you broke you clock performing the modifications. There is no connection between USB chip and ESP32 chip. Best thing is to revert the changes, to get your hardware back into original state. And hopefully it will work again.
p.s. Can you send some high resolution photos of your hardware? Where and when did you purchase it?

frankcohen, 2022-08-25:
Smitty started the project as a fun way to experiment with the Elekstube IPS clock. I’ve tried to get Elekstube to open up their firmware with no success. Over the past 2 years they have released new hardware versions. There aren’t any guarantees the firmware hack works

It’s a fun ESP32 hardware environment, and easily hackable. For example, my own experiments lead me to do this:
https://vimeo.com/568616803/453c2f24b7

It’s part of my repository at https://github.com/frankcohen/EleksTubeIPSHack

-Frank

Yami661995, 2022-08-25:
> Hi, I am not aware of IPS Gen2 clock hardware. All development described here was done on original hardware. It looks like something was changed on Gen2 and you broke you clock performing the [...]

I have reverted the hardware change but no luck, still cannot upload. I purchased it from Elekstube site: https://elekstube.com/products/elekstube-ips-rgb-digital-clock-imitate-nixie-tubes-clock-6-digit-lcd-proposal-mode-custom-style-best-boyfriend-gift-shelf-clock
Here are the pictures of the board after i reverted the change 
[image]
[image]

Yami661995, 2022-08-25:
I just remembered there's one time i accidentally shorted the 5V line with GND at the back of the board. When i connected the board to my PC it played a disconnected USB device sound and the clock was not booting. Maybe that the cause

aly-fly, 2022-08-25:
From the first look I can't spot any major hardware change. How do you know it is "Gen2"? 
Shorting 5V and GND will not do any harm if powered by PC because USB ports at the computer are protected against shorts. It should also not do any harm to the clock. But is you shorted 5V to any other signal line anything could happen...
Please review your modifications in detail. Or contact someone nearby that has electronic skiils (or logoc analyzer / oscilloscope) to track down the issue.

Yami661995, 2022-08-25:
> From the first look I can't spot any major hardware change. How do you know it is "Gen2"? Shorting 5V and GND will not do any harm if powered by PC because USB ports at the computer are protected [...]

I don't know about the original board so i just called it with the name from Elekstube site. There's no one who has electronic skill in my area so i think i'll try to figure this issue out. Currently i only have a multimeter, can i use it to measure anything suspicious?

aly-fly, 2022-08-26:
I think multimeter is not enough, as you are dealing with high-frequency signals (data transmission lines) between USB chip and ESP chip.
So you say clock is recognized by the computer (COM port appears when you plug the USB cable)? This means USB chip has power. The mentioned HW modification removes 5V supply from this chip and replaces it with 3.3V. And you reverted it back to 5V.

Yami661995, 2022-08-26:
COM port appears when i plug in USB, but it seems like the esp32 chip is not recognized. In Arduino->Tools->Get Board Info, it show up as unknown board like this:
BN: Unknown board
VID: 1A86
PID: 7523
SN: Upload any sketch to obtain it

I also tried to get some info with esptool like read_mac, flash_id, chip_id but no luck. I think the esp32 chip broke somewhere

SmittyHalibut, 2023-01-09:
It definitely sounds like something blew up with the hardware.  Sorry. :-/

This issue has been idle since August 2022.  I'm going to resolve it now.  Hopefully you were able to figure out what happened.

closed 2023-01-09 by SmittyHalibut