aly-fly/EleksTubeHAX #29: Novellife / WROVER

meddle99, 2022-12-18:
I have an elekstube clone clock made by Novellife.  It has an ESP-32 WROVER module.  I compiled the latest code and it loaded fine.  Watching the serial console, the clock connects to my network and receives the timebvoa ntp.  It sets and reads the R8025 real-time clock just fine.  The LEDs behind the displays light up.  

The issue is that nothing is displayed on the TFT displays. Any ideas on how to find out why they don't work?

frankcohen, 2022-12-18:
The original device uses a multiplexer to select the TFT CS lines. If you
have a logic analyzer you may be able to figure this out. It's probably
something fairly easy like that. Frank

meddle99, 2022-12-18:
This clock uses the 74hc595a.

SmittyHalibut, 2022-12-18:
The original did too.  Apparently my hardware hacking notes were removed from README.md at some point, but here's a link to the earlier version talking about the chip select:

https://github.com/SmittyHalibut/EleksTubeHAX/tree/3d95c2b42edc113b67633968b6d115bb3d738399#chip-select-shift-register

Compare your board to the Inputs section, see whether they moved the Shift Register to different GPIO pins.  You might have to update `GLOBAL_DEFINES.h`:

https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/EleksTubeHAX/GLOBAL_DEFINES.h#L87-L90

meddle99, 2022-12-19:
Thanks, ill take a look at that.

meddle99, 2022-12-21:
Ha!  it appears STcp (12) on the shift register is going to IO15 (23) on the WROVER and SHcp is going to IO13 (16).  Ds is still going to IO14 (13).

SmittyHalibut, 2022-12-21:
Have you tried building the firmware with that change?  Does it work?

meddle99, 2022-12-21:
No, still something else wrong.   I'll have to check the other pins.  They likely changed more.  

Another thing I noticed is if I change the board to WROVER in the IDE, it complains 
"Sketch uses 1087898 bytes (103%) of program storage space. Maximum is 1048576 bytes"  I changed the partition to 2/2 and the firmware loads, but I had to remove some of the higher numbered images to get it to fit in the 2mb SPIFFS.  

I also tried it with standard "ESP 32 DEV" module.  With that, the firmware uploads with the 1/3 partition scheme and the full graphics uploads into the SPIFFS.  

Should I have it set to WROVER, or should it work with "ESP 32 DEV"?

meddle99, 2022-12-22:
The rest of the pins are the same, except that min has a motion sensor (which I hate) instead of buttons.  

I was examining the TFT and removed the cable from the connector.  on the underside, instead of FP110H03A like the one you link to on AliExpress, the number is HD114001.  I don't know if that is significant.  the pinouts are exactly the same.  Any other idea of what i can check?

frankcohen, 2022-12-23:
I use ESP 32 DEV and no OTA. -Frank

---
Frank Cohen, Founder
***@***.***
http://www.votsh.com
Phone: (408) 364-5508 USA

meddle99, 2022-12-24:
Thinking the display might be different, I carefully pried the TFT from the board, then was rewarded with no information.  The back is totally blank.  I did check the connections between the 13 pin connector and the TFT, and they all match the elekstube.  I then enabled debug in the code, and everything looks like it is loading normally in the serial monitor.   The RTC responds, it connects to my WIFI (I commented out the WPS code and hard coded my wifi info).  The clock gets the time from NTP and the geolocation works.  Then it goes to constantly preloading images every second.  Is that what it looks like when it is operating?  

..................................................img transfer: 14
Preload img
image W, H: 123, 185
dimming: 255
img load : 40
55

meddle99, 2022-12-24:
Novellife moved the TFT enable pin to IO4.  Works now.  Except I destroyed one of the TFTs (see how I said carefully?).  Ordered a new one from Ali.

SmittyHalibut, 2022-12-24:
Excellent! Well, about figuring out the hardware, not about breaking a display. 

You want to modify the code so there's a single `#define` for each hardware type, that defines all the correct pin assignments? I take pull-requests. ;-)

meddle99, 2022-12-24:
Yes -- I'll learn to do that with github.  It will probably be a couple days.  Tomorrow might get a bit busy.

Thanks smitty and frank for this awesome work!  This is the clock I wanted now -- well, at least it will be when i get the display fixed.

closed 2022-12-26 by meddle99

frankcohen, 2022-12-26:
I bought replacement tfts on AliExpress

meddle99, 2022-12-26:
> I bought replacement tfts on AliExpress

Yep.  Already on order!  Thanks for the help