aly-fly/EleksTubeHAX #31: Back to original firmware

ducanh1603, 2022-12-30:
I’ve use winmerge but it doesn’t work. The notification show : ”1 different”. I need some help my clock BLACK OUT 😓. I would appreciate it if someone could help me. Sorry about my bad-english 😂

ducanh1603, 2022-12-30:
[image]

ducanh1603, 2022-12-31:
My MAC id: 78:21:84:6b:c3:98 
Someone please make a binary file flash for me pls. I tried 20-30 times =((((

aly-fly, 2023-01-10:
I was not able to get my clock back to working state by changing MAC address. Well it works on first power-up but does not work on a second power cycle. So not useful.
If you didn't save your original flash image, there is a very low chance to get it back into original state.
I suggest you load this EleksTubeHax firmware. it is better than original ;)

SmittyHalibut, 2023-01-10:
>  I suggest you load this EleksTubeHax firmware. it is better than original ;)

he suggests modestly.  ;-)

@ducanh1603 I'm sorry, but I don't think this is something we'll be able to help you with.  If you run into problems getting the EleksTubeHax firmware loaded, let us know.

Good luck.

closed 2023-01-10 by SmittyHalibut