aly-fly/EleksTubeHAX #32: Display Dimming only morning hours

icebreaker-ch, 2023-01-09:
Display dimming only in the morning hours, e.g. from 2 to 6 o clock does not work as expected.
I changed the code in the file EleksTubeHAX.ino according to the attached patch file (i had to zip it for uploading):

[EleksTubeHAX.ino.patch.zip](https://github.com/SmittyHalibut/EleksTubeHAX/files/10371672/EleksTubeHAX.ino.patch.zip)

SmittyHalibut, 2023-01-09:
https://github.com/SmittyHalibut/EleksTubeHAX/pull/33  Even better, I do love pull requests.  :-)

Thank you very much.  Some very minor comments on your PR, with suggestions that, if you accept, you should be able to just push a button, then I'll approve and merge.

icebreaker-ch, 2023-01-09:
Your changes are ok for me...

closed 2023-01-09 by icebreaker-ch