aly-fly/EleksTubeHAX #37: doc or src needed for BMP to CLK converter

wfdudley, 2023-02-25:
Getting a Windoze machine was a real pain in the neck.
I'd like to make a BMP to CLK converter that runs on Linux (or MacOS).
Could you document the conversion or provide source code?

aly-fly, 2023-02-25:
Yes, just give me a day or two to reach my machine.

SmittyHalibut, 2023-03-15:
w00t.

closed 2023-03-15 by SmittyHalibut

wfdudley, 2023-03-15:
just FYI, I forked the code and then added utilities that do all the
conversions:
CLK <-> PNG <-> BMP.  Code is a perl wrapper that calls ffmpeg to do the
real work.
This works in Linux should be portable to Winders and Mac.

Fork is here:

https://github.com/wfdudley/EleksTubeHAX-wfdudley

There are other improvements (to the actual clock code) besides the
graphics conversions.
Bill Dudley
This email is free of malware because I run Linux.