aly-fly/EleksTubeHAX #43: Feature request: Display arbitrary information on screens like a Turing Smart Screen

RedFox-Fr, 2023-03-15:
Hi,

I just discovered EleksTube and I would like to use it like a "Turing Smart Screen" to display arbitrary information, and as a clock of course ;o)

A " Turing Smart Screen" ( https://www.turzx.com/ ) is a dumb screen that only displays pictures send by serial connection and used to monitor ressources of a computer.
For example :  https://fr.aliexpress.com/item/1005005240742254.html
You can simulate a Turing Smart Screen with this project : https://github.com/mathoudebine/turing-smart-screen-python

Since EleksTube has WiFi connectivity, the simpler implementation could be to provide 1 to 6 URLs where to download the pictures to display, refresh every x seconds and delegate to an another project the generation of the pictures.

Is it feasible?

Best Regards,

aly-fly, 2023-03-15:
Hmmm.. this is way out of intended use of this clock. Huge modification would be needed plus external software on the PC to prepare data or images. You are free to make a Fork from this repo and build such functions yourself ;)
Your idea sounds simple, but it isn't easy to implement this, including all the setup/config and verify all possible failure modes.

SmittyHalibut, 2023-03-15:
I like the idea you're talking about, but like @aly-fly said, that's a pretty much entirely new firmware.  It's outside the scope of this project.

Also like @aly-fly said, you're welcome to branch this code, or just copy bits and pieces of it, to help you get started with your own firmware.  I just request that you give us credit for any code you use.  (A simple mention and link to this project in the README.md is sufficient.)

But it's not something we're going to build into this firmware.

closed 2023-03-15 by SmittyHalibut