aly-fly/EleksTubeHAX #45: How to control Clock with MQTT

kovu94, 2023-03-23:
Hey everyone,

I accidentially messed up the firmware of my EleksTube clock trying to update from official homepage. After two days of searching through the web I successfully managed to compile your firmware and the clock is working again (yeah!) :-)

I activated MQTT in the config and added my broker to get data to Home Assistant. 

The monitor shows some reports like IP, Signal Quality, Online status, ... and I have successfully added these to my HASS.
You mentioned in the readme that it is possible to control the clock with MQTT?
Can you help me with the commands to do this?
I'm not an expert to MQTT so I tried to change the status of myelekstube/report/powerState to OFF with MQTT-Explorer but the clock doesn't switch off. I think this should be an "cmnd"-command?

Additional information - yesterday after the first try of flashing I got the error that the images are not shown correctly like other users in issues. I was able to rotate the images but only 1/3 of the display was active, the rest was blurry.
Today I downgraded all plugins in ArduinoIDE to the recommended ones and now it's working fine.

aly-fly, 2023-03-23:
Glad the clock is working for you. I added MQTT for connection to Smartnest.cz, so the implemented commands are compatible with their broker / cloud. And from there linked to Samsung SmartThings :) 
As regarding to commands, it's easiest to dissect the code and extract them. 
Mqtt_client_ips.cpp

void callback(char* topic, byte* payload, unsigned int length) {  //A new message has been received
...
    //------------------Decide what to do depending on the topic and message---------------------------------
    if (strcmp(tokens[1], "directive") == 0 && strcmp(tokens[2], "powerState") == 0) {  // Turn On or OFF
        if (strcmp(message, "ON") == 0) {
...
        } else if (strcmp(message, "OFF") == 0) {
...
        }                                                       //      SmartNest:                         // SmartThings
    } else if (strcmp(tokens[1], "directive") == 0 && (strcmp(tokens[2], "setpoint") == 0) || (strcmp(tokens[2], "percentage") == 0)) 
...

in summary, it listens to:
directive powerState ON/OFF
--> turn on and off displays and backlight

directive setpoint 99
directive percentage 99
--> values 0 to 50 select one of the 5 graphics. Value > 90 issues a command that randomly selects one of the preloaded graphics.

aly-fly, 2023-03-23:
Great!

kovu94, 2023-03-23:
Just for documentation:
configuration.yaml for Home Assistant:

mqtt:
  switch:
    - name: "Elekstube IPS Schlafzimmer"
      state_topic: "elekstube_schlafzimmer/report/powerState"
      command_topic: "elekstube_schlafzimmer/directive/powerState"
      payload_on: "ON"
      payload_off: "OFF"
      retain: true
  sensor:
    - name: "ElekstubeIPS Online"
      state_topic: elekstube_schlafzimmer/report/online
    - name: "ElekstubeIPS Firmware"
      state_topic: elekstube_schlafzimmer/report/firmware
    - name: "ElekstubeIPS WIFI-Signal"
      state_topic: elekstube_schlafzimmer/report/signal
      unit_of_measurement: "db"
    - name: "ElekstubeIPS IP"
      state_topic: elekstube_schlafzimmer/report/ip

Gives you:
[image]

closed 2023-03-23 by kovu94

SmittyHalibut, 2023-03-23:
The buried lede here is that you figured out a "solution" (read: workaround) to the image rotation/partial display problem.  I've created #46 to dig into that more.  Though, I don't expect to get to it any time soon.  But at least this advancement is documented now.