aly-fly/EleksTubeHAX #46: Support recent depedant libraries

SmittyHalibut, 2023-03-23:
From https://github.com/SmittyHalibut/EleksTubeHAX/issues/45

User reported image issues that have been reported elsewhere: Only 1/3 of the screen is loaded, image rotated, etc.  There was a lot of speculation about the cause, but no one fixed anything.

In #45 it was mentioned that they "fixed the problem" by downgrading all the libraries to the recommended versions.

That's not a great fix, but at least we know how to work around the problem.  More importantly, it gives a lead on where the problem is.

## TODO:
* Figure out which library it is that causes the problem by downgrading them one at a time.
* Ideally, figure out which version broke it.
* Figure out what changed in that library at that version, and figure out why its breaking.
* Fix it.  :-)

bitrot-alpha, 2023-05-09:
This project doesn't compile on the latest "legacy" version of the Arduino IDE, 1.8.19 (the last version of the 1.x available)

Also, this project doesn't compile on a modern version of the ESP32 support library (v2.0+).
Context: I'm trying to use SSID and password instead of WPS.

I could trace one issue to line 152 of `WiFI_WPS.cpp`, the `WiFi.onEvent()` function call.
Arduino compiler complains that there's no function with a signature matching the argument given.

Also, even with all of the libraries and IDE installed with versions given in the README.md, the project doesn't compile with `#define USE_CLK_FILES` removed..

Lots of code to fix/bring up to date it seems.

aly-fly, 2023-05-09:
> Lots of code to fix/bring up to date it seems.

Please try with my fork, it was updated quite a lot, but I was unable to merge with original repo for some strange reasons and sync issues.
https://github.com/aly-fly/EleksTubeHAX

if you still have issues, I can send you libraries that it compiles fine with.

bitrot-alpha, 2023-05-25:
I've opened a PR (#50) with the minimum amount of updates and fixes to use Arduino IDE 1.8.19 and the latest versions of all the dependant libraries. Please take a look at it.

I've only tested with the Elekstube version of the hardware, as that is what I have.

aly-fly, 2024-02-03:
Merged into this repo and manually merged into my fork for testing.

closed 2024-02-03 by aly-fly