aly-fly/EleksTubeHAX #48: Clock face freezing randomly

bitrot-alpha, 2023-05-18:
Sometimes the clock face will freeze for a few seconds. This goes away for a while if I power cycle the clock.

I'm using the Arduino IDE and library versions from the readme.

I have MQTT and Geolocation enabled and set up with my own credentials. I disabled WPS and hardcoded my wifi settings. The clock hardware is the EleksTube board. I checked the CH340 chip and all of the power regulators with a multimeter, everything's running at 3.3V.

aly-fly, 2023-05-18:
Operations, that take longer are:
- Geolocation (to get updated timezone + DST). This runs every day at 3 at night only. 
- Time server sync. This is every couple of hours. Can be longer with bad network conditions. 
- This is every minute or so. Can be longer with bad network conditions.

bitrot-alpha, 2023-05-18:
Thanks for quick reply!

I disabled MQTT and the clock doesn't seem to freeze any more. The clock seems to struggle with my wifi connection somewhat (even though my smartphone gets -50 dBm RSSI right next to the clock), sometimes when I power cycle the clock it doesn't successfully connect.

I'll try re-enabling MQTT and setting the publish time to every 5 minutes.

closed 2024-01-18 by aly-fly