aly-fly/EleksTubeHAX #52: What hardware version is it?

DinoFontana, 2023-08-11:
Hi everyone,
I bought this Elekstube clock on pcbway
https://www.pcbway.com/project/gifts_detail/RGB_Glow_Tube_Clock_907ad35c.html
I tried to compile the source of this repositories, following all the steps as indicated.
It compiles without errors, the firmware is loaded but nothing appears on the clock (displays off). I tried choosing as HW
HARDWARE_Elekstube_CLOCK and also HARDWARE_NovelLife_SE_CLOCK
(I would discard HARDWARE_SI_HAI_CLOCK because I saw from the code that this uses DS1302 and in my case instead I have a DS3231)
Do you have any ideas or advice?
Thank you

aly-fly, 2023-08-11:
Hi, obviously hour hardware is different. Disassemble it and back-trace the wiring. Compare with shematics on this repository. After you get a new pinout for your clock, we can help you modify the code.

DinoFontana, 2023-08-15:
Thanks for the answer.
I checked the links and they match the Hardware of "HARDWARE_Elekstube_CLOCK" I chose in USER_DEFINES.h.
I am attaching a link to an image for better understanding.
What can I do or check?
Thanks

[image]

aly-fly, 2024-02-03:
Documentation updated and "HARDWARE_PunkCyber_CLOCK" option added in my fork. Will be uploaded here a bit later.

closed 2024-02-03 by aly-fly