aly-fly/EleksTubeHAX #54: How to check out pinouts?

rsj001, 2023-10-16:
Hello! I've got a new IPS Clock which seems like a fake one
and with chip ESP32-S2-WROOM
I tried to upload firmware pre-built but IPS won't display and repeatedly reboots
I'm a beginner, how to check pinouts?
Also, I tried to upload a code to interact only with buttons, but no response
Is there any advice? Thank you!

aly-fly, 2023-10-16:
Hi, use Termite (https://www.compuphase.com/software_termite.htm) or similar terminal to capture output of the clock when running. 
To check pinouts, you have to disassemble the clock to get to the PCB and trace each line from ESP32 to the destination.
Also try to compile the clock with different pinout variants, maybe one will work.
Important: make sure you have libraries of correct version installed. Newer libraries can cause problems.
Make sure you use Arduino IDE version 1, not 2.

rsj001, 2023-10-16:
The code in this repo seems incompatiable with ESP32-S2
I used the code in the PR
however it still crashed
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002d072
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x298c
entry 0x4004c18c
E (212) esp_core_dump_flash: Incorrect
IPS clock v0.4
In setup().
Config size: 88
Loaded Backlights config is invalid, using default.  This is normal on first boot.
backlights OK.
PIN SETTING.
33
PIN SET SUCCESSFULLY.
PIN SETTING.
32
ESP-ROM:esp32s2-rc4-20191025
```
I added some comment to the Serial, it crashed when setting pin 32

And I wonder if this is a proper way to test buttons?
```cpp
const int s[]={33};

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  for(int i=0;i<1;i++)
    pinMode(s[i], INPUT_PULLUP);
}

void loop() {
  // put your main code here, to run repeatedly:
  for(int i=0;i<1;i++)
    if(digitalRead(s[i]) == LOW) {
      Serial.print(s[i]);
      Serial.println("activated");
    }
  delay(200);
}
```
However the digit won't change when i set s[0] to 33 34 or 35 and crashed at 32

rsj001, 2023-10-22:
Hello I've tried to analyze it and TFTs are working now
but RTC, backlights and 74HC595D are not
(The 6 displays are showing the same digit, and RTC time is 0)
I thought I've put everything in place.
Can you do me a favor?

My DEFINES:
```cpp
  // WS2812 (or compatible) LEDs on the back of the display modules.
  #define BACKLIGHTS_PIN (-1)

  // Buttons, active low, externally pulled up (with actual resistors!)
  #define BUTTON_LEFT_PIN (7) //correct
  #define BUTTON_MODE_PIN (6) //correct
  #define BUTTON_RIGHT_PIN (5) //correct
  #define BUTTON_POWER_PIN (4) //correct

  // I2C to DS3231 RTC.
  #define RTC_SCL_PIN (33)
  #define RTC_SDA_PIN (34)

  // Chip Select shift register, to select the display
  #define CSSR_DATA_PIN (13)
  #define CSSR_CLOCK_PIN (16)
  #define CSSR_LATCH_PIN (15)

  // SPI to displays
  // DEFINED IN User_Setup.h
  // Look for: TFT_MOSI, TFT_SCLK, TFT_CS, TFT_DC, and TFT_RST

  // Power for all TFT displays are grounded through a MOSFET so they can all be turned off.
  // Active HIGH.
  #define TFT_ENABLE_PIN (21) //correct
```
[image]
[image]
[image]

sthaugen, 2023-10-22:
My ELEKSTUBE has no connection on pin 10 74HC595,  and GND to pin 13 74HC595.
So yours IO14 must hardcode this pin LOW.

rsj001, 2023-10-22:
thanks! 74hc595 works when I set io17 to high and io14 to low

rsj001, 2023-10-22:
Do you have a particular pin for LEDs on back of the displays? There isn't one for me

sthaugen, 2023-10-22:
[image]

rsj001, 2023-10-23:
my ips module is identical to yours
but i don't have a RGB pin on esp32-s2, I've tried all pins brutely but none of them is connected to backlights

closed 2023-10-28 by rsj001

Martinius79, 2025-05-28:
Hello @yt2008,

 I am not sure, which clock you are talking about. 

Can you make an overview picture? and/or details of the board?

There is technically not a big difference between the ESP32 and the ESP32-S2 for the code we are running. It is normally just to find out the used components and the pin-out.

Bye Martinius

yt2008, 2025-05-29:
> Thx for the pics!
> 
> All the info so far I can see or assume are put into this branch now: https://github.com/Martinius79/EleksTubeHAX/tree/TryOutsXunfengClock
> 
> Problem for me is, I don't own the clock. So you have to tryout the TFT pinouts on your own. Please compile and upload it...
> 
> Hope, you already have a backup of your orginal firmware?
> 
> I wasn't able to find a fork with modifications for this clock already, so I start from the scratch... Just as a tryout for now...
> 
> Where did you get it from BTW?
> 
> And what do you mean with "it suddenly bricked"?

Hi @Martinius79 
Thanks for the reply and made these commits.

So the previous "Suddenly bricked", I mean I haven't used it for a while, and one day I connected power but it suddenly screens are display nothing but are on, LEDs are on too, like this: 

[image]

So this was a gift my ex-housemate gave me, I asked him where he bought, he mentioned from Taobao but original store is no longer exist and the seller is not contactable now. So I am not sure if this S2 version is still on the market or not. I also had done a lot of research, but hardly find much information about this `Xunfeng IPS Clock with S2 board`. I had also tried flashing a few different firmwares and none of them works.

This is the original firmware I backed up, but it does not work and in the bricked status as shown in the above picture.

I just cloned the repo and rename the file `_USER_DEFINES - empty.h` to `_USER_DEFINES.h`, build and uploaded. 

Guess what?! My screens are ON~

[image]

But the screen seems frozen here? After having WPS connection to my router
I am a newbie to ESP32 and also first time try this firmware, not sure what to do from this, i tried click different buttons and also the ip, but the site is not reachable, backlights are not on and I have no idea what to do next

Martinius79, 2025-05-30:
This looks promising. 

The clock outputs all debug information via the serial port, so using "Upload and Monitor" is an effective way to capture complete logs from the start.

I've discovered that the ESP32-S2 uses different default pins for the I2C bus - GPIO 8 and 9 by default (as visible in the logs) - but we need to use GPIO 33 and 34 instead. The code wasn't initializing the TwoWire I2C bus module with the defined values for the RTC pins. On the Elekstube, the RTC pins happen to match the ESP32's default pins, which is why it worked there. I've now updated the code to properly initialize the correct pins.

I suspect the RMT GPIO error in your logs might be related to this issue. GPIO 9 should only be used by the TFT library (as evidenced by the display working). After the I2C initialization with wrong pins, the error occurs because GPIO 9 is being attempted to be used twice. Could you test the updated code and share the logs here?

Regarding the backlight LEDs, I'm making an educated guess that GPIO 26 controls them, though this is unverified. On other similar clocks, there's typically a single communication line between sockets for the LEDs. The marked pins are likely for the backlight LEDs, but with the chip soldered over the relevant traces, it's impossible to see the connections clearly. Other potential candidates are GPIO 40, 41, 42, 45, and 46. We may need additional debug output to properly identify the correct GPIO for the LEDs.

[image]

[image]

Martinius79, 2025-05-30:
AH, that's a success so far IMHO ;)
So clock is showing the time and the menu seems to be working as well (as I see from the logs).
So the buttons are also correctly defined.

I updated the defines a bit (reverse order for this clock as well, as for the PunkCyber and knowing/correct pin configs).

for the backlight:
Yes, the logical assumption would be GPIO 38, because that is the in the straight line! Maybe you can give it a try already.
I don't know where I read it, but I thought GPIO38 (or real pin 32) are reserved for something, but I checked again and it is fully useable...So let's give it a try...I guess I have to many MCUs around me in the last weeks...

And yes, if it is still not working, I will need to add some more debug output for the backlight stuff...So please try with the rest of the GPIOs (35-38 and the rest of the not used ones yet) and tell me what happend, then I will start or not ;)

yt2008, 2025-05-30:
Thanks a lot @Martinius79 
You saved my clock!

`GPIO_NUM_38` is CORRECT~

[image]

PS. From reading the logs, I'm aware of one thing from the bricked firmware: the problem may be that it lost the images or is pointing to the wrong image directory, so the screens are on but display nothing. But anyway, this firmware gave my clock a second life!

Appreciate your help over the last a few days

Martinius79, 2025-05-31:
That are good news, so a fully working clock, just 3 years after the initally question about a clock with the S2 :) And two years after someone asked for LED stuff...

> You saved my clock!

That makes me really happy to hear! That's what we are here for ;)

If you ever want to get rid of it, remeber me, I am interested in owning all the clones :) SI HAI and this Xunfeng one is still missing, as well as the "first" editions of the PunkCyber ones, from the PCBWay store.

> Appreciate your help over the last a few days

You are really welcome!

I have another clock with a S2 chip and C3 chipset here (Marvel tube and Marvel Mini) and will try to add them as well...One day...
Problem is, that it needs different board definitions and so more environments, which we want to avoid. So after I found a way to do this in a nice way, I will also integrate the Xunfeng branch permanently. For now, it stays as a branch in my fork...

You can do me a favor and make some nice pictures of the clock, in the way, the other clocks have in the documentation folder. Especially one which can be used for representing the clock in the README.md file (straight top view).

yt2008, 2025-06-01:
Sure @Martinius79 
I took some pictures for it, but I wanted to push them to a branch and create a PR for you. However, I don't have permission. I attach them here:

[image]
[image]
[image]

Do you think these are OK? I am not good at photography, but I tried my best

Martinius79, 2025-06-01:
Thx a lot for the photos!
All fine ;)

You can fork my repo/the branch, clone it locally, commit and sync your changes, then create a PR for the origin branch.

BTW: If your clock shows the "RTC reports power was lost" message after each power on/off cycle in the logs, the battery cell might be empty. Not a big deal since the clock uses NTP time when connected to WiFi, but you may want to replace it eventually.

And just to satisfy my curiosity: The attached files are from the ZIP you uploaded — these are the extracted SPIFFS partitions. The files look intact, but the 'config' file is 0 bytes, which looks strange. I have no idea what's wrong with the clock… maybe the battery? If you're feeling adventurous, try replacing the battery and flashing the original firmware.

yt2008, 2025-06-04:
> maybe the battery? If you're feeling adventurous, try replacing the battery and flashing the original firmware.

Hi @Martinius79, you're right; the battery is dead. When I disconnect the internet and power cycle it, it always starts UTC 0:00. 
I just replaced the battery today and flashed the original firmware, but it still displays nothing, so it may be other reasons that cause the brick.

Martinius79, 2025-08-24:
Hi @AlexBergGit ,

I am working on the integration of the xunfeng clock. 
For the moment, automatic firmware creation via a github workflow is missing and no  correct MQTT naming and the readme is not changed... 

But you can clone and build the feature branch: https://github.com/Martinius79/EleksTubeHAX/tree/SeperateBuildEnvs

Then flash firmware and data partition. 

I will finish the integration coming week I guess. 

Bye 
Martinus

AlexBergGit, 2025-08-25:
Thank you @Martinius79, I did as you said and now the clock works again. Odd that the original firmware bricked itself.

Martinius79, 2025-10-08:
The Xunfeng clock support is now integrated in the main branch.