aly-fly/EleksTubeHAX #55: Hardware version Gen2

DarthWeber, 2023-11-10:
Hi, is this working with Gen2 also or do i need to change something? Would like to know before i buy it....

Thanks,
Tom

DarthWeber, 2023-11-13:
// Chip Select shift register, to select the display
#define CSSR_DATA_PIN (14)      // 14
#define CSSR_CLOCK_PIN (9)      // 9
#define CSSR_LATCH_PIN (10)     // 10

works with Gen2.1 (ESP32 Pico D4 Chip)

closed 2023-11-13 by DarthWeber

aly-fly, 2024-02-12:
Added support for this clock in the _pio folder.

judge2005, 2024-02-13:
> // Chip Select shift register, to select the display #define CSSR_DATA_PIN (14) // 14 #define CSSR_CLOCK_PIN (9) // 9 #define CSSR_LATCH_PIN (10) // 10
> 
> works with Gen2.1 (ESP32 Pico D4 Chip)

Do you have a backup of the original firmware on the chip? I would like to try to get some images from it. The updates from Eleksmaker are not the full image, they only include the code portion which does not have images, I assume they stashed them in spiffs or littlefs or whatever.

Martinius79, 2024-10-03:
@judge2005 I backed up the orginal firmware and I extracted the pictures from it and they are all already present in the `data - other graphics` folder. 
So I guess nothing to do here anymore, just to finish this topic.

As far as I understand, the original FW image can be set to "your" MAC address with a little manipulation of the bin file.

Bye
Martinius