aly-fly/EleksTubeHAX #56: Numbers are on the side

tomac01, 2023-11-14:
Hi all,

I adapted the USER_DEFINES.h regarding WIFI access and compiled the code.

After the upload, the loading texts and numbers appear on the TFTs lying on their side.

Does anyone know what I'm doing wrong or what I forgot during configuration?

cu

aly-fly, 2023-11-14:
1. Make sure you have correct version of the libraries installed. Too new version may cause this.
2. Display library has an Orientation property. Change it from 0 to 3 (all possible rotations).

tomac01, 2023-11-14:
My mistake:
I hadn't commented out the first entry in User_Setup_Select.h, so User_Setup.h was still included.

After I corrected that, it works great!

Thank you!

closed 2023-11-14 by tomac01