aly-fly/EleksTubeHAX #57: wps started ..... cicle

igitlin, 2023-12-17:
My clock is a SI HAI - i try to upload new digits and clock run on boot loop
i find you repository and try to firmware - BUILD_SI_HAI_20220224.bin but after restart i see 
setup...
WIFI start
WPS Started
Press WPS button ou you router

aly-fly, 2023-12-22:
Correct. Clock is waiting to be connected to your WiFi network to get accurate time from the internet.
Press WPS button on your WiFi router to connect the clock.

igitlin, 2023-12-22:
Yes i try to press WPS on router, but nothing

aly-fly, 2023-12-22:
This works differently.  No AP.
You can add your WiFi SSID and Password in WiFi.Begin(); function. In the brackets.

igitlin, 2023-12-22:
I’m not sure that I know how to do this, I’ve only just mastered how to flash ready-made firmware, but unfortunately I couldn’t find the one that was included when I bought the watch anywhere. There was a built-in wi-fi AP and other watch faces.
My son asked me to download a new watch face, I downloaded it via Esp32ClockSpiffsTool.exe, but the watch froze.
Now I’m trying to somehow resurrect them.

What program should I download new watch faces for this?
firmware version?

aly-fly, 2024-01-18:
I hope you have first made a backup - downloaded the Flash image of your original clock before trying to upload this custom firmware?
Best is to put back the original firmware and use manufacturer's software to load new clock images.

igitlin, 2024-01-20:
The fact of the matter is that no, I didn’t think that a simple change of skins could kill the firmware. But uploading the skin killed the firmware and that’s why I’m looking for how to restore it.

igitlin, 2024-01-20:
Doesn't anyone really have the same firmware with which watches are sold on Aliexpress?
Or is there some kind of secret firmware there?
I remember well that the previously installed firmware worked without problems without Wi-fi, it simply showed the name of the Wi-Fi access point and password 12345678 when loading - I could connect from my phone and set up a connection to my router.
The firmware that I found here and tried to flash does not want to work without connecting to Wi-Fi, and my router (it works in Wi-Fi Access Point mode) does not have a WPS button.
I can’t switch it to another mode, since there are many different smart home devices connected to it.

aly-fly, 2024-01-21:
The firmware on this repository is a completely custom build and we don't support any modes or functions that were present on the clock as delivered from the factory. Development was done on original IPS clock from EleksMaker and it is only tested on this clock hardware.

Most probably your problem is as described [HERE](https://elekstube.com/en-de/blogs/news/about-the-clock-black-screen-after-using-ips-tools-program). You used incorrect utility / software for your clock. 
To restore original functionality please contact you seller, if they can provide flash image for your clock model.

Alternatively, I can build a custom flash image for you, with your WiFi SSID & Password and selected clock faces. You then only upload it to your clock. I prefer BTC, ETH or AVAX for payment.

As another option, you can get a cheap router with WPS function, program it to same SSID & Pass and use that as a temporary router to connect the clock. Then you disconnect temporary router and clock will connect to your main router.

aly-fly, 2024-02-03:
No further responses from the igitlin. Probably resolved.
Access point or config via serial terminal may be added in later revisions.

closed 2024-02-03 by aly-fly