aly-fly/EleksTubeHAX #58: Use VSCode and platformio to simplify whole build

hallard, 2024-01-15:
Thanks @SmittyHalibut for this amazing  job.

I'm planning to port this one using visual studio code and platformio so end user can build easily getting rid of checking all library version in arduino IDE and changing #define in source code, just doing change in `platformio.ini` file

If you agree and plan to merge this new pull request I will do from here, else I will fork and make available on my github, but I prefer contribute to yours.

Let me know.

Charles

aly-fly, 2024-01-15:
Hi Charles, I have made quite a few updates to the code on my fork (https://github.com/aly-fly/EleksTubeHAX) but due to some conflicts I was not able to merge it here... In the mean time also this repo received a few updates, so it has become quite a task to merge both repositories together, before porting to Platformio :)
Unfortunately time is quite limited, that's why I left this on the backlog... 
Otherwise I have good experience with Platformio and would like to see the ported code sooner or later.

aly-fly, 2024-01-15:
Also, this guy already made a port to Platformio: 
https://github.com/SmittyHalibut/EleksTubeHAX/compare/main...judge2005:EleksTubeHAX:main

SmittyHalibut, 2024-01-15:
And if I'm being honest, this is abandonware for me.  I haven't put any work into this for (crikey!) years now.  @aly-fly is basically the owner now, despite it being in my account.

Should we officially move the repo to you now?

aly-fly, 2024-01-15:
> Should we officially move the repo to you now?

It was officially your project (+ Frank's if I remember correctly?) so lots of kudos for the first developers. 
If you are in for the move, I will be glad do accept and maintain it :)

SmittyHalibut, 2024-01-15:
@Frank-Bemelman You have an opinion here?  @aly-fly is right, you put in as much effort as anyone else here.

For me personally, it was a 4-week project for me to do while I had some time off work, that I no longer have time to maintain.  I'm happy to hand primary ownership off to someone else for on-going maintenance, and to my observation, @aly-fly has been most active in that.

hallard, 2024-01-16:
Woo I was not expecting so much animation here, nice :-)

Another solution is that @SmittyHalibut give permission on this repo so that defined contributors (@Frank-Bemelman , @aly-fly) can merge or publish (same or new branch) here like that, owner stays owner but other can works and do modifications seen directly from here, and other are not confused with too much forks, may be a good deal.

whatever I can do that but can't set a timeline because looks like everyone there 24H/day is not enough :-)

SmittyHalibut, 2024-01-16:
Done.  @aly-fly you're now a contributor.  I don't object to transferring ownership to you if you'd prefer.  Let me know.

Also, I knew I forgot someone.  @RedNax67 was the other person I was thinking of who contributed a lot early in the process.  Sorry for forgetting you!

SmittyHalibut, 2024-01-16:
To address the ACTUAL ask of this Issue!  :-)

I have personally moved to PlatformIO and VScode for all my development as well, so I certainly don't object to migrating this project to that platform.  My only concern is the additional support burden of helping new folks program their hardware.

Though, from the looks of the other tickets we've received, I don't know if this hardware platform is even still available. As it is, there won't really be any new users anyway.  At least, not without a fair bit of work to the code to abstract out the different hardware versions and make it easy to add new ones.

In any case.  As far as I'm concerned, please do migrate it to PlatformIO.  Arduino IDE (at least the v1.8 branch) is less than ideal.  (I haven't used v2 yet, so have no opinions on it.)

frankcohen, 2024-01-17:
Hi Smitty, +1 on anything you want to do with the project. AND, right on time, the CEO at EleksMaker today replied to my post on Reddit, inviting a response to his email. Anything you or anyone wants me to reply to them? -Frank

aly-fly, 2024-01-18:
> Done. @aly-fly you're now a contributor. I don't object to transferring ownership to you if you'd prefer. Let me know.
I received and confirmed the status. Thank you. For now, I think the status is perfectly fine as is, because this repo is linked on many internet sites.  Will take time in following few weeks to merge and test all changes from my branch into this one.

aly-fly, 2024-01-18:
> Arduino IDE (at least the v1.8 branch) is less than ideal. (I haven't used v2 yet, so have no opinions on it.)

I tried Arduino v2. Many projects don't compile. Also the Data Upload tool is not available. It's almost impossible to migrate from v1.8 to v2. PlatformIO will be the way to go.  @hallard Did you check out that fork that was already migrated? Is it useful?

aly-fly, 2024-01-18:
> AND, right on time, the CEO at EleksMaker today replied to my post on Reddit, inviting a response to his email.

Found it: https://www.reddit.com/r/arduino/comments/mq5td9/comment/ki7r3u2/?utm_source=reddit&utm_medium=web2x&context=3

Well, I am not sure if the author is happy that their design is being hacked and also copied by Chinese guys. And then we support all sorts of hardware copies that reduce author's profit. :)
At the moment I don't have any questions for the EleksMaker...

hallard, 2024-01-18:
@aly-fly before I found the vscode version I tried with this specific fork version https://github.com/bitrot-alpha/EleksTubeHAX/ may be time of my search I tough it was the most "up to date" and I succeeded to fix my clock (but as library management is a mess (don't blame any authors, it's Arduino way). 
Anyway only after I took the idea to migrate this one to vscode and then found other contributors already done, so no I did not try any with vscode yet. But for sure I will as soon as it is in this repo or let me know the fork to use, I can give a try.

I also plan to re org the digits, having all in a folder with both (`bpm`, `jpg` or `png`) AND `.clk` so it's easier to see in explorer or finder and have a script to select the one to put in data folder easily without having to rename file name because we all prefer different digits from each others.

aly-fly, 2024-01-18:
I like the idea of having all image files in the same folder. Let us know when you will have a working script :)

hallard, 2024-01-18:
chatgpt done the hard work (here basic example), I even used it to crop original images also :-)

```python

from PIL import Image

# Loop through the image files named RETRO0.jpg to RETRO9.jpg
for i in range(10):
    # Open each jpg image file
    img = Image.open(f"RETRO{i}.jpg")

    # Crop the image to the specified coordinates (left, upper, right, lower)
    #cropped_img = img.crop((12, 22, 120, 220))

    # Save the cropped image as a bmp file named 40.bmp to 49.bmp
    #cropped_img.save(f"4{i}.bmp")
    img.save(f"4{i}.bmp")

```

hallard, 2024-01-18:
I put here as a reminder, 

During lunch time I asked chatgpt to convert pascal code to generate `.clk` file (because I'm on mac and file converter need windows platform) and here the result; I'll take some time later to see if it's working, but looks good.

```python
from PIL import Image
import os

def convert_image(input_file_path, output_file_name):
    # Load the image
    img = Image.open(input_file_path)
    W, H = img.size

    # Construct the output file path
    output_file_path = os.path.join(os.path.dirname(input_file_path), output_file_name + '.clk')

    # Prepare the header
    header = bytearray([ord('C'), ord('K'), W & 0xFF, (W >> 8) & 0xFF, H & 0xFF, (H >> 8) & 0xFF])

    # Create a memory stream (byte array) and write the header
    memory_stream = bytearray()
    memory_stream.extend(header)

    # Process and write pixel data
    for y in range(H):
        for x in range(W):
            R, G, B = img.getpixel((x, y))
            OutPixel = ((R & 0xF8) << 8) | ((G & 0xFC) << 3) | (B >> 3)
            memory_stream.extend(OutPixel.to_bytes(2, byteorder='little'))

    # Save to file
    with open(output_file_path, 'wb') as file:
        file.write(memory_stream)

    print('Saved.')

# Example usage
# convert_image('path_to_input_image.jpg', 'output_name')
```

frankcohen, 2024-01-18:
Ok, I'll just chill. Good luck on the project. I got involved to learn how
to hack hardware - I'm a software guy who now does both! 🙂 I even went to
Hackaday Supercon last year. I made my own board for my Reflections
project. https://github.com/frankcohen/ReflectionsOS

aly-fly, 2024-02-04:
Alight, I took whole weekend and merged together more or less everything useful from many different forks out there :D Also tested compilation of different options. And functionality on my hardware.
There are two main tasks still open:  merging manual timezones from RedNax67 and analyzing all sorts of updates from wfdudley. 

Looking at Platformio migration made by judge2005 - this doesn't seem to be very difficult. We might just merge ideas from there :) 
@hallard If you have experience doing this, you can proceed using existing code here, create a pull request and then we will continue development using Platformio.

p.s. I made a backup of this repository before this mass update, if anyone needs it, it's on my disk.

frankcohen, 2024-02-04:
Thanks for the update. And good job on merging it all together.

I don’t remember if we communicated before. If not let me introduce you to my contributions. Perhaps you would be interested in incorporating them into the new super project.

I added a Web UX to control the EleksTube IPS clock. The code is here:
https://github.com/frankcohen/EleksTubeIPSHack

And, I added a hack that lets you split an image in a browser page and display them on the IPS. I call it Photobooth. A video showing it is here:

https://vimeo.com/568616803/453c2f24b7

Hacking the IPS helped me to build the Reflections project. It is a wrist watch size ESP32-S3 environment for making entertaining experiences. Reflections is at:
https://github.com/frankcohen/ReflectionsOS

I am no longer working on IPS code. And I’d be glad to support your effort.

-Frank

SmittyHalibut, 2024-02-05:
@aly-fly said:
> p.s. I made a backup of this repository before this mass update, if anyone needs it, it's on my disk.

And, ya know, git revision history. ;-)

aly-fly, 2024-02-06:
> And, ya know, git revision history. ;-)

I know, but better safe than sorry. I sleep better this way knowing I can manually revert anything If I made a big mistake somewhere.

aly-fly, 2024-02-06:
>  I added a Web UX to control the EleksTube IPS clock. The code is here: https://github.com/frankcohen/EleksTubeIPSHack 

Thanks for this, will be useful to look at. That is in the plan to add, if we will have enough space. Goal at the moment is to pack many features into 1 MB and keep 3 MB free for as many clock faces as possible loaded at the same time.

Good job on your wrist watch, wend through the project, looks impressive!

hallard, 2024-02-06:
Nice job here, really sorry guys, I'm currently out of time with huge project to get some spare time right now. But will test ASAP.

aly-fly, 2024-02-11:
Hi everyone, I am getting back to the original topic: Migration to PlatformIO.
I am playing with some of my other small projects (CO2 monitor + clock) to gain experience in PlatformIO.

Steps:
1. updated all libraries in Arduino 1.8 IDE to latest version, including ESP core.
2. migrated the project with as little modifications as required.

Results:
Compilation and running success. 
**BUT - code size is significantly larger! About 30%, what I really don't like.** 

```
Sketch uses 1040501 bytes (99%) of program storage space. Maximum is 1048576 bytes.
Global variables use 46412 bytes (14%) of dynamic memory, leaving 281268 bytes for local variables. Maximum is 327680 bytes.

RAM:   [==        ]  16.0% (used 52424 bytes from 327680 bytes)
Flash: [======    ]  63.2% (used 1325369 bytes from 2097152 bytes)
```
Arduino: 1040501 bytes.
PlatformIO: 1325369 bytes.
_(Ignore the % values as I had to adjust partition sizes.)_

I am learning that default compiler settings may have -O0 settings for [optimization](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html). Does anyone has insights what are default -O settings for both Arduino and PlatformIO?

aly-fly, 2024-02-12:
Alright, I successfully migrated the project to PlatformIO. 
Using manual file copying instead of "import arduino" function. Kept libraries in the same folder (not moved into .\lib). 
Build code size is about the same as before. Yay! 
Set up manual partitions as under Arduino (No OTA, 1M app, 3M SPIFFS). 

**Please download, test and report back!**

Issue that I noticed is configuring the pins for the TFT display. You can either manually overwrite the config file in the library (as before) or define pins in `platformio.ini`. Currently project is set-up using the latter approach. Still investigating some other approaches.

hallard, 2024-02-12:
@aly-fly awesome thanks.

For PIO, a idea is do define the differents boards into `platformio.ini` and then set pinout for each board, like that you have all paremeters in one file and can be different for each board

Here a quick and dirty example I made some time ago for LVGL boards
https://github.com/hallard/Makerfabs-LvglWidgetsDemo/blob/main/platformio.ini

aly-fly, 2024-02-12:
That is one way, yes, with different build environments.
I was looking into a solution to make a pre-build .py script that[ before every compilation](https://docs.platformio.org/en/latest/scripting/actions.html) rewrites the TFT config file `User_Setup_Select.h` to include the `global_defines.h` into the library and this takes away the probability of the user to incorrectly select the environment on the PIO startup.

judge2005, 2024-02-12:
Whichever approach is taken, I would prefer that it not modify any GIT files as this makes it hard to fork the repo and issue pull requests.

I took the platform.ini approach in https://github.com/judge2005/EleksTubeIPS as I'm building releases for others to use as well as myself. I have already accidentally flashed the wrong image!

aly-fly, 2024-02-12:
I made a .py script that copies config files into the TFT library folder. 
Could be a bit more polished (using system variables for file location) but I spent too much time failing to get it working, and finally gave up and reverted to hard coded paths. If anyone is more fluent in this, please correct it.
This way all settings are together in a sinlge file `_user_defines.h` and no need to switch environments.

aly-fly, 2024-02-17:
Hi all,
Did anyone had time to test this PIO project?
Are build paths for copying files same on Linux as on Windows?
Is everything working fine and can we delete the Arduino folder?

SmittyHalibut, 2024-02-19:
Sorry, I haven't had a chance to try it yet.  Lots going on at work right now.

Martinius79, 2024-03-05:
Hello!

I was able to build and upload the firmware with the PlatformIO extension/IDE in VS Code using the files from the "EleksTubeHAX_pio" folder in the main branch. 
Just needed to make the "normal" changes to the code (select model, rename user-defined header file, set up WiFi). I didn't have any "prerequisites" installed before, just VS Code. 

VS Code detected the need for PlatformIO and C/C++ extensions and installed them. 
PlatformIO then checked and downloaded the needed libraries. 

Just two minor problems occurred: 
The first line "Import (env)" from the little Python helper script needed to be commented out, and I needed to install Python (executable missing). 

And because I have a variant of the NovolLife clock (SPECTRUM-V2.0), I need to reduce the serial port speed to 512000 baud, but this has nothing to do with the PlatformIO migration. Maybe I can commit some pictures of it over the weekend. 

The firmware itself is working fine!

My setup: 
Windows 11 Pro 23H2 + VS Code 1.87.0 + fresh GIT clone from repo. 
Actually, Arduino 1.8.19 is also installed, so I didn't do a completely "clean build" yet on a vanilla Windows in a fresh VM, but don't think that is needed.

Great work so far @aly-fly  :)

I guess the Markup files with the instructions need to be created/renewed, and a bit of "fine-tuning" is possible, like using different build environments (to preselect the settings for the different variants and versions of the clock), as @hallard mentioned in his post and showed in his platformio.ini.

I don't know if I will find the time to do any of these tasks in the near future, but I will try...

Some questions from my side are left: 
What about the SPIFFS upload? I didn't check how to do it with PlatformIO yet. 
Should be possible to use the esp32fs jar call variant, I guess?
Which versions of the precompiled firmware should be included in the repo in the future? the Arduino build ones or the paltformIO ones? Or both? Any naming convention for them?

Bye
Martinius

aly-fly, 2024-03-06:
Hi Martinius, thanks for the detailed report! Glad it works fine for you.
Not sure if build environments are needed, as user must set other things too, besides the hardware type.
Will add a note in the Readme regarding baud rate and SPIFFS upload tool.
Arduino folder is already behind on the changes, I will delete it now, as you are confirming that PIO files are all good.

closed 2024-03-06 by aly-fly