aly-fly/EleksTubeHAX #59: DS1307RTC library by Michael Margolis is maintained by Paul Stoffregen

Fastdruid, 2024-01-18:
The instructions are explicit to use only the libraries listed however when searching for "DS1307RTC by Michael Margolis (developed on v1.4.1)" in the library manager this isn't found and instead only one by "Paul Stoffregen" is found. 

[image]

On looking this up however its written by Michael Margolis but maintained by Paul Stoffregen https://www.arduinolibraries.info/libraries/ds1307-rtc

Fastdruid, 2024-01-18:
Same goes for "Time by Michael Margolis (developed on v1.6.0)"

aly-fly, 2024-01-18:
Thanks for reporting. It looks like names changed since the instructions were written. 
Please report back if the project builds and works fine, we will update the instructions.

aly-fly, 2024-02-03:
Added updated lib descriptions in [my fork](https://github.com/aly-fly/EleksTubeHAX). Will be merged here a bit later.

closed 2024-02-03 by aly-fly