aly-fly/EleksTubeHAX #60: Build fails if MQTT or Geolocation is not configured

Fastdruid, 2024-01-18:
My USER_DEFINES.h block as it refers to MQTT and Geolocation was unmodified from the repo version.

```
//  *************  Geolocation  *************
// Get your API Key on https://www.abstractapi.com/ (login) --> https://app.abstractapi.com/api/ip-geolocation/tester (key) *************
//#define GEOLOCATION_API_KEY "key"

// ************* MQTT config *************
//#define MQTT_ENABLED  // enable after creating an account, setting up the device on smartnest.cz and pilling in all the data below:
//#define MQTT_BROKER "smartnest.cz"         
//#define MQTT_CLIENT "deviceid" 
//#define MQTT_USERNAME "Username"             
//#define MQTT_PASSWORD "MQTT_PASSWORD"
#define MQTT_PORT 1883                         // Broker port
#define MQTT_RECONNECT_WAIT_SEC  30   // how long to wait between retries to connect to broker
#define MQTT_REPORT_STATUS_EVERY_SEC  71 // How often report status to MQTT Broker

```

This fails as per the below:

```

C:\Users\fastdruid\sketchbook\EleksTubeHAX\Mqtt_client_ips.cpp: In function 'void sendToBroker(char*, char*)':

Mqtt_client_ips.cpp:66:32: error: 'MQTT_CLIENT' was not declared in this scope

     sprintf(topicArr, "%s/%s", MQTT_CLIENT, topic);

                                ^~~~~~~~~~~

C:\Users\fastdruid\sketchbook\EleksTubeHAX\Mqtt_client_ips.cpp:66:32: note: suggested alternative: 'MQTT_PORT'

     sprintf(topicArr, "%s/%s", MQTT_CLIENT, topic);

                                ^~~~~~~~~~~

                                MQTT_PORT

"C:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.5\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/heap/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/log/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/soc/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/soc/esp32" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/hal/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_common/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_system/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/xtensa/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/driver/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/driver/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_pm/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_ringbuf/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/efuse/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/efuse/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/vfs/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_wifi/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_event/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_netif/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_eth/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/tcpip_adapter/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_phy/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_phy/esp32/include" "-IC:\\Users\\fastdruid\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_ipc/include" "-IC:\\Users\\fastdruid\\AppD [... cut ...]
Using library ArduinoJson at version 6.21.1 in folder: C:\Users\dsharp\sketchbook\libraries\ArduinoJson 
Using library PubSubClient at version 2.8 in folder: C:\Users\dsharp\sketchbook\libraries\PubSubClient 
Using library Wire at version 1.0.1 in folder: C:\Users\dsharp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire 
exit status 1
'GEOLOCATION_API_KEY' was not declared in this scope
```

If we uncomment the #define for GEOLOCATION_API_KEY (as well as the one for MQTT_CLIENT) it builds successfully.

aly-fly, 2024-01-18:
MQTT is known issue. You need only ENABLE commented, all other are required in the compiler (and later ignored by the linker).
Geolocation needs additional ENABLE line in defines and in code so that it can be effectively disabled.

aly-fly, 2024-01-31:
Both issues fixed in my fork; will be merged to this repo a bit later, after full testing.

closed 2024-01-31 by aly-fly