aly-fly/EleksTubeHAX #61: Python script for mass image renaming and clk conversion

hallard, 2024-01-18:
During lunch time I asked chatgpt to convert pascal code to generate `.clk` file (because I'm on mac and file converter need windows platform) and here the result; I'll take some time later to see if it's working, but looks good.

```python
from PIL import Image
import os

def convert_image(input_file_path, output_file_name):
    # Load the image
    img = Image.open(input_file_path)
    W, H = img.size

    # Construct the output file path
    output_file_path = os.path.join(os.path.dirname(input_file_path), output_file_name + '.clk')

    # Prepare the header
    header = bytearray([ord('C'), ord('K'), W & 0xFF, (W >> 8) & 0xFF, H & 0xFF, (H >> 8) & 0xFF])

    # Create a memory stream (byte array) and write the header
    memory_stream = bytearray()
    memory_stream.extend(header)

    # Process and write pixel data
    for y in range(H):
        for x in range(W):
            R, G, B = img.getpixel((x, y))
            OutPixel = ((R & 0xF8) << 8) | ((G & 0xFC) << 3) | (B >> 3)
            memory_stream.extend(OutPixel.to_bytes(2, byteorder='little'))

    # Save to file
    with open(output_file_path, 'wb') as file:
        file.write(memory_stream)

# Example, loop through the image files named RETRO0.jpg to RETRO9.jpg
# convert them to bmp renaming them from 40 to 49 and finally convert the
# bmp to clk file
for i in range(10):
    # Open each jpg image file
    img = Image.open(f"RETRO{i}.jpg")

    # Crop the image to the specified coordinates (left, upper, right, lower)
    #cropped_img = img.crop((12, 22, 120, 220))

    # Save the cropped image as a bmp file named 40.bmp to 49.bmp
    #cropped_img.save(f"4{i}.bmp")
    img.save(f"4{i}.bmp")

    # Convert the bmp to clk
    convert_image(f"4{i}.bmp", f"4{i}.clk")
```

deansouth8, 2024-04-13:
Here is some really bad PowerShell, enter the 10th as a single digit then drag and drop.  I'm sure there is much better stuff out there. 

`# Load the required .NET assemblies
Add-Type -AssemblyName System.Windows.Forms
Add-Type -AssemblyName System.Drawing

# Function to resize and save an image as 8-bit Color BMP
function ConvertTo8BitColorBMP($sourceFile, $destinationFile, $width, $height) {
    Write-Host "Converting $($sourceFile) to $($destinationFile)"
    # Load the image
    $image = [System.Drawing.Image]::FromFile($sourceFile)

    # Resize the image
    $resizedImage = New-Object System.Drawing.Bitmap($width, $height)
    $resizedGraphics = [System.Drawing.Graphics]::FromImage($resizedImage)
    $resizedGraphics.DrawImage($image, 0, 0, $width, $height)
    $resizedGraphics.Dispose()

    # Create a new bitmap with desired dimensions and format
    $bmp = New-Object System.Drawing.Bitmap($width, $height)

    # Draw the resized image onto the bitmap
    $bmpGraphics = [System.Drawing.Graphics]::FromImage($bmp)
    $bmpGraphics.DrawImage($resizedImage, 0, 0)
    $bmpGraphics.Dispose()

    # Save the image as 8-bit Color BMP
    $bmp = $bmp.Clone([System.Drawing.Rectangle]::new(0, 0, $bmp.Width, $bmp.Height), [System.Drawing.Imaging.PixelFormat]::Format8bppIndexed)
    $bmp.Save($destinationFile, [System.Drawing.Imaging.ImageFormat]::Bmp)

    # Dispose the images
    $image.Dispose()
    $resizedImage.Dispose()
    $bmp.Dispose()
}

# Create a form for inputting the starting number
$numberForm = New-Object Windows.Forms.Form
$numberForm.Text = "Enter Starting Number"
$numberForm.Size = New-Object Drawing.Size(300, 100)
$numberForm.StartPosition = "CenterScreen"

# Add a label
$label = New-Object Windows.Forms.Label
$label.Location = New-Object Drawing.Point(10, 20)
$label.Size = New-Object Drawing.Size(280, 20)
$label.Text = "Enter the starting number for the output files:"
$numberForm.Controls.Add($label)

# Add a text box for input
$textBox = New-Object Windows.Forms.TextBox
$textBox.Location = New-Object Drawing.Point(10, 50)
$textBox.Size = New-Object Drawing.Size(100, 20)
$numberForm.Controls.Add($textBox)

# Add a button to confirm input
$button = New-Object Windows.Forms.Button
$button.Location = New-Object Drawing.Point(120, 50)
$button.Size = New-Object Drawing.Size(75, 23)
$button.Text = "OK"
$button.Add_Click({
    $numberForm.DialogResult = "OK"
    $numberForm.Close()
})
$numberForm.Controls.Add($button)

# Show the form and get the input
$numberFormResult = $numberForm.ShowDialog()

# If the OK button is clicked, proceed with conversion
if ($numberFormResult -eq "OK") {
    $startingNumber = [int]$textBox.Text

    # Create a form for drag and drop functionality
    $form = New-Object Windows.Forms.Form
    $form.Text = "Drag and Drop Images"
    $form.Size = New-Object Drawing.Size(400, 300)
    $form.AllowDrop = $true

    # Define the desired width and height for the images
    $width = 120
    $height = 174

    # Define the output file prefix
    $outputPrefix = $startingNumber.ToString("D2") + "_"

    # Define the drag drop event handler
    $form.Add_DragEnter({
        if ($_.Data.GetDataPresent([Windows.Forms.DataFormats]::FileDrop)) {
            $_.Effect = [Windows.Forms.DragDropEffects]::Copy
        }
    })

$form.Add_DragDrop({
    $files = $_.Data.GetData([Windows.Forms.DataFormats]::FileDrop)
    Write-Host "Files dropped: $($files)"
    foreach ($file in $files) {
        if (Test-Path $file -PathType Container) {
            # Get all image files from the directory
            $imageFiles = Get-ChildItem -Path $file -Include *.png,*.jpg,*.jpeg -File
            Write-Host "Image files found: $($imageFiles.Count)"
            foreach ($imageFile in $imageFiles) {
                $outputFileNumber = ($startingNumber * 10) + [int][System.IO.Path]::GetFileNameWithoutExtension($imageFile.Name)
                $outputFile = Join-Path $file ($outputFileNumber.ToString() + ".bmp")
                Write-Host "Output file: $($outputFile)"
                ConvertTo8BitColorBMP $imageFile.FullName $outputFile $width $height
                Write-Host "Conversion complete."
            }
        } else {
            $outputFileNumber = ($startingNumber * 10) + [int][System.IO.Path]::GetFileNameWithoutExtension($file)
            $outputFile = Join-Path (Split-Path $file) ($outputFileNumber.ToString() + ".bmp")
            Write-Host "Output file: $($outputFile)"
            ConvertTo8BitColorBMP $file $outputFile $width $height
            Write-Host "Conversion complete."
        }
    }
})

    # Show the form
    $form.ShowDialog()
}`

MikeyC3040, 2024-08-06:
I took your idea a step further, and since you are using PIL created a PIL plugin for CLK files and an example prepare script (with option to view CLK files)
https://github.com/MikeyC3040/EleksTubeHAX/blob/main/Prepare_images/CLKImagePlugin.py
https://github.com/MikeyC3040/EleksTubeHAX/blob/main/Prepare_images/CLKprepare.py

hallard, 2024-08-06:
Nice, what will your PR2 contains, new awesome features?

aly-fly, 2025-04-08:
BMPs with palletes have smaller size than CLK files, therefore the .CLK file format is deprecated.

closed 2025-04-08 by aly-fly