aly-fly/EleksTubeHAX #62: Numbers reversed (new / different hardware)

Fastdruid, 2024-01-18:
I've a different version *without* even a CH340 on board (the USB merely powers it), after some investigation of some interesting looking headers and some effort I got things to compile but the numbers are backwards. 

ie if the time now was 15:45:10 it would display 01:54:51

I fixed it by changing the indexing scheme in GLOBAL_DEFINES.h (eg swapping the number order so that SECONDS_ONES (0) became SECONDS_ONE (5) and similar for all the others). 

This worked. I've seen a few others reporting issues with backwards numbers so a note in the documentation or an option to redefine those 6 constants in reverse order would be handy.

Fastdruid, 2024-01-19:
I bought mine from Aliexpress but I believe it's the same as the PCBWAY one here https://www.pcbway.com/project/gifts_detail/RGB_Glow_Tube_Clock_907ad35c.html as there are people reporting the same "reversed digits" after flashing as well as the lack of a menu or any way to set the clock! 

Its labelled as "PUNKCYBER IPS CLOCK". 

There *should* be a CH340 chip on the board but its missing (unpopulated space) on mine. Others may have it populated. 

The processor on this is an ESP32D0WDQ5 (revision 1)

While there is no CH340 there are however three headers marked BOOT, EN and GRT.

BOOT is connected to GPIO0 (pin 25)
EN is connected to EN (pin 3)
G is connected to Ground (pins 1/38/15)
R is connected to GPIO1 / TxD (pin 35)
T is connected to GPIO3 / RxD (pin 34)

Connecting to flash therefore is a simple matter of a loop between the two boot pins, Gnd from the CH340 to G, Rx to R and Tx to T. Power from USB and short the two EN pins to reboot. 

[Photo of top of board](https://i.imgur.com/hubWh9p.jpg)
[Photo of bottom of board](https://i.imgur.com/RdmZnws.jpg)

aly-fly, 2024-01-31:
@Fastdruid  Thank you for your contributions and all data posted here. I have used your information on my fork (https://github.com/aly-fly/EleksTubeHAX), which is currently in Alfa stage for pre-release testing. Could you please download my code, test it on your clock model, and if you confirm everything works on your clock, I will merge changes into this repository.
Thanks.

aly-fly, 2024-01-31:
> ... RGB_Glow_Tube_Clock...
> Its labelled as "PUNKCYBER IPS CLOCK".

@Fastdruid I was looking at their website and video. Images and videos show two slightly different models, the most visible difference is labeling on the mainboard. Do you have any clue if this is just marketing fail and this is same clock?

Fastdruid, 2024-01-31:
Marketing fail.  The one I bought had pictures and video of a totally different one as well (the video was of the Elekstube one).  There are references in comments from the pcbway one that it is branded PUNKCYBER (as well as people saying that the numbers are reversed when they've flashed firmware).

What I can't say is if all PUNKCYBER ones are the same as mine,  certainly there should have been a CH340 on my board which wasn't fitted and I've not heard of anyone else who had to solder headers on!

Fastdruid, 2024-01-31:
The exact one I bought was this one https://vi.aliexpress.com/item/1005005910062138.html

The picture and video show it as a ELEKSTUBE IPS DOCKER V6.1.11

closed 2024-02-03 by aly-fly

Pacattacks, 2024-05-15:
I replaced the 340c from a ESP32 dev board, along with other missing components. I also rewired 74HC138D, as this makes the EleksTube software reverse the displays or I could changed it to a 74HC595 which is correct chip, which EleksTube used.

[image]

aly-fly, 2024-05-15:
Isn't it way easier to just reverse the defines in the clock firmware? I think we already implemented this setting with reversed displays.

Pacattacks, 2024-05-15:
On the clocks firmware, you need change the ESP32 3 pin select pins. Probably easy to do on SmittyHalibut custom source code, as I don't have EleksTube source code, its easier for me to change the 3-to-8 line decoder/demultiplexer chip for me.

aly-fly, 2024-05-16:
>  I don't have EleksTube source code

Oh well, here we are discussing directly in the repository of the custom firmware :)