aly-fly/EleksTubeHAX #68: Where is the prebuilt firmware?

mattfox27, 2024-03-01:
Where can I find pre built firmware?  Is it in the Repo?

aly-fly, 2024-03-01:
Not done yet. Can start with that, maybe this weekend. Which config you need first?

jwm0001, 2024-03-06:
Looking got gen2[EleksTubeIPS Gen 2 hardware], i brick it by flashing it with wrong bin file

victorvuelma, 2024-03-25:
@aly-fly 

What do you think about using GitHub releases to make multiple pre-built available? We can setup a GitHub action to automate the compilation process

jwm0001, 2024-03-25:
You talking about you creating those per-built. I have no experience in the compile of the software.

aly-fly, 2024-03-25:
> What do you think about using GitHub releases to make multiple pre-built available? 
That could be a solution.

> We can setup a GitHub action to automate the compilation process
Not sure if this will work, as it is required to build a complete flash image, including partition table, app, NVS section, SPIFFS with image files.. Sure it can be done with build tools, but the proven and simple way is to build and upload everything into actual ESP32 and then dump the .bin file from that.

mattfox27, 2024-03-27:
I ended up compiling it myself, which was not easy but I figured it out ;)

jwm0001, 2024-03-28:
thank you for your help. I really appreciate it.

aly-fly, 2024-03-28:
Done.
You can find files and notes here:
https://github.com/SmittyHalibut/EleksTubeHAX/tree/main/pre-built-firmware

closed 2024-03-28 by aly-fly

jwm0001, 2024-03-29:
I saw this morning, takes for your hard work. Will try this after work.

https://github.com/SmittyHalibut/EleksTubeHAX/tree/main/pre-built-firmware