aly-fly/EleksTubeHAX #72: [Contribution] Extend supported hardware to IPSTube

Holbormon, 2024-05-01:
Hi, and thanks for sharing your work !

I was gifted a [Chinese copy of the EleksTube](https://www.ipstube.com/en/product/IPStube-401.html), and decided to see if I could reverse-engineer it. Doing some light USB probing (mainly looking at `/dev/ttyUSB0` when pressing the reset button, output pasted below) reveals that it is very probably using some kind of ESP chip behind a CH340 USB-to-serial converter. [Further disassembly](https://github.com/SmittyHalibut/EleksTubeHAX/assets/160845407/0f6191c9-ea35-4d96-9791-b0995aab1162) shows what I believe to be an ESP32 chip (circled in red), wired to a USB port and a button (circled in green). 

All this has me thinking that your firmware might be applicable, unfortunately I lack knowledge about the programming flow of such IOT devices. In particular, I have looked up documentation about and succesfully setup ESP-IDF and ESPtool on my linux machine, but I can't get these programs to communicate with the clock over USB (despite getting some data as shown below)

Output of `/dev/ttyUSB0` when resetting the clock via button long press:

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (826) esp_core_dump_flash: No core dumÉÑÑœ¹found!
E (826) esp_core_dump_flash: No core dump partition found!
Software reset CPU
Software reset CPU
```

Is there anyone interested in extending support to this new device and can point me in the right direction ? Thanks in advance

aly-fly, 2024-05-02:
Interesting hardware. They opted to place ESP32 and supporting components directly onto the PCB instead of the readily available module. 

What you need to do:
1. Install esptool and download the existing firmware from the clock. Check flash size (4 or 8 MB) and load complete image.
2. Delete IDF and install PlatformIO.
3. Read the part number of the RTC chip next to 32 kHz crystal. 
4. Build the clock version that supports your RTC chip. 
5. Load and test. If not working, you will have to do signal tracing of the complete PCB, report here and then we can include it.

Holbormon, 2024-05-03:
Sorry if this is a trivial question, but as mentioned I have no experience with ESP32 chips, so I don't know how to boot mine in bootloader mode. I found [these instructions](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html) but I don't have the necessary hardware skills to process them. In particular, how do I control the `RTS` and `DTR` Serial pins of my computer ?

More generally, do you happen to have more detailed instructions of all the steps you mentioned, targeted at someone still learning about ESPs ?

aly-fly, 2024-05-03:
The ESPtool software will handle all the RTS and DTR lines for you (if properly wired on the board by the Chinese guys). And it will automatically reboot the clock into the bootloader mode.
Read the manuals how to use ESPtool (on Linux). Or google it. 
I can't offer much help, because on Windows everything is a bit different. I think if you put correct phrases into your fav search engine, it should give you the correct path directly. Ex: "save flash image esptool linux". For windows the path is somewhat similar, see here: https://github.com/SmittyHalibut/EleksTubeHAX?tab=readme-ov-file#backup-first for which switches and settings are needed to save your backup. It seems it will be a bit steep learning curve for you, but over one weekend you will be able to google and learn all of it :)

eku, 2024-05-03:
I could backup the original firmware with esptool. It's 4MB.

```
Detecting chip type... ESP32
Chip is ESP32-D0WDR2-V3-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

Holbormon, 2024-05-05:
Every time I try to use ESPTool I get "No serial data received". According to [their documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#no-serial-data-received), this means the hardware is not properly wired, hence my earlier question of whether I need to do something more to put the chip in bootloader mode ?

Martinius79, 2024-05-05:
Hello Holbormon!

I assume, you get the "No serial data received" message, while you are trying to write your backuped firmware back to the clock again? So something like "esptool --port /dev/ttyUSB0 --baud 921600 write_flash --erase-all 0x0000 backup1.bin", right? 
Or you wasn't able to read out the firmware yet at all? Means, no backup file of the orginal firmware yet?
Did you try to lower the Baud-rate like to 115200?

And you can also try to press this little button on your PCB, while plugging-in the "power cord" to the clock and hold it for a few seconds while the device is booting. Sometimes, this brings the ESP into the bootloader mode (depends on the wiring).

And like aly-fly indirectly mentioned, you need to identify the hardware used on the clock, so make pictures of the chips with labels/prints on them, so the letters can be read. Smartphone camera with zoom and different angles and light mostly will do the trick here. Most interesting for you is the ESP and the clock chip.

If you are really interested, you can also try to get the wiring of the PCB by taking photos from both sides, a multi-meter, some time and interest :) Then you can identify, how the ESP is soldered and if you can/need to solder some "helper" pins to put it into bootloader mode, if needed.

Bye
Martinius

eku, 2024-05-05:
AFAIR the button next to the  usb connector has to be pressed and hold while connecting the usb cable. Then start `esptool` to backup the original firmare. Same for writing any firmware to the device.

See also https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/pre-built-firmware/_ESP32%20save%20flash%204MB.cmd.

Holbormon, 2024-05-05:
> AFAIR the button next to the usb connector has to be pressed and hold while connecting the usb cable. Then start `esptool` to backup the original firmare. Same for writing any firmware to the [...]

Thank you for the precision. This seems to indeed boot the clock in a different mode (All screens are backlit but blank). I also had to use a different USB Cable because the one shipped with the chinese clock resulted in pySerial I/O errors. I have finally succeeded in running `flash_id`:

```
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WDR2-V3-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e4:65:b8:cd:1d:44
Uploading stub...
Running stub...
Stub running...
Manufacturer: 0b
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...
```

Now that my dev enviromnent is finally setup properly I will proceed to backup the chip and keep following the repo's steps.

eku, 2024-05-05:
> but a quick inspection of the resulting 4.0MB file shows empty regions at the beginning and start, suggesting that the actual firmware may be smaller

The flash is partioned. Don't know how to read the partition table back, but I guess there are at least 2 OTA and a FFS partition.

Holbormon, 2024-05-05:
> The flash is partioned. Don't know how to read the partition table back, but I guess there are at least 2 OTA and a FFS partition.

Just did thanks to [these instructions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html) and `gen_esp32part.py` from `esp-idf`. [I have attached the output for reference](https://github.com/SmittyHalibut/EleksTubeHAX/files/15213205/IPSTube-parts.csv). Interestingly, my clock was shipped with one more app/partition than the layout shown [here](https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/EleksTubeHAX_pio/partition_noOta_1Mapp_3Mspiffs.csv)

eku, 2024-05-05:
@Holbormon thanks for the info. Looks like we need to dump 8MB to also safe the original pictures in spiffs.

Could you please provide a more sharp foto of the pcb, so we could read the component identifiers?

Holbormon, 2024-05-05:
> Looks like we need to dump 8MB to also safe the original pictures in spiffs.

Done.

> Could you please provide a more sharp foto of the pcb, so we could read the component identifiers?

[image]

Speaking of the PCB, can anyone advise whether my hardware looks susceptible to the power problem that the original Elekstube has (5V being fed to the ESP32 as described [here](https://github.com/SmittyHalibut/EleksTubeHAX/blob/main/Hardware%20modification/README.md)) ?

eku, 2024-05-06:
Flash is XT25F64 from XTX Technology Limited, which is 64x1024kB = 8MB. RTC is a DS1302 replica.

So far so good.

It would be interesting to know which LCDs are used and how they are connected. Can you find out?

eku, 2024-05-06:
@Martinius79 
> And you was successfull with uploading a custom firmware?
> So which one did you use? One of the pre-builded ones?

I'm still on original firmware and waiting for more hardware details.

I generally have the confidence to compile the software myself once I have all the details for the configuration.

Holbormon, 2024-05-06:
> Flash is XT25F64 from XTX Technology Limited, which is 64x1024kBit = 8MB. RTC is a DS1302 replica.
> 
> So far so good.
> 
> It would be interesting to know which LCDs are used and how they are connected. Can you find out?

Again, I am much more versed in software than I am in hardware. Also, I don't have access to basic tools such as multimeters and soldering irons. Would more pictures of the other side of the PCB suffice to tell ?

Martinius79, 2024-05-06:
> Again, I am much more versed in software than I am in hardware. Also, I don't have access to basic tools such as multimeters and soldering irons. Would more pictures of the other side of the PCB [...]

It is about the six little TFT-LCD panels, which are displaying the "digits" of the clock. 
Can you take photos of them as well or identify the manufacturer or model number? 
Then you can watch for the datasheet and check if the resolution is the same like for other models and which driver is needed for them.
So yes, more pictures are always helpful ;)

aly-fly, 2024-05-07:
It is required to get full reverse-engineering of the hardware, all the chip details, which pin of which component is connected to which pin on the other component(s).
There are only two options, either one of the clock owners does the detailed hardware reverse engineering, or send the clock to someone that can do that. 

Because the clock has 8 MB flash (double of what everyone else has) there are two ways forward:
1. keep this project as-is and only use half of available flash (simple, but ineffective), or
2. reworking the partition table for larger flash for this build type only.
Firmware should be able to automatically count number of uploaded clock fonts.

Regarding the voltage issue from the original clock - best is to measure the supply voltage of the USB chip. If it is 3.3V then it is ok. If it is 5V, you have to search for any resistors in series with the Data TX line.

Holbormon, 2024-05-10:
Got some more pics of the PCB/LCDs. Unfortunately unable to determine manufacturer/model by mere visual inspection. Does this help ?

[image]
[image]
[image]
[image]

Martinius79, 2024-05-10:
Hello again Holbormon,

If the link is right in your first post, the clock is from "IPSTube" and it is model "401". 
If I search for it on the usual suscpect websites for china tec, it seems, as there are two versions of this clock available.
One with LCD Lights on the bottom and one without (see the screenshots).

Which one of them do you own?

[image]
[image]

Bye
Martinius

Holbormon, 2024-05-10:
> Which one of them do you own?

It's a bit misleading because the case of mine looks like the bottom picture (black and gold lines), but the base is most definitely not lit up with LEDs. I'd say it's the top model, no lights.

Martinius79, 2024-05-10:
Ah, my bad, the first one seems to be the model 402. Your is like this one, I guess:
[image]

Holbormon, 2024-05-10:
> Ah, my bad, the first one seems to be the model 402. Your is like this one, I guess:

That's exactly it !

Martinius79, 2024-05-10:
> Got some more pics of the PCB/LCDs. Unfortunately unable to determine screen manufacturer/model by mere visual inspection. Glare on the last one shows a bit of LCD wiring. Does this help ?
> 

Sadly, not really. It just shows us, that the LCD panels are not "plugable" and not using a socket, like it is for the most Eleksmaker IPS clock copies. And it shows also, that they "glued" some stickers with a dragon on it :)

Most of these clock manufactures are just copycats, so the chance, that they are using nearly the same panels with the ST7789 driver as all the others, are high. But most of these small panels using 8 lines, so you can't be sure. 
This device is definetly different from the usal copies. Other ESP32 with bigger flash chip (as eku mentioned).

The only way, is to follow the wirings to find the right pins ;) and then try and error, if the usual driver is working.

If you are willing to risk to damage the device, you can "unplug" one panel and try to remove it and use an ardiuno board to communicate with it :)  Or try to remove the stickers, but that will probably damage the panel and there is no gurantee that you will find any model numbers under it.

So it depends, on how much time you want to invest and if you want to risk damaging the device to go on from here.

Bye
Martinius

eku, 2024-05-11:
I'm currently setting up the development enviroment to build and test self comiled firmware with the IPS Tube.

The instructions recommend the use of VSCode with plug-in for PlatformIO. As a fan of the command line, I prefer to use Makefiles. Can anyone help me with this?

Holbormon, 2024-05-11:
> I'm currently setting up the development enviroment to build and test self comiled firmware with the IPS Tube.
> 
> The instructions recommend the use of VSCode with plug-in for PlatformIO. As a fan of the command line, I prefer to use Makefiles. Can anyone help me with this?

That's actually what I've also been trying to do this whole time. Here's where I stand: I have installed `esp-idf` using the [official guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html) (Under Manual Installation), and am currently trying to fit the code from `EleksTubeHAX_pio/src` within a default idf project template in order to just run `idf.py build`. I think the bootloader.bin is actually generated fine but I'm still getting errors compiling the main code, probably related to using PlatformIO source files with the idf toolchain (undefined references, etc...). Will report back if I find a way to make it fit.

Holbormon, 2024-05-11:
> As a fan of the command line, I prefer to use Makefiles. Can anyone help me with this?

I was actually also working on this myself. For now I have succesfully followed [these instructions](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html#manual-installation) up to the point of having to run `idf.py build`. Now I figure it's just a matter of cramming the source code of this repo's [pio folder](https://github.com/SmittyHalibut/EleksTubeHAX/tree/main/EleksTubeHAX_pio/src) into an idf project. Will report back if I manage to do that.

Martinius79, 2024-05-12:
> I'm currently setting up the development enviroment to build and test self comiled firmware with the IPS Tube.
> 
> The instructions recommend the use of VSCode with plug-in for PlatformIO. As a fan of the command line, I prefer to use Makefiles. Can anyone help me with this?

Even if you don't want to use Visual Studio Code (which I can highly recommand for every OS), there is the CLI of PlatformIO (see https://docs.platformio.org/en/stable/core/index.html). It is doing all the "magic" in the backgound. You should be able to use it without any GUI (see https://docs.platformio.org/en/stable/core/installation/shell-commands.html#piocore-install-shell-commands).
The platformio.ini is the "make file" for this tool and can be used to get the depenencies, build, link and upload the firmware.
Of course, you can try to build all tools and libraries locally and with the belonging compiler or interpreter, but I don't see the use of it, except you really want to learn a lot :)
If you analyse the source code of PlatformIO or put it the PIO system on verbose logging you will see, what it is doing in the background. Actually nothing spectecular. Grabbing the dependencies you definded and put them on the right place, compile your code with the right includes and the right switches for the target architecture and to run the linker. The binary is then uploaded with a variant of the ESP32 tool is this project.
PlatformIO makes it WAY easier, then doing all the single steps in seperate tools/consoles. And with the GUI in Visual Studio Code it is all integrated in one place. 

Bye
Martinius

eku, 2024-05-20:
The instructions for building the firmware yourself need a little revision. The chapter "_Configure the TFT_eSPI library_" comes before "_Configure, Build and Upload New Firmware_", although it requires files from the following step. 

The Pythen scripts do not contain a shebang and it is not clear whether they require version 2 or 3.

I'm still fighting my way through it. To be continued...

eku, 2024-05-20:
Configuring the firmware is tricky.

I only see 5 circuits, ESP32, Flash, RTC, CH340 and voltage regulator. No shift register for selecting the LCDs. No control buttons.

Any ideas?

aly-fly, 2024-05-21:
> The Pythen scripts do not contain a shebang and it is not clear whether they require version 2 or 3.

PY comes together with PlatformIO + VS code bundle. Therefore no need to know which revision is it.  If you still insist on going barebones (CLI), this is up to you.

> I only see 5 circuits, ESP32, Flash, RTC, CH340 and voltage regulator. No shift register for selecting the LCDs. No control buttons.

Probably everything is connected directly to the ESP32. If there are no buttons, this frees additional IO pins, so there might be enough of free pins for all displays.
If there are no buttons, how do you operate the clock with stock firmware?

eku, 2024-05-21:
> If there are no buttons, how do you operate the clock with stock firmware?

They provide an [Android App](https://www.ipstube.com/en/download/ipstube-app.html) that connects via  Bluetooth. 
No registration is required for simple settings. Registration is required to install and select additional clock images.

aly-fly, 2024-05-22:
You will have to modify existing code quite a bit to remove all buttons, etc.
The control of "our" clock here can be done via the MQTT.

Martinius79, 2024-05-22:
I finally received my parcel from china with two of the Model H401 IPSTUBE clocks (with and without "glowing" bottom). 
I just need to find the time to analyse them :( 
Maybe eku will be faster ;)

Foamier, 2024-06-08:
hi all,
Got myself an IPSTube version as well in the hope this software could be made to work. It looks promising, ESP32 2MB PSRAM, 8MB flash, as shown by others above. Mine of type H401 like the one above.
Software/apps can be taken from [http://ipstube.com/](http://ipstube.com/).

Yesterday I took a look at the pinouts of the IPSTube PCB and there are some differences compared to the other versions.
The displays are connected with all pins together, except for pin8 (CS). The CS lines connect directly to the ESP32. The displays and the background LEDs are always on when the device has power (not very nice, perhaps a switching transistor between LED and/or display power pins would be nice to make it more like this project).

`Pin  name   ESP32 name     `
`1    LED    3V3   3V3      `
`2    GND    GND   GND      `
`3    RESET        unused?  `
`4    RS     15    GPIO26   `
`5    SDA    14    GPIO25   `
`6    SCL    12    32K_XP   `
`7    3V3    3V3   3V3      `
`8    CS     See next table `

`Display CS pin   ESP32 pin `
`FPC1 (L)         20        `
`FPC2             18        `
`FPC3             17        `
`FPC4             16        `
`FPC5             22        `
`FPC6 (R)         21        `

The displays (FP-114H03A JY) seem to have the following specs and connect with an FPC connector to the board.
- IPS 1.14 inch HD TFT LCD Display Screen
- Size: 1.14 inch
- LCD Type: TFT
- Outline Dimensions: 17.6*31.0*1.6mm
- Active Area: 14.864*24.912mm
- Number of Dots: 135*240
- Controller: ST7789
- PIN NO.: 8P
- Pin Pitch: 0.5mm
- Display Color: 262K
- Backlight: 1 White LED
- Polarizer Mode: Transmissive
- Data Interface: SPI

DS1302Z Clock/calendar chip
`pin  purpose  ESP32 pin    `
`1    3V3                   `
`2    X1                    `
`3    X2                    `
`4    GND                   `
`5    CE       42    GPIO21 `
`6    IO       38    GPIO19 `
`7    SCLK     39    GPIO22 `
`8    Battery               `

25F64    8MB PSRAM        
`pin      ESP32            `
`1        30    SD_CMD     `
`2        32    SD_DATA_0  `
`3        29    SD_DATA_3  `
`4        GND              `
`5        33    SD_DATA_1  `
`6        31    SD_CLK     `
`7        28    SD_DATA_3  `
`8        26    VDD_SDIO   `

The is only one button on the device. Perhaps this could be good enough as mode or power button while the rest is controller through the web interface.
`pin       ESP32                        `
`1         23 GPIO0 with 10k pull down  `
`2         GND                          `

The RGB LEDs behind the displays are likely NeoPixel Mini 3535 RGB LEDs.
The first LED is behind left display, chaining to right display like most ws2812B do as well.   

`pin  type   ESP32 pin         `
`1    DIN    35    GPIO18      `
`2    VDD                      `
`3    DOUT    to next LED DIN  `
`4    GND                      `

The last (6th) LED (R) DOUT chains to RGB connector under board, I suppose to add more LEDs for underneath lighting.

There is also a USB serial port converter of type CH340N, which connects to U0TXD/U0RXD.

Have not looked at the code in this project close enough, but it seems that some things need to be adjusted:
If hardware is HARDWARE_IPSTube_CLOCK:
- Directly control the display CS lines (in TFTs.cpp or ChipSelect.cpp to abstract the hardware?)
- Allow for use of the NeoPixel Mini 3535 RGB LEDs.
- Adjust the partition table to ~7MB picture storage.
- Do something useful with the single button at the back.

In the coming days/weeks I try to figure something out (unless someone can help with at least the direct drive of the CS lines).

If anyone is interested in photos of this devices let me know and I'll upload to a new hardware folder of your choosing.

Foamier, 2024-06-08:
FYI, reading the info of the ESP32 and backup of the flash works well (you need to have Arduino or PlatformIO installed with the ESP32 package added). COM port info can be taken from Windows device manager or otherwise.
Hold the button on the IPSTube and hold for a few seconds while applying the power through the USB port on your PC, so esptool can upload the stub program to perform actions. Power off when finished to have it run the program again.

From Windows command line or powershell:
`>esptool.py --port COM8 --baud 115200 chip_id                                               `
`esptool.py v4.5.1                                                                           `
`Serial port COM8                                                                            `
`Connecting.......                                                                           `
`Detecting chip type... Unsupported detection protocol, switching and trying again...        `
`Connecting...                                                                               `
`Detecting chip type... ESP32                                                                `
`Chip is ESP32-D0WDR2-V3-V3 (revision v3.1)                                                  `
`Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None        `
`Crystal is 40MHz                                                                            `
`MAC: e4:65:b8:12:34:56                                                                      `
`Uploading stub...                                                                           `
`Running stub...                                                                             `
`Stub running...                                                                             `
`Warning: ESP32 has no Chip ID. Reading MAC instead.                                         `
`MAC: e4:65:b8:12:34:56                                                                      `
`Hard resetting via RTS pin...                                                               `

Backup of the original flash image, so it can be put back if needed:
`> esptool.py --port COM8 --baud 115200 read_flash 0x0 0x800000 IPSTube_1_2.bin              `
`esptool.py v4.5.1                                                                           `
`Serial port COM8                                                                            `
`Connecting...                                                                               `
`Detecting chip type... Unsupported detection protocol, switching and trying again...        `
`Connecting...                                                                               `
`Detecting chip type... ESP32                                                                `
`Chip is ESP32-D0WDR2-V3-V3 (revision v3.1)                                                  `
`Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None        `
`Crystal is 40MHz                                                                            `
`MAC: e4:65:b8:12:34:56                                                                      `
`Stub is already running. No upload is necessary.                                            `
`8388608 (100 %)                                                                             `
`8388608 (100 %)                                                                             `
`Read 8388608 bytes at 0x00000000 in 780.2 seconds (86.0 kbit/s)...                          `
`Hard resetting via RTS pin...                                                               `

aly-fly, 2024-06-11:
> hi all, Got myself an IPSTube version as well in the hope this software could be made to work. It looks promising...

You are on the right track. The hardware description is very good and with such da5ta it is 95% chance to get it working on the first try.
Let us know if you need any help or are you independent in making those modifications?

Foamier, 2024-06-11:
Hey, thanks for the reply. I'll try some basic code first to see if I can get some text on the displays so I know all pins etc. are working as expected. Then I'll have a look to see if adding another hardware type to the code is doable for my limited C++ experience. Might take some weeks to scramble some time after work though.

Yesterday the IPSTube stopped working after I verified all the pins again. The 3V3 regulator was getting very hot, so I expect a short circuit somewhere. Need to look into this further but I expect that the when the PCB is screwed securely to the aluminum milled base there is a short between 3V3 and GND. Might be why the screws were not tightened when I got the unit new.

Hardware wise it is not a bad build for the cost (45 euro), the case is aluminum and the tubes are actually made of glass. The displays are replaceable without soldering if needed, but this requires some warming up of the soldered-in support PCB of the displays. The dragons look to be copper on the support PCB. To nice these up I painted the sides black.

Depending on how the Elekstube features/control can be done on this hardware, I might consider some hardware changes like adding buttons, a LED strip expansion and temperature or other sensors. Assigning unused ESP32 pins is a lot harder with the ESP chip directly on PCB. The pins are so tiny that even my sharp measuring pins moving over the ESP pins caused some minuscule shorts between some pins, even without a lot of force while identifying the PCB traces. My guess is these PCBs are not ROHS compliant, as in lead being used in the solder...

Martinius79, 2024-06-11:
Hello @Foamier!

You were faster and better than me :) I was stuck for some time because I am really not good at reverse engineering a PCB and was distracted in so many ways. So your description is completely right and after your post, I finally was brave enough to upload my modified code to the last working H401 model I have (I destroyed the ESP32 on one of my boards because I measured too much).

For now, a lot is already working. The RTC seems to be fine, and the RGB-LEDs on the bottom of the TFTs are working fine. 

Sadly, the main part is not working for me: The displays are not showing anything at the moment. Not sure why :)
Also the "bottom" RGB-LED stripe is not working. But the connectors for this should also on your board already.

I uploaded my modifications to a branch in my fork of this project. Including my bad quality pictures and confusing wiring diagram as KiCAD files in the H401 folder:
https://github.com/Martinius79/EleksTubeHAX/tree/H401TryoutsAndOtherStuff

That is a start, IMHO, but maybe @aly-fly has more knowledge on addressing the TFT-LCDs than me. I am nearly sure these are also ST7789-based panels because the JPEGs in the original firmware are also 135x240 pixels, and it is also an 8-pin layout.

The modified board description for PIO is also there. After erasing once, I was able to flash the firmware image and then the filesystem image. The annoying part is having to put the chip into bootloader mode every time.

And maybe it is ROHS compliant (hazardous material and so), but they are using an "unshielded" version of the ESP32, so the EMC is not ensured, I bet. Nice for learning a lot about microcontrollers for MC-noobs like me. But there is a reason why you can buy these MCs in packages and very rarely the pure ESP32 :) They also use different versions of the ESP32. Like the pictures from the original poster here, has no I-PEX connector or external antenna (so U-version and not).

I also noticed that if I reflash the original firmware, there is a "firmware error" shown because the core dump partition seems to be missing. The firmware is working for me, but I never tried the original app with it. Any ideas on this?

I also have a lot of unrelated code in the fork, like setting a "hard" timezone or using an emulated ESP32 from the Wokwi stuff. Just ignore that...

Bye
Martinius

aly-fly, 2024-06-11:
Regarding displays. I suggest that you make a code from scratch. An example from the eTFT library. Configure the example in such way that only one display is active - only one CS line. Just ignore the others - set ESP pins to High (other displays ignore the SPI bus). When you get it working, you can easily port the required knowledge into "clock" project.

Foamier, 2024-06-11:
> Martinius

Great effort already from your side, was just about to find out how to create a json for a the board, but you already uploaded one! Excellent. Will have to check the schematic later, only so much time in a day. My method was a bit more blunt, by doing connectivity testing between the pins of the displays and the ESP32. Then some more voltage testing after power on.
ChatCPT4 suggested some code for Arduino, so will be molding it into something I hope will work to test the access to the displays (with AI and my own skills it will not be pretty, but guess that's not the point).

Foamier, 2024-06-12:
Brief update on getting the displays going (or better said, not going...).
I've tried to get something to show on the displays, but it does not work so far.
Normal code that talks to the serial port uploads fine and works as expected.
Using Arduino 2.3.2 for now for simplicity before jumping to VSCode and PlatformIO.

Using an Adafruit graphicstest example there is no response.
https://learn.adafruit.com/adafruit-1-14-240x135-color-tft-breakout/arduino-wiring-test

Adafruit_ST7789 tft = Adafruit_ST7789(20, 14, 12, 13, 15);

The pinout that I use is a follows:
`Display pin  name     ESP pin  Libray name `
`3            RESET    15       TFT_RST     `
`4            RS       14       TFT_DS      `
`5            SDA      12       TFT_MOSI    `
`6            SCL      13       TFT_SCLK    `
`8            CS       20       TFT_CS      `

A simpler example using the TFT_eSPI library (Walking 1 write and read pixel test) also does not show anything on the display.
The serial output shows that it (tries to) write to the display but subsequent reading always returns zero.
User_Setup.h:
`#define USER_SETUP_INFO "User_Setup" `
`#define ST7789_DRIVER                `
`#define TFT_SDA_READ                 `
`#define TFT_WIDTH  240               `
`#define TFT_HEIGHT 135               `
`                                     `
`//IPSTube pins for ST7789 FP-114H03A `
`#define TFT_MISO  -1                 `
`#define TFT_MOSI  12  //SDA          `
`#define TFT_SCLK  13  //SCL          `
`                                     `
`#define TFT_CS    20  //CS           `
`#define TFT_DC    14  //RS           `
`#define TFT_RST   15                 `
`                                     `
`#define SPI_FREQUENCY   1000000      ` Also tried other frequencies

I wonder if I'm missing anything obvious for getting these displays to work. Does anyone have any ideas for this? Maybe the CS lines of the other displays need to be defined decently (either on or off)?

The next step might be to disconnect the IPS displays and connect a LED on each of the GPIO connection and use a simple blink program to make sure all connections are the ones I think they are.

Btw, did no try yet to put back the backup of the original H401 v1.2 firmware. I'm not too interested in signing up to a cloud service just to be able to add clock faces or pictures to the IPSTube clock. We want more of course ;-)

Martinius79, 2024-06-13:
Hello @Foamier,

my bad, I guess! I used the PIN numbers and not the GPIO numbers :(

This is beeter:
```
  #define TFT_ENABLE_PIN (GPIO_NUM_4)    // needes?

  // configure library \TFT_eSPI\User_Setup.h
  // ST7789 135 x 240 display with no chip select line
  #define ST7789_DRIVER     // Configure all registers
  #define TFT_WIDTH  135
  #define TFT_HEIGHT 240
  //#define CGRAM_OFFSET      // Library will add offsets required
  #define TFT_SDA_READ      // Read and write on the MOSI/SDA pin, no separate MISO pin
  #define TFT_MOSI (GPIO_NUM_32)
  #define TFT_SCLK (GPIO_NUM_33)
  #define TFT_CS   (GPIO_NUM_27) // ONE display ONLY for now
  #define TFT_DC   (GPIO_NUM_25)  // Data Command, aka Register Select or RS
  #define TFT_RST  (GPIO_NUM_26)  // Connect reset to ensure display initialises
```

The above is giving at least a responds from the LCD driver.

The 1 pixel write and read example from the TFT_eSPI repo has a strange behavior then.
It seems to shift the values by one bit, with each write/read cycle in the register of the LCD.

Output from serial is: 

```
time spent in loop (ms): 500
 Pixel value written = 1
 Pixel value read    = 1
 PASS 
time spent in loop (ms): 500
 Pixel value written = 2
 Pixel value read    = 40
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 4
 Pixel value read    = 800
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 8
 Pixel value read    = 1
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 10
 Pixel value read    = 40
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 20
 Pixel value read    = 800
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 40
 Pixel value read    = 1
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 80
 Pixel value read    = 40
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 100
 Pixel value read    = 800
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 200
 Pixel value read    = 1
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 400
 Pixel value read    = 40
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 800
 Pixel value read    = 800
 PASS 
time spent in loop (ms): 500
 Pixel value written = 1000
 Pixel value read    = 1
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 2000
 Pixel value read    = 40
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 4000
 Pixel value read    = 800
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 8000
 Pixel value read    = 1
 ERROR                 ^^^^
time spent in loop (ms): 500
 Pixel value written = 1
 Pixel value read    = 40
 ERROR                 ^^^^
```

I have no clue, why...

I will do a more basic example with the pure adafruit library for the ST7789 now...

Bye
Martinius

Martinius79, 2024-06-13:
I tried the Adafruit libary and now I know for sure, which pin is which CS line ;)
The selected screen is at least flickering and showing "something" from time to time, so I could make this list:

```
GPIO_NUM_15 //seconds ones
GPIO_NUM_2   //seconds tens
GPIO_NUM_27 //minutes ones
GPIO_NUM_14 //minutes tens
GPIO_NUM_12 //hours ones
GPIO_NUM_13 //hours tens
```

I created a small example
https://github.com/Martinius79/EleksTubeHAX/tree/OneDisplayTestH401/Tryout1/TryOut1

Tried different SPI modes and frequencies.

There is something missing here...More power from the transistor?

But I am gone for the weekend, so some else must go on from here for now.

Bye
Martinius

Foamier, 2024-06-14:
Hi @Martinius79 ,
Thanks for this one. It was the golden piece of info to get this going!
Noob like me did not realize that the pin numbers on the ESP chip package were not the pins in Arduino for a GPIO. Even the term GPIOx would not be accepted, I guess because this is not defined in the board information. Simply putting the GPIO pin numbers and numbers did the trick.
Also, as suggested by @aly-fly the CS pins do need to be defined and either put high or low, otherwise the displays seem to interfere on the SPI bus.

Code for filling all screens with changing colors:

```c++
//Color cycle IPSTube displays

// Used libraries:
// TFT_eSPI at version 2.5.43
// SPI at version 2.0.0 in esp32\2.0.16
// FS at version 2.0.0 in esp32\2.0.16
// SPIFFS at version 2.0.0 in esp32\2.0.16
//
// Arduino 2.3.2 with ESP32 Dev Module board.

#include <TFT_eSPI.h>
#include <SPI.h>

/* ---------------------------------------------------

IPSTube settings for ST7789 FP-114H03A in Libraries\TFT_eSPI\User_Setup.h

#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
#define TFT_SDA_READ       // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only
#define TFT_WIDTH  135
#define TFT_HEIGHT 240
#define SPI_FREQUENCY  55000000 // fastest tested to be working

#define TFT_MISO  -1
#define TFT_MOSI  32  //SDA
#define TFT_SCLK  33  //SCL

#define TFT_CS    13  //CS
#define TFT_DC    25  //RS
#define TFT_RST   26  
#define TOUCH_CS  -1
 --------------------------------------------------- */

TFT_eSPI tft = TFT_eSPI(); 

#define TFT_CS_1   13 //(GPIO13)
#define TFT_CS_2   12 //(GPIO12)
#define TFT_CS_3   14 //(GPIO14)
#define TFT_CS_4   27 //(GPIO27)
#define TFT_CS_5   2  //(GPIO2)
#define TFT_CS_6   15 //(GPIO15)

int cspins[] = {TFT_CS_1,TFT_CS_2,TFT_CS_3,TFT_CS_4,TFT_CS_5,TFT_CS_6};

#define BLACK 0x0000
#define BLUE 0x001F
#define RED 0xF800
#define GREEN 0x07E0
#define CYAN 0x07FF
#define MAGENTA 0xF81F
#define YELLOW 0xFFE0
#define WHITE 0xFFFF

int c=0;

void setup() {
  Serial.begin(115200);
  for(int i=0; i<6; i++) {pinMode(cspins[i],OUTPUT);}
  
  Serial.println("Initialising all displays.");
  
  //Actiavte and init all dispays
  for(int i=0; i<6; i++) {digitalWrite(cspins[i], 0);}
  tft.init();

  Serial.println("Initialised all displays.");
  }

void loop() {
    tft.fillScreen(RED); delay(50);
    tft.fillScreen(BLUE); delay(50);
    tft.fillScreen(GREEN); delay(50);
    tft.fillScreen(CYAN); delay(50);
    tft.fillScreen(YELLOW); delay(50);
    tft.fillScreen(MAGENTA); delay(50);
    }
``` 

I noticed that you need to wait at least like 5 seconds after removing power to upload to the IPSTube device with the button pressed, otherwise it will not reboot correctly.

The display are actually quite fast. SPI_FREQUENCY up to 55000000 is working fine. I did notice that the program crashes after a few minutes. not sure why. The same happens around the same time at the lower SPI frequencies.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d44ce  PS      : 0x00060a30  A0      : 0x800d2ffa  A1      : 0x3ffb21d0  
A2      : 0x3ffbdbb4  A3      : 0x00001001  A4      : 0x00000003  A5      : 0x00000001  
A6      : 0x007bf388  A7      : 0x003fffff  A8      : 0x80088f67  A9      : 0x3ffb2190  
A10     : 0x3ffc2290  A11     : 0x3ffbf388  A12     : 0x3ffc2294  A13     : 0x00060a23  
A14     : 0x00060a20  A15     : 0x00000001  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000003  LBEG    : 0x40086431  LEND    : 0x40086441  LCOUNT  : 0xffffffff  

Backtrace: 0x400d44cb:0x3ffb21d0 0x400d2ff7:0x3ffb21f0 0x400d1f9e:0x3ffb2220 0x400f3585:0x3ffb2250 0x400d147f:0x3ffb2270 0x400d58c5:0x3ffb2290
```

eadmaster, 2024-06-14:
does wifi work as expected?
i'd like to try streaming custom text/images from a pc.

eku, 2024-06-15:
> does wifi work as expected?

His test code does only use the LCDs. We are far away from complete CFW.

eku, 2024-06-15:
> Guru Meditation Error

You may want to decode the dump with [Arduino ESP8266/ESP32 Exception Stack Trace Decoder](https://github.com/me-no-dev/EspExceptionDecoder) or [ESPressif's Execption Decoder](https://community.platformio.org/t/how-do-i-use-esp32-exception-decoder/22789).

Foamier, 2024-06-15:
Hi all, 
Maybe it's just my programming skills or something that needs to be addressed.
When running the code below, it does update the displays with the required text, however the left display on GPIO13 also always updates with every update. It looks like GPIO13 always goes LOW with each display update. GPIO13 is a strapping pin on ESP32, not sure if that matters if the pin is in output mode.
With the code below you can see that display 0 gets the text from every update.
Is this perhaps that others could test and see?
```c++
//Test test IPSTube displays

// Used libraries:
// TFT_eSPI at version 2.5.43
// SPI at version 2.0.0 in esp32\2.0.16
// FS at version 2.0.0 in esp32\2.0.16
// SPIFFS at version 2.0.0 in esp32\2.0.16
//
// Arduino 2.3.2 with ESP32 Dev Module board.

#include <TFT_eSPI.h>
#include <SPI.h>

/* ---------------------------------------------------

IPSTube settings for ST7789 FP-114H03A in Libraries\TFT_eSPI\User_Setup.h

#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
#define TFT_SDA_READ       // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only
#define TFT_WIDTH  135
#define TFT_HEIGHT 240
#define SPI_FREQUENCY  55000000 // fastest tested to be working

#define TFT_MISO  -1
#define TFT_MOSI  32  //SDA
#define TFT_SCLK  33  //SCL

#define TFT_CS    13  //CS
#define TFT_DC    25  //RS
#define TFT_RST   26  
#define TOUCH_CS  -1
 --------------------------------------------------- */

TFT_eSPI tft = TFT_eSPI(); 

#define TFT_CS_1   13 //(GPIO13)
#define TFT_CS_2   12 //(GPIO12)
#define TFT_CS_3   14 //(GPIO14)
#define TFT_CS_4   27 //(GPIO27)
#define TFT_CS_5   2  //(GPIO2)
#define TFT_CS_6   15 //(GPIO15)

int cspins[] = {TFT_CS_1,TFT_CS_2,TFT_CS_3,TFT_CS_4,TFT_CS_5,TFT_CS_6};

#define BLACK 0x0000
#define BLUE 0x001F
#define RED 0xF800
#define GREEN 0x07E0
#define CYAN 0x07FF
#define MAGENTA 0xF81F
#define YELLOW 0xFFE0
#define WHITE 0xFFFF

void setup() {
  Serial.begin(115200);
  for(int i=0; i<6; i++) {pinMode(cspins[i],OUTPUT);}
  
  Serial.println("Initialising all displays.");
  
  //Activate and init all dispays
  for(int i=0; i<6; i++) {digitalWrite(cspins[i], LOW);}
  tft.init();
  tft.setRotation(0);
  //Deactivate CS for all dispays
  for(int i=0; i<6; i++) {digitalWrite(cspins[i], HIGH);}
  delay(500);
  Serial.println("Initialised all displays.");
  }

void loop() {
  for (int i=0; i<6; i++) {

    Serial.print("Display# is :");
    Serial.print(i);
    Serial.print(" Pin is: ");
    Serial.println(cspins[i]);

    digitalWrite(cspins[i], LOW);
    delay(100);
    tft.fillScreen(BLACK);
    tft.setTextColor(WHITE,BLACK);
    tft.setCursor (0, 128);
    tft.print("Text test on display ");
    tft.print(i);
   
    // deselect display i
    digitalWrite(cspins[i], HIGH);
    delay(500);
    }
}
``` 
Analyzing the voltage on the pin shows it is going up and down with every display update, not just the update intended for the specific display. This kind of rules out a physical aspect I suppose (PCB short to GND or similar).

Foamier, 2024-06-15:
> does wifi work as expected? i'd like to try streaming custom text/images from a pc.

yes, WiFi should work as well like any other ESP32, once we get to it...

aly-fly, 2024-06-20:
Glad to see that you are making progress.

Foamier, 2024-06-20:
> > does wifi work as expected? i'd like to try streaming custom text/images from a pc.
> 
> yes, WiFi should work as well like any other ESP32, once we get to it...

In the mean time I've tested the WiFi with a basic ESP32 WiFi manager sketch and it seems to be working fine. The RSSI is not great, but I guess that's to be expected with the PCB printed antenna in an almost closed aluminum case. I expect it will work fine in normal environments.

Martinius79, 2024-07-09:
Hello again,

I finally found the time, to implement the support for the IPSTUBE H401 clock.
Some stuff is missing (see PR)
https://github.com/SmittyHalibut/EleksTubeHAX/pull/74

With my version of the clock, everthing works stable.

Best way to control it in the moment is, to use MQTT messages, because usage of the menu with one button only is limited.
With my version of the clock, everthing works stable.

I only have clocks without "real" buttons, so please can someone check, if the real button clocks are still working as expected?

Looking forward for any comments on this.

Bye
Martinius

Martinius79, 2024-07-15:
I "reduced" the amount of changes to support just the IPSTUBE hardware and closed the PR mentioned above and created a new one:
https://github.com/SmittyHalibut/EleksTubeHAX/pull/77

MacGyver4B, 2024-09-07:
@Martinius79 I see a way to add more buttons.
The pin used is 23 GPIO0 right?
It is also known as ADC2_CH1 so voltage divider with switchable resistors should allow to connect more than one Button to the same IO.
Like in this example: https://www.instructables.com/How-to-Multiple-Buttons-on-1-Analog-Pin-Arduino-Tu/
I have the H401 without the bottom light stripe and I am planing to add some special mode where it displays a 6 digit Hex number. 
Is the number of dialplates increased for hardware with 8MB? or same as for the 4MB versions?
The original H401 firmware has one default and 2 alternative sets of dialplates plus 6 photo plates, my idea was to use one alternative set and the photo set to get the 16 hex digits in total.
I'm building a modern version of the MOS KIM-1 based on the WDC W65C02SXB and want to use the clock as a cool wireless display together with a N970 Pro BT Numeric 28-Keyboard connected via a Serial to Bluetooth adapter.

aly-fly, 2024-09-09:
> I "reduced" the amount of changes to support just the IPSTUBE hardware and closed the PR mentioned above and created a new one: #77

@Martinius79  This is a huge change list to digest and merge. In the mean time also the Main had a few updates. Should we just try and merge and debug after merging? The # 77 shows only 2 minor conflicts at the moment.

Martinius79, 2024-09-10:
Actually, this PR is closed since a while ;) I worked on another and even this will be closed soon :)

Martinius79, 2024-09-11:
I created a new pull request, with the actual main merged 
https://github.com/SmittyHalibut/EleksTubeHAX/pull/86

Martinius79, 2024-09-11:
> If I understand correctly, you are creating a new branch on your fork with streamlined changes, so that it will be merged more easily? Will you include also #80 with it?

New branch is created, but changes from PR [80](https://github.com/SmittyHalibut/EleksTubeHAX/pull/80) are not in there yet, but I can also include it. 
Should I?

Martinius79, 2024-09-13:
> @Martinius79 I see a way to add more buttons. The pin used is 23 GPIO0 right? It is also known as ADC2_CH1 so voltage divider with switchable resistors should allow to connect more than one Button to the same IO. Like in this example: https://www.instructables.com/How-to-Multiple-Buttons-on-1-Analog-Pin-Arduino-Tu/ I have the H401 without the bottom light stripe and I am planing to add some special mode where it displays a 6 digit Hex number. Is the number of dialplates increased for hardware with 8MB? or same as for the 4MB versions? The original H401 firmware has one default and 2 alternative sets of dialplates plus 6 photo plates, my idea was to use one alternative set and the photo set to get the 16 hex digits in total. I'm building a modern version of the MOS KIM-1 based on the WDC W65C02SXB and want to use the clock as a cool wireless display together with a N970 Pro BT Numeric 28-Keyboard connected via a Serial to Bluetooth adapter. One question to the round with this hardware I read somewhere that restoring the original firmware gave some errors, as something is missing is this solved is it possible to revert to stock now?

Sounds like an interesting project! :)
It should be possible to solder four "analog" buttons to the wires of the existing button, and theoretically, it is also possible to switch the pin to analog input. Have you made any progress with this?
I am not sure what you mean by "dial plates"... I guess you mean a set of pictures representing one "clock face." And yes, the IPSTUBEs all have 8MB flash, and nearly 6.8MB are usable for picture data. That is nearly double the size for clock faces then in the 4MB models, but the code currently limits the number of sets to a maximum of 8.
There is also no "picture" or album mode like in the original firmware, just the clock mode. There are mods of the repo doing this, AFAIK.
Using the displays to show a static picture is not a big deal. ;)

And if I flash back the original firmware to the H401, I receive an error, but the clock still works for me. I have never tried the original software/app, so I have no idea if it will be unhappy or will work... I’ll have to try it out.

Bye,
Martinus

Martinius79, 2024-09-14:
@MacGyver4B:

I flashed back my self created backup of the original FW of the H402 model. No errors was shown. 
The clock is working normaly.
Then I connected via the IPSTUBE original app from the Google store. 
All is working as far as I can tell...
So no problem here, to go back to the orginal firmware.

Bye,
Martinius

[6 more replies]

closed 2025-03-22 by Martinius79