aly-fly/EleksTubeHAX #75: The project cannot be compiled, program size is greater than maximum

gamba69, 2024-07-10:
VSC Terminal show:
=====
...
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1062425 bytes) is greater than maximum allowed (1048576 bytes)
RAM:   [===       ]  34.4% (used 112864 bytes from 327680 bytes)
...
=====
I would be grateful for any recommendations

aly-fly, 2024-07-10:
Which clock variant and options are you trying to compile?
It is possible to increase the program space a bit and sacrifice a little of the images sector.

gamba69, 2024-07-10:
Default from repo, as-is. But when I change partition_noOta_1Mapp_3Mspiffs.csv to:

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  factory, 0x10000, 0x110000,
spiffs,   data, spiffs,  0x120000,0x2E0000,
coredump, data, coredump,0x400000,0x10000,

all working fine. Is this right way?

aly-fly, 2024-07-10:
Yes this is correct modification.

Martinius79, 2024-07-15:
I noticed the same problem with the newest versions of EspressIF 32 development platform from PIO.

So I digged a bit deeper.

I tried with the newest and some older versions of the EspressIF 32 development platform module, but I always get much larger firmware files, then with my "original local version". 

So i watched my original local version more precisely and found, that my PIO initally used a fork of the EspressIF 32 development platform from the Tasmota project. I have no idea, why this happend, but the resulting ELF files are much smaller.

My "orginal local version" is from the GIT repo: https://github.com/tasmota/platform-espressif32.git
with version "2023.06.04" or Commit hash "8235aa5249f16aaffda3d29db6036edab64f4f35". 

The "real version" is from: https://github.com/platformio/platform-espressif32.git
Newest version "6.7.0"
Also available via PIO registry: https://registry.platformio.org/platforms/platformio/espressif32

I tried also to run it on GitHub Actions and got the same results:
With the fork and the described version -> 68-75pct of the size used (based on the selection of the hardware and features).
With original version -> 88-105 pct of size used.

See my tests in the GitHub Actions:
https://github.com/Martinius79/EleksTubeHAX/actions

To use the forked version modify the platform.ini as followed:

```
...
[env:esp32dev]
platform = https://github.com/tasmota/platform-espressif32.git#8235aa5249f16aaffda3d29db6036edab64f4f35
;or
;platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
board = esp32dev
framework = arduino
...
```

@aly-fly:
You have an idea, why the forked version was used? 
And what to do now? 
Using the Tasmota fork? Which version?
Or stay with the "orginal" from PlatformIO team?
Then alyways use the latest version?
And then the partition files needs to be modified, because the actual code results in binaries using over 100pct of the allowed size.

Bye
Martinius

aly-fly, 2024-07-16:
I don't like to use old (and potentially obsolete) libraries. Prefer always the most recent one with bug fixes and added functionality.
When I was building one of my other projects I saw different serial data structure on ESP32 startup/reboot. Probably that was an indicator of a new ESP environment. 
Dunno where the old version came from. It was never specified. Probably that was a bug in the PIO configuration or on their server side. It seems we will have to change the default flash mapping table.

Martinius79, 2024-07-21:
I created a PR for this:
https://github.com/SmittyHalibut/EleksTubeHAX/pull/78
Please approve

aly-fly, 2024-07-22:
Please don't make dependencies onto fixed revision unless explicitly necessary. Otherwise someone will have to update & test it on the regular basis. If version is not defined, project will automatically receive bug-fixes.

aly-fly, 2024-08-22:
Merged.