aly-fly/EleksTubeHAX #76: remote control: show custom numeric values/strings/images (WIP POC)

eadmaster, 2024-07-11:
this is currently a wip on [my repo](https://github.com/eadmaster/EleksTubeHAX/tree/AddedIPSTUBEH401).

It allows you to show custom numeric values/strings/images in the displays by sending commands over the serial port. e.g.:
````
SENSORS: CPU: 10, MEM: 20, TEMP: 50

TXT: Long text example this is very looong

52543002232 lrc line
````

example of the sensor command:
[image]

Currently only tested on IPSTUBE H401, it works fine as a POC but still needs some polishing.

aly-fly, 2024-07-12:
I suggest that you keep this code on your fork. The clock's main purpose is to be a standalone unit to show time on any shelf in the house. Optional commands transmitted over WiFi over MQTT. 
Maybe it might get included if you rework the code so that your PC sends data through the MQTT server and then to the clock (BMPs are too large for MQTT). But not with a constant cable link.

eadmaster, 2024-07-13:
no problem, indeed i've considered remote control using MQTT and even added [some stub code for it](https://github.com/Martinius79/EleksTubeHAX/blob/ba0c89314ea417a3cb5828f6a81c17a68842c3ac/EleksTubeHAX_pio/src/Mqtt_client_ips.cpp#L99), but i the end i've switched to Serial over USB to have the lowest latency. I've also considered a regular HTTP server with a json API, which would eliminate the need of a external server (i prefer standalone devices).

In addition, there are other parts that still needs polishing, so feel free to send me PMs if you work on any of these:
- TXT command may show a bigger text splitted on all the lcds, instead of repeating ([partially implemented here](https://github.com/Martinius79/EleksTubeHAX/blob/ba0c89314ea417a3cb5828f6a81c17a68842c3ac/EleksTubeHAX_pio/src/TFTs.cpp#L101))
- base64 img decoding not working correctly atm, still debugging this part (see the [code here](https://github.com/Martinius79/EleksTubeHAX/blob/ba0c89314ea417a3cb5828f6a81c17a68842c3ac/EleksTubeHAX_pio/src/TFTs.cpp#L101#L268))
- improve [spectrogram](https://github.com/Martinius79/EleksTubeHAX/blob/ba0c89314ea417a3cb5828f6a81c17a68842c3ac/EleksTubeHAX_pio/src/TFTs.cpp#L76#L146) aestetic (blend with bg img?) + support a custom number of bands

closed 2024-07-24 by aly-fly

eku, 2024-07-24:
> Optional commands transmitted over WiFi over MQTT.

That would be the way to go.