aly-fly/EleksTubeHAX #81: Question: Where and how is this temperature sensor (DS18B20) used?

Martinius79, 2024-08-05:
Hallo @all

I am wondering, if anyone knows anything about the temperature sensor mentioned in the code (DS18B20 with "One wire bus")?
I see that it should be connected via the GPIO4 to the original EleksTube clock, but I don't see any sockets or other connectors for it.

So was/is this just soldered to the ESP32 somehow?

Does anyone ever use this and can me tell more?

Bye
Martinius

aly-fly, 2024-08-19:
Hi, the sensor was soldered to one of the free pins on the board. That was only an experimental setup.
The libraries used are very clumsy. I was never been able to include them properly, the code only worked if all code was in the .h file. In the latest versions the sensor libraries caused the project to compile, but crash on startup. I was not able to resolve this, so it was deleted. 
If you want to have this functionality it is best to start from scratch and use different library than it was used here, before.

aly-fly, 2024-08-30:
Libraries disabled in `platformio.ini`.

closed 2024-08-30 by aly-fly

Martinius79, 2024-09-03:
First, thanks for the info!
And that didn't mean, that you should remove the support for it ;) 
I just wanted to know, where the sensor was connected to and if it is possible to use it on other clocks...
But I guess, the usage of this was realy limited, so removing the libs maybe is the best idea.

aly-fly, 2024-09-03:
It was useful as a "hidden" temperature sensor on the desk. Visible as a small text on the LCD, accessible over MQTT (I had in linked to Samsung SmartThings for automations and alarms). And with latest version it could be integrated into HA. 
Besides the code crashing, libraries used were also blocking all other functions for > 1 second during temperature conversion and readout. This should be re-done from scratch - if required.

Martinius79, 2024-09-03:
I noticed the HA support just now and happy about it ;) I already used the MQTT messages for other stuff and I also use HA for my home automation, so integration is very welcome :) 
Must integrate it into my branches now ;)

aly-fly, 2024-09-03:
Great, test it out. Any reports or fixes are welcome 👍