aly-fly/EleksTubeHAX #82: [Info] Home Assistant Edition with extensive MQTT support

gamba69, 2024-08-15:
Home Assistant Edition with extensive MQTT features:
-  Device detected as two different lights, main and back
-  Supported on/off, brightness, effects (main and back), color (back)
-  Supported 12/24 hours and blank zeroes settings switches
-  Supported effect's speed change for backlight
-  All options are discoverable via Home Assistant MQTT

https://github.com/gamba69/EleksTubeHAX

eku, 2024-08-16:
> Home Assistant Edition with extensive MQTT features

Would you like to incorporate the extensions into this project so that the developments do not diverge?

aly-fly, 2024-08-19:
> Home Assistant Edition with extensive MQTT features:

It would be great to incorporate this, under a new switch, I recommend `#define MQTT_HOMEASSISTANT`.
Please create a Commit.

gamba69, 2024-08-20:
Done (as PR).

closed 2024-08-22 by aly-fly