aly-fly/EleksTubeHAX #90: Can someone owning an open H401 clone check something?

MacGyver4B, 2024-10-15:
Would this 🕹️ buttons be able to replace the button and still fit in the case?
https://cdn-shop.adafruit.com/datasheets/SKQUCAA010-ALPS.pdf

Martinius79, 2024-10-25:
Hello MacGyver4B!

I don't think that this "joystick button" will fit easily with the H401 or H402. The problem is the casing, and the fact that the soldered button is not directly accessible from the outside. Instead, there is a long plastic stick and a round plastic piece around it.

I can just post some of the pictures I took to make it clearer.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Bye
Martinius

MacGyver4B, 2024-10-25:
Thank you very much.
I think the original button is 6x6 mm right?
The joystick is 10x10 mm so it would be needed to remove a bit of the PCB to make room for the joystick.
I would say still possible but really no fun to do so.
Was just an idea.
Best regards 
Stefan

closed 2024-10-25 by MacGyver4B