aly-fly/EleksTubeHAX #93: Windows Defender Blocks Project.

strykereye2, 2024-10-26:
Project has an Identified Severe Malware: Trojan:Script/Wacatac.B!ml
[image]
I am unsure if this is a false positive.

Martinius79, 2024-10-27:
Hello strykereye2!

If you donwloaded the ZIP from this repo (main branch), this is a false positive from defender.

Trojan.Script/Wacatac.B!ml is a "generic trojan" detection, which means, it is not based on the known signatures, so defender  "guesses". And this is known to be error-prone.

The best way to check something like this, is to do an upload or re-check on virustotal.com, because more then one scanner is used here.

I uploaded the GitHub generated ZIP file from the repo today: https://www.virustotal.com/gui/file/83b392706faa8d4b0b6aa70190a8aa4428855d644308a14dca8926efae3f9464

In the whole repo are only two executables. One for the ESP communication and one for the CLK file generation. The rest is pure C++ code for the clock (which runs only on an ESP32) and belonging ressources (Except some Python scripts and some Delphi code). So I really don't know, why the heuristic scanner was giving an alarm to you.

I am not sure, what kind of "threats" the defender has detected, but with the actual virus defenition files and settings, I don't see a detection on my Windows 11 machines.

[image]
[image]
[image]
[image]

Just try to update your defender definition files (via Windows Update), redownload the ZIP from the repo, check manually with Defender, make an VirusTotal check and tell us, if you still see this.

BTW: You can also use GIT to clone the repo, instead of downloading the generated ZIP file generated by GitHub.

Bye
Martinius

Martinius79, 2025-03-20:
I rescanned the zip content today and still no detection, as expected. So closing this one.

[image]

closed 2025-03-20 by Martinius79