aly-fly/EleksTubeHAX #96: NovelLife SE clock always in off state

hthiel75, 2024-10-31:
Happy newbie here, I flashed my NovelLife SE clock via https://www.nixies.us/projects/elekstubeips-clock/elekstube-ips-v1-2/

I'm having the problem that no matter what time setting I apply, the status is always "off," so I can only get a dim or matrix display. Is this a general issue, or could it be due to the way I flashed it?

Martinius79, 2024-11-01:
Hello hthiel75!

If you flashed the clock via the WebApp on the given URL, you are using a slightly different after market firmware then the one on this github page. 

It is maintained by @judge2005 and you can find the github project under: https://github.com/judge2005/EleksTubeIPS

I never tried to flash the firmware on my NovelLife, so can't help much in the moment...but what do you mean with "matrix" display? The clock starts up but is not showing anything on the displays?

hthiel75, 2024-11-01:
Thank you, I didn't realize I was on the wrong path here. So let's cut to the chase; I'd like to give this a try with your help.

Unfortunately, I failed when trying to flash. I flashed the device with the pre-built firmware:

shell
Code kopieren
esptool --port COM3 --baud 921600 write_flash --erase-all 0x0000 FW_NovelLife_SE_HAX_1.0.bin 
pause
It ran through, but unfortunately, everything just flickered.

It would be really nice if someone could give a newbie a little support here.

Martinius79, 2024-11-02:
It took me a while to try out the things ;)

First of all, yes, you are right, the image in the repo seems to not working correctly :(. 
I flashed it to my Novellife SE and it was just made the displays flicker a bit in black.

I created a branch in my fork with a working version. 
Default settings for this files are set (WPS on, serial debug output, no MQTT, no night time dimming, no geolocation).
See here: 
https://github.com/Martinius79/EleksTubeHAX/blob/AutoCreateFirmwareFileToFlash/pre-built-firmware/FW_NovelLife_SE_HAX_V1.0_HA_Edition.bin

I flashed it with 
```esptool.exe --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash 0x0000 .\FW_NovelLife_SE_HAX_V1.0_HA_Edition.bin --erase-all```

So please give it a try and tell us the result.

If you want to change the settings, you will have to install VS Code.

The (modified version of the) README.MD is a good starting point for that.
See: https://github.com/Martinius79/EleksTubeHAX/blob/AddedIPSTUBEs/README.md
under "How to build this firmware".

Bye
Martinius

hthiel75, 2024-11-02:
Hello, that was really enlightening, then I haven't done anything wrong at first. That's reassuring :-)

I've now done it exactly as you described. Only the USB port is logically different.

Connection via WPS works perfectly, clock starts. Unfortunately, it then stops immediately and cannot be reached via WEB. Too bad. Now I have no idea, my knowledge is too limited.

Many thanks for your trouble with me!

Translated with DeepL.com (free version)

Martinius79, 2024-11-02:
Happy to hear that the firmware image is more or less working for you...
But what do you mean with "it stops"? Does it not show the digits anymore or what happens?

And what do you mean with reachable via Web? There is no web server implementation yet in the code, so the only way to send commands to the clock, is via MQTT messages.

And did you try to build the firmeware on your own?

hthiel75, 2024-11-02:
Thank you for the effort you're putting in to help a newbie like me :-)

The clock starts, gets the time, and then keeps displaying it without any updates. So, the display basically freezes.

Oh, no web interface? I thought that was "normal." I think I'll have to bow out then, unfortunately. I have absolutely no idea what MQTT is and am just an ambitious user. Too bad—I’ll try asking over at https://github.com/judge2005 to see if he can solve the issue with "always off / or always dim." Thanks for now!

Martinius79, 2024-11-03:
You are welcome!

And I see, you raised an issue in the ElekstubeIPS project, so I guess, you can close this on here (https://github.com/judge2005/EleksTubeIPS/issues/21)

I answered you also there.

BTW: The gesture sensor is working in the version I provided, so the "control" of the clock can be done "manually", like in the original version...It is a bit "unhandy", because the sensor is not that precise and detects not all the movements, like I wish it would do.
But I guess, you want to control the clock via a web interface, so the ElekstubeIPS version is the better choice for you in the moment.

hthiel75, 2024-11-05:
Thank you all for the effort you put in. Thanks to the workaround, I now have a perfectly functioning clock.

closed 2024-11-08 by aly-fly