aly-fly/EleksTubeHAX #99: MQTT Error

ant456, 2024-11-06:
I keep getting this error when plugging in click on Smartnest.

Any help?

 Reason: Publishing to forbiden topic: homeassistant/light/672b8bdf10bc080d351604f2_main/light/config

aly-fly, 2024-11-07:
Most probably you have enabled both - Smartnest and HomeAssistent. You should enable only one of them.

aly-fly, 2024-11-07:
If you use Smartnetst web service, uncomment `MQTT_ENABLED` and fill in Username, Password, Client ID.
If you have HA, uncomment   `MQTT_HOME_ASSISTANT`   and all other with _MQTT_HOME_ASSISTANT_ in the name.

Martinius79, 2024-11-08:
If you want to use the Home Assistant Integration, you normally need some kind of MQTT broker running locally, which supports the HA discovery and communication messages.
If you have a HA running on HAOS, you will have a broker running already, like Mosquitto. Or you can set it up easily.
See: https://www.home-assistant.io/integrations/mqtt/

So if you set up the broker, you can enter the IP, port, username and password of it into the matching MQTT_* defines into your _USER_DEFINES.h to connect to your local hosted MQTT broker and then also enable the MQTT_HOME_ASSISTANT_* defines.

I am not sure, if there are any "public internet" based MQTT broker services, which supports "HA messages". Smartnest is not supporting it AFAIK, so the limit, when using smartnest, is controlling an "emulated temperature" sensor, which is then mapped to some very specific functions of the clock.

ant456, 2024-11-08:
Thanks for all the help! I figured it out now. I ended up using HA mosquitto broker and those credentials instead of Smartnest. Maybe add that to the readme unless I’m the only one that didn’t understand. No to figure out how to hide the seconds and add a space between hours and minutes so it looks cleaner. Thanks Again!

closed 2024-11-08 by aly-fly