aly-fly/EleksTubeHAX #108: additional maintainer(s)?

aly-fly, 2024-12-09:
@SmittyHalibut are you still around? :) 
Can you please promote the @Martinius79 to a co-developer of this project, so that I don't need to check and validate every pull request - there are many of them 💯

SmittyHalibut, 2024-12-09:
Invite sent.  Thanks for the heads up, you're right that I'm not paying much attention to activity here.  I do pay attention to mentions, however.

I'm still willing to just transfer ownership to you, @aly-fly.  You've done way more to support this project than I ever did.  Let me know if you want to make that happen.

Leaving this issue open until you decide.

aly-fly, 2024-12-09:
> I'm still willing to just transfer ownership to you, @aly-fly. 

I am in. Ok let's do this.

SmittyHalibut, 2024-12-09:
[image]

SO CLOSE!  :-)

Make sure your local repo is merged into mine, then delete yours.  Tell me when that's done, and I'll attempt the transfer again.

aly-fly, 2024-12-09:
I renamed it. My repo is an original fork, already stale for some time. Will delete it after a short review.

SmittyHalibut, 2024-12-09:
[image]

Bugger.  I think it needs you to delete it first.  It probably doesn't like that its a fork.

aly-fly, 2024-12-09:
Done. Went to electronic heaven.

SmittyHalibut, 2024-12-09:
[image]

Yay!

aly-fly, 2024-12-10:
It looks like that transfer was successful! Thank you.

closed 2024-12-10 by aly-fly

Martinius79, 2024-12-10:
Hello again!

Happy to see that you were "promoted" aly-fly :)

I guess, because the ownership changed, the invitation from SmittyHalibut is not valid anymore.

[image]

Maybe you can invite me again?