aly-fly/EleksTubeHAX #109: Feature Request: Countdown Timer for Elekstube

fuisl, 2024-12-11:
## Description
Add a countdown timer feature to the Elekstube ISP to enhance its functionality and usability for various timing tasks (pomodoro clock).

## Feature Details
- **Input**: Countdown time (e.g., in hours, minutes, and seconds) set via code/app or control buttons on the clock.
- **Display**: The countdown should be displayed prominently on the device.
- **Alert**: When the timer reaches zero, visual indication _(blinking 00:00:00 or just stays at 0s)_.

## Notes
- I've always wanted to have a countdown timer for my study session but not sure how can I turn this scene from Oppenheimer to reality.
- Maybe some guidance would be helpful if this is do-able for a beginner...

[image]

aly-fly, 2024-12-11:
Interesting idea. Yes it is doable. Setting and starting the input time will be a bit clumsy - there is no app for this clock. Some of the models only have one button.  
Also making a countdown timer for the last 99 hours of this year is somewhere in the planning. :)

masterhui, 2025-01-25:
Any progress on this topic? I would love the ability to start a countdown timer via a mqtt command :-) Then I could add it to my OpenHAB setup and make it Alexa controlled...

aly-fly, 2025-01-27:
Hi, unfortunately I have so many other open projects that this is low on the priority list... But I have it in mind. Maybe some other developer(s)have a bit more spare time..

masterhui, 2025-01-27:
Hi Aljaz, thank you for getting back to me. I will try to implement this feature myself then, currently still waiting for the hardware to arrive. Once I'm up and running I will get started. Maybe you could point me at the relevant code sections so I have a better idea where to start :-)

aly-fly, 2025-01-28:
>  Maybe you could point me at the relevant code sections 

If this would be on the list of requirements from the start, it would be differently implemented. But in this case we have to make a modification with minimal impact on the rest of the code.

1. Make a Flag (boolean) to switch between clock and countdown display.
2. You have to modify all the functions that refresh the displays and block the write requests if the Flag is set.
3. Add a new function that will perform the countdown. Include it in the main loop.
4. Add a new function that will write countdown to the displays. Include it into the function 3 above.
5. Add a new function to set and start the countdown timer and toggle between countdown and clock displays (possibly also with a different font).