aly-fly/EleksTubeHAX #110: HA: effect change

aly-fly, 2024-12-14:
Just integrated the clock into the HA. Works great.
Small annoyance is, that when you want to change the effect, it always sends the pair of two commands ON + Effect. 
The ON command flashes displays to black, back to last display, reloads last display and then loads new clock face. 
I think that when receiving the ON command we should  first check if already enabled and skip resetting the existing display.

[image]

Martinius79, 2025-01-19:
I needed some time to figure out, that this only happens on the PunkCyber clock ;) All other clocks are fine...

Maybe the TFTs are a bit different there...

Will watch this in the near future.

aly-fly, 2025-01-20:
I have the "original" clock where the TFT Enable signal removes the power to the displays (logic and backlight). 
This is the reason why trash is displayed when enabled. 

Fix: Check if Enable pin is already active and return immediately - skip the toggle and display rewrite (with old clock font).

aly-fly, 2025-01-20:
Fixed :)

closed 2025-01-20 by aly-fly

Martinius79, 2025-01-20:
LOL, just finished the branch with the fix, but you was faster ;) https://github.com/Martinius79/EleksTubeHAX/tree/BugfixMQTTBrigthnessChangeTriggersReinit

MqttCommandPowerReceived don't need the check, it is not trigered by any brightness command ;)
But doesn't harm to check ;)

aly-fly, 2025-01-21:
I just wrapped all instances of `tfts.reinit()` command.

aly-fly, 2025-01-21:
> [@aly-fly](https://github.com/aly-fly) do you have a buy me a coffee type link?

@jkosharek  Sorry, no; but thanks for the idea :)

jkosharek, 2025-01-21:
Love the project and would donate to the cause 🤙🏻