aly-fly/EleksTubeHAX #111: Switching to night mode does not refresh all screens

aly-fly, 2024-12-14:
After the night mode kicks in (integrated hardcoded lower brightness), screens are not updated. 
Force update must be added into the code.

[image]

Martinius79, 2025-01-16:
Sorry, I was really busy in the last weeks, but I will take care of this...

Martinius79, 2025-01-18:
Fixed this problem with the PR https://github.com/aly-fly/EleksTubeHAX/pull/112

Problem was, that the precached image in the RAM was just "printed" to the TFT screen again, without having the "dimming" value applied again. Depending on the actual clock time, it drawed the wrong digit X-times to the screens and that stayed until the next redraw.
So invalidating the buffer to force a reload from SPIFFS is fixing this.

PR is in main now.

aly-fly, 2025-01-20:
Works! :)

closed 2025-01-20 by aly-fly