aly-fly/EleksTubeHAX #113: Regression - Setting BACKLIGHT_DIMMED_INTENSITY to 0 does not fully turn off RGB rear lights when dimming is active

bitrot-alpha, 2025-01-26:
Device: ELEKSTUBE V1
FW Version: [a337352](https://github.com/aly-fly/EleksTubeHAX/commit/a3373528a15c5dbabd3ac2ae32bcae4cb369c659)
PIO version: Core 6.1.16 Home 3.4.4
OS: OpenSUSE Leap 15.6
Python: 3.6.15 (yes I got the deprecation warning "Python < 3.7" in VS console, however firmware seems to compile fine)

Setting BACKLIGHT_DIMMED_INTENSITY to 0 in `_USER_DEFINES.h` does not fully turn off RGB rear lights when dimming is active.

Manually setting intensity setting on the clock to 0 does turn off the rear lights.

bitrot-alpha, 2025-01-26:
Specifically, this is with the rainbow pattern. The new rotation speed code seems to be causing the issue.

aly-fly, 2025-01-29:
It's an easy fix. Maybe you can move it into a new PR, so that it gets integrated separately from the WPS issue.

PR #118 (2025-01-30): Make rainbow pattern go dark during dimming and backlight dimmed intensity is 0
small PR that resolves #113 
uses preprocessor directives since BACKLIGHT_DIMMED_INTENSITY itself is a macro and doesn't change during runtime
closed 2025-01-30 by aly-fly