aly-fly/EleksTubeHAX #114: EleksTube IPS Classic Edition 6-Bit Digital Clock for sale

frankcohen, 2025-01-28:
I have 2 new - never used - EleksTube IPS clocks. 

What's in the box:
1 x IPS Clock Base
6 x IPS Display Screen
6 x Quartz Tube
1 x USB-C Cable
1 x User Manual

All in the original packaging. All working.

$75 USD each. Venmo or Paypal accepted.

I can ship anywhere, you pay shipping.

-Frank

aly-fly, 2025-01-28:
I's a pity you are in US, otherwise I would be interested. Overseas shipping is prohibitive expensive...

jkosharek, 2025-01-28:
> I have 2 new - never used - EleksTube IPS clocks.
> 
> What's in the box: 1 x IPS Clock Base 6 x IPS Display Screen 6 x Quartz Tube 1 x USB-C Cable 1 x User Manual
> 
> All in the original packaging. All working.
> 
> $75 USD each. Venmo or Paypal accepted.
> 
> I can ship anywhere, you pay shipping.
> 
> -Frank

Which version? I’m in Florida.

frankcohen, 2025-01-28:
Which country are you in? I’m glad to get a price. -Frank

frankcohen, 2025-01-28:
It’s the original version, firmware version: Ver 3.10, the "Classic Edition”. I have 2 of them in the original packaging, never used. -Frank

jkosharek, 2025-01-28:
> Which country are you in? I’m glad to get a price. -Frank
> […](#)

US, zip 33707

frankcohen, 2025-01-28:
> Do you have a picture Ver 3.10 is supposed to be newer, at least I thought.

https://photos.app.goo.gl/KhvVgoxCrRQn2Nce6