aly-fly/EleksTubeHAX #119: Failure to get correct NTP time may cause the RTC to get overwritten and the RTC to stop entirely

bitrot-alpha, 2025-01-30:
Device: Elekstube v1

After reflashing and power cycling my clock a few times in the span of 5 minutes, I've somehow caused it to lock itself out of the NTP server (hopefully temporarily).

Writing all zeroes to the RTC will cause it to stop because that also writes a 0 to the enable bit of the RTC (shared with the seconds bits).

I'm not sure why I just encountered this and I'm not sure how to reproduce it at this point.

Implementing a zero check in the NTP code should fix this.

bitrot-alpha, 2025-01-30:
waiting until the Geolocation refreshes starts the clock again, although on an incorrect time. a power cycle afterwards fetched the correct time from the NTP server and reset the time to the correct one.

this bug probably won't affect normal users of the clock unless they have extremely high packet loss or the device powering their clock is about to explode.

bitrot-alpha, 2025-01-31:
I'm unsure if this is actually a bug that affects the current version of the repo. I was using a different FW version from @Martinius79 to figure out #115, but I'm unable to get the bug to happen again on the regular FW.

There appears to already be a (normally) sufficient check in place to make sure that the NTP response is valid.

I'll hope it was a spurious hardware glitch since I myself cannot reproduce the bug..

closed 2025-01-31 by bitrot-alpha

aly-fly, 2025-01-31:
First implementations of the NTP used external libraries. After careful examination I found bugs and  many possible ways of unexpected behavior, so I copied them into this project and updated the source files. Most probably you tested one of the early versions of the NTP code.

bitrot-alpha, 2025-01-31:
It happened after I flashed https://github.com/Martinius79/EleksTubeHAX/tree/FixWiFiReconnectProblemsTryOut and then reflashed the main branch without a power cycle. Perhaps the NTP code doesn't clear some memory before reading/writing to it.