aly-fly/EleksTubeHAX #121: Compile fails because of 'SPIFFS'

eku, 2025-03-18:
I've set up a fresh platformio build environment

```
 $ virtualenv ~/src/platformio-core
 $ cd ~/src/platformio-core
 $ . bin/activate
 $ pip install -U platformio
```

and try to build the project. Downloading the dependencies fails for SPIFFS with `Could not find the package with 'SPIFFS' requirements for your system 'linux_x86_64'`.

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2024.12.30) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.0+sha.35321e8b
 - tool-esptoolpy @ 4.8.4
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing SPIFFS
Warning! Could not find the package with 'SPIFFS' requirements for your system 'linux_x86_64'
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit NeoPixel @ 1.12.5
|-- Time @ 1.6.1
|-- DS1307RTC @ 0.0.0-alpha+sha.c2590c0033
|-- TFT_eSPI @ 2.5.43
|-- PubSubClient @ 2.8.0
|-- ArduinoJson @ 7.3.1
|-- Wire @ 3.1.0
|-- SPI @ 3.1.0
|-- FS @ 3.1.0
|-- SparkFun APDS9960 RGB and Gesture Sensor @ 1.4.3
|-- RTC @ 2.5.0
|-- WiFi @ 3.1.0
|-- Preferences @ 3.1.0
Building in release mode
```

Let me know if you need any further information.

Martinius79, 2025-03-18:
Hello eku!

I am not sure on which Linux distribution you are working, so maybe you give me a bit more info?

And which command do you use to do the pio build?

On my Ubuntu Linux, I created the venv, checked out the main branch, run the pio pip install and I am able to build with the command ```pio run --environment EleksTubeHax```. See below for the output.

The biggest differences I see, is that you are using framework-arduinoespressif32 in version 3.1.0+sha.35321e8b and also have the 3.1.0 version of the internal libs (Wire, FS, SPI).

How does your platformio.ini looks like? Specific lib or framework versions set?

---

Actually, some of the libs listed in the platformio.ini are not needed anymore, because they come with the arduino framework for esp32. So it can be, that the LDF is failing here and tries to download a version of the libs, which does not exist for linux from the source he wants to use. 
I removed the unneeded libs in the following branch.
https://github.com/Martinius79/EleksTubeHAX/tree/refs/heads/FixLinuxBuildUnneededDependencies

Please checkout this branch and try it again

---

Output Ubuntu build:

```
(platformio-core) ubuntu@HOMEPC:~/EleksTubeHAX/EleksTubeHAX_pio$ pio run --environment EleksTubeHax
Processing EleksTubeHax (board: esp32dev; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing espressif32
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@6.10.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20017.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@3.20017.241212+sha.dcc1105b has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40501.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40501.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Library Manager: Installing adafruit/Adafruit NeoPixel
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Adafruit NeoPixel@1.12.5 has been installed!
Library Manager: Installing paulstoffregen/Time
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Time@1.6.1 has been installed!
Library Manager: Installing paulstoffregen/DS1307RTC
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: DS1307RTC@0.0.0-alpha+sha.c2590c0033 has been installed!
Library Manager: Installing bodmer/TFT_eSPI
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: TFT_eSPI@2.5.43 has been installed!
Library Manager: Installing knolleary/PubSubClient
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: PubSubClient@2.8.0 has been installed!
Library Manager: Installing bblanchon/ArduinoJson
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@7.3.1 has been installed!
Library Manager: Installing sparkfun/SparkFun APDS9960 RGB and Gesture Sensor
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: SparkFun APDS9960 RGB and Gesture Sensor@1.4.3 has been installed!
Library Manager: Installing makuna/RTC
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: RTC@2.5.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit NeoPixel @ 1.12.5
|-- Time @ 1.6.1
|-- DS1307RTC @ 0.0.0-alpha+sha.c2590c0033
|-- TFT_eSPI @ 2.5.43
|-- PubSubClient @ 2.8.0
|-- ArduinoJson @ 7.3.1
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- SparkFun APDS9960 RGB and Gesture Sensor @ 1.4.3
|-- RTC @ 2.5.0
|-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- Preferences @ 2.0.0
Building in release mode
===== copying TFT config files =====
```

aly-fly, 2025-03-18:
Windows user here. Pulled the changes. Visual Studio Code + PlatformIO plugin. Clean the project and build the 4 MB environment. Build passes without any hiccups.
` *  Executing task in folder EleksTubeHAX_pio: C:\Users\aljaz\.platformio\penv\Scripts\platformio.exe run --environment EleksTubeHax `

```
Environment    Status    Duration
-------------  --------  ------------
EleksTubeHax   SUCCESS   00:01:22.753
```

Martinius79, 2025-03-18:
After trying out some things, I guess, you have an outdated version of the espressif32 platform or belonging framework-arduinoespressif32 package installed.

Please try to use a specific version, by adding 

```
...
[env]
platform = espressif32@6.1.0
...
```

in the `platformio.ini`

Alternatively try to delete your global `.platformio` directory.

```
rm -rf ~/.platformio/*
```

and the `.pio` dir

```
cd ~/EleksTubeHAX/EleksTubeHAX_pio
rm -rf .pio/
```

then rerun 

```
pio run --environment EleksTubeHax
```

and tell us the output.

eku, 2025-03-19:
> After trying out some things, I guess, you have an outdated version of the espressif32 platform or belonging framework-arduinoespressif32 package installed.
> 
> Please try to use a specific version, by adding
> 
> ```
> ...
> [env]
> platform = espressif32@6.10.0
> ...
> ```
> 
> in the `platformio.ini`

Yes, I use a virtualenv for all platformio based projects here.

Of course, this assumes that **each project clearly defines the versions** of the dependencies used. This is **not** the case with this project.

By specifying the version of the espressif framework, as you suggested, this project will now also build.

Please specify the corresponding versions in the project's own `platformio.ini`.

aly-fly, 2025-03-19:
> Please specify the corresponding versions in the project's own `platformio.ini`.

This was intentionally left out, so that every fresh checkout (or total rebuild) fetches the most recent libraries and frameworks (to get the most recent bug fixes).
I don't know why this was not working for you. (I don't know much about virtualenv..)

Martinius79, 2025-03-19:
> 
> Yes, I use a virtualenv for all platformio based projects here.
> 

Sadly, PIO is not really using virtual envs like other python projects. It has a hybrid structure. Platform and framework packages (like for arduino and esp32) are installed into the global platformio folder, like ```~/.platformio```. Libriaries and other stuff is installed in the "project" folder, like ```.pio"```.
So if you ever installed another frameword package, it is still "there" in the global folder.
As I told, I have no idea, why your PlatformIO installation choosed this version, instead of getting the newest (6.10.0) one.

I had a simular problem some time ago on Windows. It got the esp framework version forked by the tasmota project. I didn't noticed for some time and have still no idea, what happend there exactly.

That is one of the reasons, why we are using the GitHub actions, to verify, that the main is buildable on a "clean" system for Linux (ubuntu in this case) and Windows.

> Of course, this assumes that **each project clearly defines the versions** of the dependencies used. This is **not** the case with this project.
...
> 
> Please specify the corresponding versions in the project's own `platformio.ini`.

I totaly agree with you! I also mentioned some time ago, as a DevOps Engineer, I always want to have "unique" builds with precise versions.
But like aly-fly wrote, the idea is, to have the newest version of everything.

I added some comments about that in the platformio in the PR:
https://github.com/aly-fly/EleksTubeHAX/pull/122

> By specifying the version of the espressif framework, as you suggested, this project will now also build.

Ok, so the platform package was the problem there!
Solved for you?

eku, 2025-03-19:
> This was intentionally left out, so that every fresh checkout (or total rebuild) fetches the most recent libraries and frameworks (to get the most recent bug fixes).

This also involves a risk of breaking.

If you don't specify a particular version than any version satisfies the dependencies. This includes any old version already downloaded. This happend to me.

aly-fly, 2025-03-19:
Yes, that is true. As every coin has two sides, this is also relevant here. With fixed versions you always get buildable and working project, but project maintainers always have to look for newer versions and manually update and test accordingly.
Without fixed versions, you should always get the latest version of the libraries (and framework) and maintainers only have to jump in when a breaking change is introduced in one of the dependencies. That's why Platformio.ini has both sets of libraries - with and without versions and each user can set them as own personal preference.

Martinius79, 2025-03-20:
I guess, we will not come to a "good solution for everyone" with the versionsnumber topic :)

We can discuss this in another issue then...So closing this one, because the problem was solved...

closed 2025-03-20 by Martinius79