aly-fly/EleksTubeHAX #123: HA MQTT discovery

aly-fly, 2025-03-22:
After modifying the MQTT code for another application (btw. thanks for the excellent template to kick-off my other project) I found out in the HA documentation that Discovery messages must be re-published after HA is started or rebooted.
Therefore, Clock should be subscribed to the `homeassistant/status` and listen for the `online` message. And send the Discovery messages after receiving that status. At least this is what I understand from the docs: https://www.home-assistant.io/integrations/mqtt/#discovery-messages-and-availability

Will try to implement this onto my other development work and see if I come to the working solution.

Another similar topic - _Last Will / availability status_.
I am a fan of seeing when device has gone offline. Usually this is accomplished with posting a Last Will message at the MQTT broker, which is automatically published if connection is dropped. I have done this with other simpler MQTT projects, not yet in connection with HA. If someone has a working code, it would be great to add it.

Martinius79, 2025-03-28:
Hello aly-fly,

I created a branch with some code changes to implement the Last Will and Birth messages for both sides.
See https://github.com/Martinius79/EleksTubeHAX/tree/AddMQTTBirthAndLastWillMessagesFromTheDevice

The device itself is setting a LWT automatically with the changed initalization of the PubSubClient.

And by subscribing to the homeassistant/status topic and handling it in the MQTTCallback function, a new discovery is triggered, if HA was offline and is coming back online.

If HA is offline, no MQTT messages will be send.

You can try it, by stopping the Mosquitto AddOn in HA and restart it then. Device is showing "NOMQTT" and then switching back and sending the discovery messages again.

Please review the code and tell me, if it is what you was looking for?

aly-fly, 2025-03-28:
In the mean time I have heavily reworked and tested the MQTT driver myself. 
It is under a different project: https://github.com/aly-fly/BigWallClock/blob/main/Firmware/src/Mqtt_client_HA.cpp
If you go looking at the code, note that in that version I deleted section for connecting to an outside service.

Note I also changed and tested the Message Retain function, and it kind of works better if all messages are not retained. 
_#define MQTT_RETAIN_DISCOVERY_MESSAGES                    false         // https://www.home-assistant.io/integrations/mqtt/#discovery-messages-and-availability
// retain discovery = true:  Configuration is stored in the MQTT broker. HA directly responds to the existing messages. Configuration can be deleted on the MQTT broker (use MQTT Explorer program).  Restart of the broker, device or HA does not clear the config.
// retain discovery = false: Configuration is stored in the HA and can be deleted in Settings -> Devices -> MQTT. Restart of the broker, device or HA does not clear the config._

I like the idea to suppress sending the messages if HA is offline.  Will add that to my code as well.
I think we both made a pretty good job and now it is hard work to merge everything together :)

aly-fly, 2025-03-28:
One more thought on _If HA is offline, no MQTT messages will be send._
As long as Broker is online, there is nothing wring with sending the messages, even if no-one is listening. Maybe for some reason (network issues,...) we miss the "online" message and we don't send the messages. Also if you boot the device later than HA, you will not see the "online" message, because it is not persistent.
I think it's best to delete this function.

Martinius79, 2025-03-28:
Sidetopic: You seem to have the same "stupid" idea of having a bigger version of the clock, like me ;)
Just in a different way.
I just started with it, by ordering some bigger LCDs, because I am not sure how to do it. Just using six or more ESP32 dev modules and syncing them, or Arduinos. Or using an older PI...

Martinius79, 2025-03-28:
> One more thought on _If HA is offline, no MQTT messages will be send._ As long as Broker is online, there is nothing wring with sending the messages, even if no-one is listening. Maybe for some [...]

Yes, after checking the 'online' message for the `homeassistant/status` topic, I see that HA doesn't set the 'retain' flag, and the QoS is 0, so there is no guarantee of receiving the message.
So I totally agree.

Martinius79, 2025-03-28:
> In the mean time I have heavily reworked and tested the MQTT driver myself. It is under a different project: https://github.com/aly-fly/BigWallClock/blob/main/Firmware/src/Mqtt_client_HA.cpp If you go looking at the code, note that in that version I deleted section for connecting to an outside service. And added entity `device_class` and `icon` into the discovery, along with mandatory `platform` section.

I need to check the platform section stuff... I’ve never looked at the discovery stuff so closely before...

> Note I also changed and tested the Message Retain function, and it kind of works better if all messages are not retained. _#define MQTT_RETAIN_DISCOVERY_MESSAGES false // [...]

I fought with the retained messages of the clocks a lot because I have so many devices, and sometimes I missed giving them unique names. So, I agree that non-retained (non-retain marked) messages might be a good idea.  

BTW, some time ago, I started to add some "extra" defs in my personal `_USER_DEFINES.h` to avoid this problem.

```
// --- MQTT - Just for naming ---
#if defined(HARDWARE_Elekstube_CLOCK)
#define DEVICE_MANUFACTURER "EleksTube"
#define MQTT_SUFFIX "_Elekstube"
#define HARDWARE_VERSION "1.x"
#elif defined(HARDWARE_Elekstube_CLOCK_Gen2)
#define DEVICE_MANUFACTURER "EleksTube"
#define MQTT_SUFFIX "_ElekstubeGen2"
#define HARDWARE_VERSION "2.x"
#elif defined(HARDWARE_SI_HAI_CLOCK)
#define DEVICE_MANUFACTURER "SI HAI"
#define MQTT_SUFFIX "_SI_HAI"
#define HARDWARE_VERSION "1.x"
#elif defined(HARDWARE_NovelLife_SE_CLOCK)
#define DEVICE_MANUFACTURER "NovelLife"
#define MQTT_SUFFIX "_NovelLifeSE"
#define HARDWARE_VERSION "1.x"
#elif defined(HARDWARE_PunkCyber_CLOCK)
#define DEVICE_MANUFACTURER "PunkCyber"
#define MQTT_SUFFIX "_PunkCyber"
#define HARDWARE_VERSION "1.x"
#elif defined(HARDWARE_IPSTUBE_CLOCK)
#define DEVICE_MANUFACTURER "IPSTUBE"
// #define MQTT_SUFFIX "_IPSTUBEH401"
#define MQTT_SUFFIX "_IPSTUBEH401_WITH_LEDSTRIPE"
// #define MQTT_SUFFIX "_IPSTUBEH402"
#define HARDWARE_VERSION "1.x"
#else
#define DEVICE_MANUFACTURER = "Unknown"
#define MQTT_SUFFIX "_Unknown"
#define HARDWARE_VERSION "x.x"
#endif
#define MQTT_BASE_NAME "EleksTubeHAXClock" // Base name of the MQTT topics
...
#define MQTT_HOME_ASSISTANT // Uncomment if you want Home Assistant (HA) support (MQTT_ENABLED must be also enabled)
// all following MQTT_HOME_ASSISTANT_* settings require MQTT_HOME_ASSISTANT to be enabled
#define MQTT_HOME_ASSISTANT_DISCOVERY                                         // Uncomment if you want HA auto-discovery (requires MQTT_HOME_ASSISTANT)
#define MQTT_HOME_ASSISTANT_DISCOVERY_DEVICE_MANUFACTURER DEVICE_MANUFACTURER // Name of the manufacturer shown in HA
#define MQTT_HOME_ASSISTANT_DISCOVERY_DEVICE_MODEL MQTT_BASE_NAME MQTT_SUFFIX // Name of the model shown in HA
#define MQTT_HOME_ASSISTANT_DISCOVERY_SW_VERSION FIRMWARE_VERSION             // Firmware version shown in HA
#define MQTT_HOME_ASSISTANT_DISCOVERY_HW_VERSION HARDWARE_VERSION             // Hardware version shown in HA
```

...
> Comment on your `String` variant: I don't really like it, because this will allocate RAM and join constants in RAM before sending. With `concat123` variant everything is already done and optimized by the compiler without using ESP's additional resources.
...

Even if we have "enough" DRAM free most of the time, I agree that it is not *needed* to use a `String` object here. We use "fixed" strings most of the time.
I am just lazy and sometimes use the *easiest* option for myself. 
I am not sure if the `String` class can use PSRAM instead (using `heap_caps_malloc` or something)... I will check that. 
BTW, we have a wild mix of ways to define and work with strings in the project. 😉

...
> I think we both made a pretty good job and now it is hard work to merge everything together :) I might have some time to do it and test it.
...
Thx! I will also try to improve the branch a bit :)  
Tell me if you plan to merge in the near future.

aly-fly, 2025-03-28:
> Sidetopic: You seem to have the same "stupid" idea of having a bigger version of the clock, like me ;) 

Mine is more or less already working. More analog style, driven with a stepper motor and synced to NTP. 
1 meter diameter  🤠

[image]

Martinius79, 2025-03-28:
One meter, nice work!!!

I just want a "big" version of a pseudo tube clock for the wall. Like 5"-7" or more for each digit. 
So it should a bit look like this one:
https://bigtimeclocks.biz/collections/countdown-clocks/products/extra-large-8-led-countdown

But not like a LED clock, should look more like the "original" ones, just bigger. 
https://www.daliborfarny.com/product/puri-nixie-clock/

No exact ideas yet, just planning and searching for components in the moment.

Martinius79, 2025-03-28:
I see, you seem to use a J-Link debugger...Do you have a "real" Segger one or a clone?
In both cases, which di you have and which one can you recommend?
Does it work well for you, at least with the ESP32 dev boards?

aly-fly, 2025-03-28:
Original Segger.  [Base Compact](https://www.segger.com/products/debug-probes/j-link/models/j-link-base/). "Borrowed" from my daily work. I don't need tracing and time optimization, just simple and reliable debugger. Work gets so much easier than `Serial.println()`. 🥇  Highly recommend it for working on complex projects. Flash code download is also a few times faster than serial connection. Works well on ESP32 and doesn't have issues with two cores or RTOS.

Recently I am moving towards ESP32-S3, which has built-in debugger. The software part in PlatformIO is not as reliable as Segger, but it saves quite a few IO pins. Most of the modules come with 16 MB flash and 8 MB PSRAM which makes development much faster (but less optimized haha).
My favorite dev board is Camera board (but without camera) - it has built-in SD-Card slot connected to MMC controller (20-40 MHz clock!) a small addressable LED and all pins exposed. With 16M + 8 M storage. ([link](https://www.aliexpress.com/item/1005007495763537))

aly-fly, 2025-03-28:
@Martinius79  I finally took some time to write the Readme of my clock. You are welcome to take a look :)
https://github.com/aly-fly/BigWallClock

Regarding your clock - have a look at CYD - Cheap Yellow Displays. You can get a TFT + ESP32 + 4 or 5 or 7 inch LCD for a reasonable money. Here is a list of (almost) all variants: https://github.com/rzeldent/platformio-espressif32-sunton?tab=readme-ov-file#supported-boards

Martinius79, 2025-03-28:
> [@Martinius79](https://github.com/Martinius79) I finally took some time to write the Readme of my clock. You are welcome to take a look :) https://github.com/aly-fly/BigWallClock

Wow! That's a lot of work but looks great! You even build the clockwork yourself! Impressive!

The readme is a good overview to start and see what you did and which components are used.
Everyone with a 3D printer also knows how to use the STL files I guess ;)
You should also explain or show, how you did the wood work and the LED stripe wiring and fixing.
But I am even more interested in building my own clock now ;)

> Regarding your clock - have a look at CYD - Cheap Yellow Displays. You can get a TFT + ESP32 + 4 or 5 or 7 inch LCD for a reasonable money. Here is a list of (almost) all variants: https://github.com/rzeldent/platformio-espressif32-sunton?tab=readme-ov-file#supported-boards

Thx for the link! I directly ordered a 7 inch display with S3 board :) I just saw them with HDMI connector yet...

Martinius79, 2025-03-28:
> Original Segger. [Base Compact](https://www.segger.com/products/debug-probes/j-link/models/j-link-base/). "Borrowed" from my daily work. I don't need tracing and time optimization, just simple and [...]

A real one, nice! I thought about buying a V12 clone, but still a lot of money, just for placing around with it.
I decided to start with the (cheap) ESP32-Prog board, because I am so tired of the "Serial.Print" orgies I am doing ;)
 
> Recently I am moving towards ESP32-S3, which has built-in debugger. The software part in PlatformIO is not as reliable as Segger, but it saves quite a few IO pins. Most of the modules come with 16 [...]

I already have some S3 and C3 ESP boards flying around, but never tried the onboard debugging featueres, will give it a try.

aly-fly, 2025-03-29:
> You should also explain or show, how you did the wood work and the LED stripe wiring and fixing.

For wood work I relied on the relatives with the woodworking skills :)
Other points I have added to the description.

> Thx for the link! I directly ordered a 7 inch display with S3 board :) I just saw them with HDMI connector yet...

Such big displays only come with HDMI or parallel RGB interface. SPI is too slow to transfer so much data.

Martinius79, 2025-03-29:
I updated the branch https://github.com/Martinius79/EleksTubeHAX/tree/AddMQTTBirthAndLastWillMessagesFromTheDevice

Availabilty status is working now for the device. After 30 seconds unavailable the entities are grayed out in HA.

HA online check removed.

String is replaced with const char buffer, because PSRAM usage is only possible by overriding malloc project wide. Don't know the effects, so didn't try.

Platform is not needed to be set in the discovery messages, it is not interpreted at all. HA selects the platform automatically only by the given topic (<DEVICEID>/<PLATFORM>/blah/blub).

Tried to choose some nice icons from the MDI library.

Using not-retained messages now for discovery and state messages. Only status stays retained.

Added some note about retain flag for messages in the _USER_DEFINES.h.

Sidetopics: Renamed the flag for TFT enabled state. Fixed that error state was shown on the display, even if they are "turned off".

aly-fly, 2025-03-29:
Ok, cool.
Should I check-out your branch and send you files with my comments, or will you push a commit and I make a fork out of that?

Martinius79, 2025-03-29:
All is fine for me ;) 
I can create a PR and you comment/commit there. 
You can check out the branch and then do a PR on it (in my then fork). 
Or you send me the files otherwise.

aly-fly, 2025-03-29:
ok, I pulled your branch. Working on it now.

I am thinking, why do we nee to split #HA and HA_Discovery.  Probably everyone that has HA also likes to have Discovery, so I would delete all `#ifdef MQTT_HOME_ASSISTANT_DISCOVERY` lines.. And discovery is always active if HA is enabled?

aly-fly, 2025-03-29:
Can you give me write access to your branch? Otherwise here attached are modified files.

[MQTT_client_ips.zip](https://github.com/user-attachments/files/19521994/MQTT_client_ips.zip)

Martinius79, 2025-03-29:
Invited you as Collaborator :)

Martinius79, 2025-03-30:
I integrated your changes in this commit:
https://github.com/Martinius79/EleksTubeHAX/tree/2db191c07cc83edf47ec3d22e3ee0d8e7edd7589

And the code looks way cleaner with the MQTTPublish functions!

And I also agree, that using Home Assistant without discovery messages makes no sense. So I will remove the defs for it in the next commit.

Martinius79, 2025-03-30:
Removed the discovery condition: https://github.com/Martinius79/EleksTubeHAX/tree/AddMQTTBirthAndLastWillMessagesFromTheDevice

I also changed the retain flag to true for the connection/alive messages.

I will create a Pull Request now, so you can either comment there, or change the code directly in my branch and commit. PR will be updated then.

Martinius79, 2025-03-30:
PR created: https://github.com/aly-fly/EleksTubeHAX/pull/124

aly-fly, 2025-03-30:
> EDIT: Something seems to be not as it should be with the discovery messages, posted to often. Will check tomorrow.

I just briefly tested the code, was not too rigorous. I think we have to fully test it before commit.
My code is from other project where I have posting of Discovery and Online messages in the main loop, not in the startup code. And they are enabled / disabled by boolean flags. Here is a bit different, so this may be the cause. A review and test is required. Best that we do it in your branch before touching the main branch.

Martinius79, 2025-03-30:
I updated the code (and tested a bit).
Main changes:
Discovery messages have the retain flag set to true.
Commented some unneeded functions and variables and definition -> can be deleted!?
Fixed some bugs after merging.
removed unneeded second component string in topic (homeassistant/[component]/[unique_id] instead of homeassistant/[component]/[unique_id]/[component])
removed the discovery defines

And yes, I totally agree, that we need to test this properly.
I will test it on all my devices and with diffirent setting (pure MQTT, HA MQTT, without, device on, device off, broker on, broker off, etc.)

Martinius79, 2025-03-30:
I updated the code a bit more.
https://github.com/Martinius79/EleksTubeHAX/tree/AddMQTTBirthAndLastWillMessagesFromTheDevice

Tested it on all my devices.
Readme.md and pictures needs to be updated...

@aly-fly: Maybe you can check again?

aly-fly, 2025-04-02:
- Tested MQTT with SmartNest and fixerd the errors
- Tested the HA - works fine
- Removed the general MQTT define 
- Tested the clock without any MQTT service enabled
- Updated the Readme (using the https://www.jekyllpad.com/tools/online-markdown-wysiwyg-editor)

As far as i am concerned, this branch is good to merge.
You can test it too and then merge it.

Martinius79, 2025-04-03:
I watched the changes and it looks great :) 
I wasn't able to test the code with all clocks yet and I also want to exchange so of the pictures.
So give my a bit more time to do the round up...

And shame on me...So many mistakes in the Readme file! I should use more AI stuff to check my writings...:)

aly-fly, 2025-04-03:
I used just a spellchecker integrated into the web browser. 

I have also checked the code using the `cppcheck` tool in platformio. There are a few minor issues in the libraries - dependencies, but all-in-all the code is good looking :)

Martinius79, 2025-04-03:
Fixed a little problem if TLS is used (SPIFFS.h needed).
Adjusted the MQTT retry logic to avoid too many reconnects.
Adjusted the MQTT connection retry timeout to 30 seconds, not 3 seconds.

All clocks tested and working with HA.

Also tested smartnest.cz (Plain) and HiveMQ (Plain and HA) as brokers with NovelLifeSE. Mosquitto local (Plain and HA).

So all seems to be working now ;)

aly-fly, 2025-04-04:
This was a nice marathon :)
GitHub checks have passed and code was merged. 
This branch can be deleted if you wish. Or you can keep it as I see you have a full list of old branches :)

I don't have skills to easily build binary files, if you have a tool at hand, that would be great.

Martinius79, 2025-04-04:
Yes, it was ;) 

And yes, I try to delete the "unused" branches, but still to many left ;) So much to do and so many ideas and not enough time...

BTW: I received a 5" and a 7" display already and I will play around with it soon...

And there is no "nice" way in the moment to create the matching bin files...
I use the "default" USER_DEFINES.h with the features active like described in the firmware readme and then I set the firmware version string to the "next expected" version from the github workflow run. In the moment this would be 1.0.13, (if the fictional PR is the next to be integrated into the main branch)...Then I set the CREATE_FIRMWAREFILE define in the platform.io and build for every device (changing the hardware defines for each device). Then renaming, copying and commiting...

MQTT is not active in the pre-builded firmware files, so I guess it is not "needed" to have the bin files updated...I will skip it for this release...

I am working on a fix for the RTC support of the NovelLifeSE...The manufacturer used a second I2C bus to connect the RTC chip...So I need to modify a lib, because by default the code is only supporting one I2C bus...
Is it OK, to place a modified lib in the /lib dir of the pio folder? It is working, but it is some kind of "fork" then...Don't know how this I then with the IP and the license stuff...We already using some "modified" code (for NTP i.e.) and maybe it is a good idea to "extract" it and use a modified version of the lib then...

Hope we will do soemthing like this again :)

aly-fly, 2025-04-05:
> The manufacturer used a second I2C bus to connect the RTC chip.

Do we use two I2C buses at once? If not, this means that you should be able to use the default hardware and reroute the pins through the io matrix.

> I need to modify a lib, because by default the code is only supporting one I2C bus... 

Most probably there is someone that has already done that - and then you just call the appropriate lib.
I found this - use `Wire `and `Wire1 `. This are two I2C hardware masters.
https://forum.arduino.cc/t/using-twowire-i2c2-with-esp32-two-pressure-sensors/1061318/10

```
void setup()
{
  Wire.begin();         // this starts the first I2C bus at 21, 22
  Wire1.begin(26, 25);  // this starts the second I2C bus
}
```

> I need to modify a lib, because by default the code is only supporting one I2C bus... Is it OK, to place a modified lib in the /lib dir of the pio folder? 

Yes this is an alternative, if the above does not work.

aly-fly, 2025-04-06:
One more bug to be fixed. Just found this in HA logs:

> Log Details (WARNING)
> Logger: homeassistant.components.mqtt.light.schema_json
> Source: components/mqtt/light/schema_json.py:156
> integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
> First occurred: 20:43:14 (2 occurrences)
> Last logged: 20:43:14
> 
> **Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed.** Got: {'device': {'identifiers': ['clock'], 'manufacturer': 'EleksMaker', 'model': 'Elekstube IPS', 'sw_version': '1.1', 'hw_version': '2.3.04', 'connections': [['mac', '40:91:51:AC:B4:3C']], 'name': 'Elekstube IPS'}, 'unique_id': 'clock_main', 'object_id': 'clock_main', 'availability_topic': 'clock/status', 'icon': 'mdi:clock-digital', 'schema': 'json', 'state_topic': 'clock/main', 'json_attributes_topic': 'clock/main', 'command_topic': 'clock/main/set', 'brightness': True, 'brightness_scale': 255, 'color_mode': False, 'effect': True, 'effect_list': ['Nixie Tube', 'Blue Ribbon', 'Colors', 'Modern Neon', 'Yellow Dots', 'Cyan and Pink', 'Thin Blue'], 'name': 'Main'}. **This will stop working in Home Assistant Core 2025.3**
> 
> Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'device': {'identifiers': ['clock'], 'manufacturer': 'EleksMaker', 'model': 'Elekstube IPS', 'sw_version': '1.1', 'hw_version': '2.3.04', 'connections': [['mac', '40:91:51:AC:B4:3C']], 'name': 'Elekstube IPS'}, 'unique_id': 'clock_back', 'object_id': 'clock_back', 'availability_topic': 'clock/status', 'icon': 'mdi:television-ambient-light', 'schema': 'json', 'state_topic': 'clock/back', 'json_attributes_topic': 'clock/back', 'command_topic': 'clock/back/set', 'brightness': True, 'brightness_scale': 7, 'color_mode': True, 'supported_color_modes': ['hs'], 'effect': True, 'effect_list': ['Dark', 'Test', 'Constant', 'Rainbow', 'Pulse', 'Breath'], 'name': 'Back'}. This will stop working in Home Assistant Core 2025.3

Martinius79, 2025-04-07:
> Do we use two I2C buses at once? If not, this means that you should be able to use the default hardware and reroute the pins through the io matrix.
>
>
> Most probably there is someone that has already [...]

Yepp, that's the problem, the libs for RX8025T RTC chip support use of Wire.begin hardcoded. So I modified one of the libs to have and init function with pins given and using Wire1 then...
see https://github.com/Martinius79/EleksTubeHAX/tree/TryOutsForRTCOnNovelLifeSE

Martinius79, 2025-04-07:
> One more bug to be fixed. Just found this in HA logs:

Seems, as if we need to set both `color_mode`and `supported_color_mode`

see:
https://developers.home-assistant.io/docs/core/entity/light/#color-modes
and
https://developers.home-assistant.io/blog/2024/02/12/light-color-mode-mandatory/

I will try to implement it as soon as I find the time...

Martinius79, 2025-04-07:
https://github.com/aly-fly/EleksTubeHAX/pull/127

Quick fix ;)
The state messages should still report the color_mode! Just in the discovery_messages they are deprecated...

Working for my clocks. After restarting my HA host, the error and the warning is gone.

Please review and approve

closed 2025-04-08 by aly-fly